Koenkk/zigbee2mqtt #1315: Implement nedb setAutocompactionInterval

tb-killa, 2019-03-24:
As we often see the nedb (objectbox use them) as storage base for zigbee-shepherd we got multiple duplications inside this database file.

After searching for a valid way to get this compressed i found this simple hint:

https://github.com/louischatriot/nedb#persistence

> You can manually call the compaction function with yourDatabase.persistence.compactDatafile which takes no argument. It queues a compaction of the datafile in the executor, to be executed sequentially after all pending operations. The datastore will fire a compaction.done event once compaction is finished.

> You can also set automatic compaction at regular intervals with yourDatabase.persistence.setAutocompactionInterval(interval), interval in milliseconds (a minimum of 5s is enforced), and stop automatic compaction with yourDatabase.persistence.stopAutocompaction().

I simple this this test with patch the ``node_modules\objectbox\lib\db.js`` with this

``
function Db (fileName) {
    if (typeof fileName !== 'string')
        throw new Error('fileName must be a string.');

    this._db = new Datastore({ filename: fileName, autoload: true });
    this._db.ensureIndex({ fieldName: 'id', unique: true }, function (err) {
        if (err) throw err;
    });
	this._db.persistence.setAutocompactionInterval('5000');
}
``
This would set automatic compaction to every 5secs.

I know this is maybe to short (60 Secs would be better) but this is a first test to check if this could work probably.

I see that this really compress my database entrys.

Now we need check if this data are valid after some more days of use.

lolorc, 2019-03-24:
there's something funny, the compaction should occur when zigbee2mqtt opens the database

this simple piece of code does it for instance.
```
var Datastore = require('nedb')
  , db = new Datastore({ filename: 'database.db', autoload:true });
```

Koenkk, 2019-03-25:
I would advise not to modify the database when zigbee-shepherd is started.

I think this should be put in the `stop` routine (https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/controller.js#L171) after `this.zigbee.stop(callback);`. This makes sure that every after zigbee2mqtt stop the database is clean.

lolorc, 2019-03-25:
oh I wouldn't run this while it's running, I'm just running it before I start zigbee2mqtt and also before having to alter database.db by hand.

tb-killa, 2019-03-26:
After two days of run with intervall  with 5 min it run without problems in periodic way.

But yes we should handle the database close in right way as @Koenkk point.

As mentioned on the first post this is a testing way, because I'm not happy with the often produced duplicated entry's in the database file if I search for specific device entry's.

lolorc, 2019-05-27:
(this issue should probably stay opened)

closed 2019-08-02 by stale[bot]

tb-killa, 2019-09-21:
hi @Koenkk 
do you think we should add this to your new herdsman implementation or do you would implement something like this as your own ones ?
I think this would bring benefits.

Koenkk, 2019-09-22:
Yes we can implement it. Two things however:
- it seems that tgis is already done now at every startup (not sure if this was the case before)
- https://github.com/Koenkk/zigbee-herdsman/pull/21 will remove the complete nedb dependency.

tb-killa, 2019-09-22:
Hi nice to read that we would see something new.
Actually we need an Z2M restart to get the multiple entry's of the database themselves completed to ones per device. 
If the auto trigger worked this happened directly and I doesn't see any failures from this even if the trigger comes up on same time like device joined or removed.
But let's see if your new herdsman pr is merged what benefits would come up :)

Koenkk, 2019-09-23:
@tb-killa latest dev branch removes usage of nedb. `database.db` will always contain one entry now.