Koenkk/zigbee2mqtt #1316: status code: 205. No network route. Please confirm that the device has (re)joined the network.

frank3523, 2019-03-24:
Did you read the FAQ?
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device?
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question?
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
- First make sure your on the latest version of zigbee2mqtt.
- Make sure you are running the latest CC253X firmware from https://github.com/koenkk/z-stack-firmware.
- Provide a clear description of the problem.
- If applicable, provide steps how to reproduce the problem.
- Provide a log with `log_level` set to `debug`, see https://www.zigbee2mqtt.io/configuration/configuration.html
- DON'T copy logs directly here, post a link to https://hastebin.com/ or https://pastebin.com/.

frank3523, 2019-03-24:
After installing a new HAssio because my sd card was crashed I can't get my Ikea lamp to work.
I restored to latest snapshot.
I already flashed the latest firmware tried to do the trick with the pan_id but that did not work.

I can't pair the light bulb anymore. 
Any help would be appreciated.

Koenkk, 2019-03-25:
Can you provide the complete log when trying to pair?

pistoletov1974, 2019-04-18:
I have the same problem to. After i reflash stick to https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/max_stability/CC2531/CC2531_MAX_STABILITY_20190315.zip this firmware my Xiaomi lamp became unresponsible. 
When i connect lamp to 220 v i see the mqtt messages with state ON and current Brightness & ColorTemp values.  I try to reconeect lamp by many times on-off lamp but she does not go to link mode (lamp blinking). I attached my log. Please help with some advice. 
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3094037/log.txt)

Koenkk, 2019-04-19:
@pistoletov1974 can you re-pair the device?

pistoletov1974, 2019-04-19:
I tried to do this. But lamp did not enter in link mode. I on-off many times lamp from 220  but unfortunatelly.

closed 2019-06-26 by stale[bot]