Koenkk/zigbee2mqtt #1317: Rejoin CC2530 + CC2591 router issues

CMDR-Sloma, 2019-03-24:
I'm on the latest stable firmware and hassio addon. I quite happy with the way it works now with no major issues. I followed a guide:

https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

Bought all the parts and built it as pictured there. It was working fine for a while but due to power loss it has been kicked out of the network and since has never rejoined. I tried restarting router but it didn't rejoin. I also tried re-flashing the firmware but it didn't join (I'm not sure if it was because retain IEEE address flag was checked). 

Since there is no reset switch on the board is there any other hardware way to reset the router like I can do with my other CC2531 sniffer/router? Unscrewing the enclosure every time there is the power loss is not perfect solution to this problem.

Is there any software way to remove it from the database apart from MQTT device removal command? I already tried removing and rejoining but it didn't work.

Any help would be really appreciated.

tunip, 2019-03-24:
I rejoined all my 4 CC2530 + CC2591 routers a few days ago successfully.
First I removed it over MQTT cmd. Then I flashed the devices. Sometimes I need to flash a second time, otherwise it would not pair. I wait round about 2-3 minutes until the device paired successfully. If not, then I flash a second time. Never flashed a third time for joining...

CMDR-Sloma, 2019-03-24:
But this involves disassembly of entire thing... I would like to know if there is a software way to reset the router without re-flashing or if I can mod it by adding hardware reset switch so it can rejoin without flashing again. If not my only option is CC2531 with antenna mod I guess. So my question still stands, is there other way?

Koenkk, 2019-03-25:
@ptvonfo shouldn't this be the case with the latest router firmware?

ptvoinfo, 2019-03-25:
The last (and the previous) firmware scans all possible channels. From my experience, pairing may take up to 10 minutes.

CMDR-Sloma, 2019-03-25:
And how can I check if there are two different networks present?

tunip, 2019-03-27:
tested this and it will not come back after 10 minutes.

I powered off a router (CC2530 + CC2591) and wait for ~5 minutes and powered it back on. since 20 minutes now, this router is not online.

tunip, 2019-03-27:
> СС2530 can be re-paired if you power on/power off it three times (power on, wait 2 seconds, power off, repeat this cycle three times).

This worked for re-pairing.

Shred99, 2019-05-03:
I've had exactly the same issue.

Everything was working and stable. I experienced a prolonged power outage (router lost power immediately, coordinator and Raspberry Pi died 3.5 hours later when the UPS ran dry).  Now the router has fallen off the network.  I've tried turning it off for 5 minutes and leaving it on for hours and hours... but it never connects.  I've restarted the coordinator and the software on the RPi.

The end devices that were connecting via the router have also disappeared from the network - even though 2 of the 3 devices can easily communicate with the coordinator directly.

closed 2019-07-09 by stale[bot]