Koenkk/zigbee2mqtt #1318: Coverage and devices not always responding

copystring, 2019-03-24:
Did you read the FAQ? Yes
https://www.zigbee2mqtt.io/information/FAQ.md

Unsupported device? No
https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

A question? Not sure if that helps... seems to be a deeper problem
Read the documentation: https://www.zigbee2mqtt.io/

Bug report:
Im using the latest version of the firmware (20190223) and latest version of zigbee2mqtt (1.2.1)
The CC2531 router has CC2531_router_2018_09.zip as firmware.

I have about 48 devices all together and I have been watching the mesh network closely for weeks. It has always been a little trouble which needed me to restart my raspberry pi every x weeks. When that happened, no device responded and no shutter contact or temperaturesensor updated any values. It did not bother me that much so I lived with it.
However, it got a lot worse since upgrading to the latest firmware and version of zigbee2mqtt.

The most problematic device at the moment is the router and my outside gardenpoles. The gardenpoles are not always reaction to the commands I sent via fhem.
The router has hardly any connected devices. The closest devices are "Briefkasten" and "Eingangslicht"
I do find it very odd, that a lot of devices are not connected with a "line" to anything and are still online... is that a bug? Do I misunderstand the system?

With about 3 routers (2 power plugs + 1 dedictated CC2531 router) plus some lamps/light which are not always on, there should be plenty of coverage, right?

My main concern is that the CC2531 router has hardly any connections even though it should and "Eingangslicht" is not able to react to the commands I sent.

I will attach a picture of the mesh.

https://imgur.com/UfECRoP

If there is any way to produce a helpfull log, let me know. I have looked at the logs and found nothing usefull.

Kind regards

Felix

tb-killa, 2019-03-25:
Hi, the MAP function isn't perfect as it point out. We should maybe give a big hint that this isn't fully working and only be in an experimental state.

From what I could confirm are the problematic with the link quality and the strange thing that often devices doesn't be in a complete way joined.

copystring, 2019-03-25:
Yes. I remembered this when I went to bed.

Thinking about it I do get `Error: AF data request fails, status code: 205. No network route. Please confirm that the device has (re)joined the network.` fairly often when trying to control "Eingangslicht" which is the light outside the house near the door.
The light is a Gardenpole Mini RGBW OSRAM. That was not supported out of the box but, because of the similarities, I assumed its the same as Gardenpole RGBW-Lightify just the small version.
To make it work I added `
    {
        zigbeeModel: ['Gardenpole Mini RGBW OSRAM'],
        model: '4058075047853',
        vendor: 'OSRAM',
        description: 'Smart+ Gardenpole Mini RGBW OSRAM',
        extend: generic.light_onoff_brightness_colortemp_colorxy,
    },` to devices.js

Maybe more code is needed to make it work properly. I assume for a light a sniffer is needed, right? Maybe I can help once mine arrived from china.

I am also using the availability function for my devices and fairly often the light outside is marked as offline.
When that happens in the log it shows: `3/25/2019, 7:49:52 AM - error: Failed to ping 0x7cb03eaa00ae8d43`
 I'm sure there is something wrong with the connection of that device even after repairing it several times. There are two other routers just 3 meters away from it with one wall inbetween. Assuming that is enough and other devices much further away and battery powered are fine, it must the Gardenpole that is the issue.

tb-killa, 2019-03-25:
What version of Z2M and coordinator firmware you use?
The ping is reworked on latest dev branch and seems to work better.

copystring, 2019-03-25:
Im using the latest version of the firmware (20190223) and latest version of zigbee2mqtt (1.2.1)
The CC2531 router has CC2531_router_2018_09.zip as firmware.

ekaats, 2019-03-25:
I am running into similar issues. My firmware and zigbee2mqtt versions are the same as those of copystring. So far only battery powered nodes sometimes fail. They cannot reconnect without being reset near the coordinator. Powered (tradfri) bulbs and switches reconnect just fine.

A possible issue could be the custom Home Assistant component [Circadian Lighting](https://community.home-assistant.io/t/circadian-lighting-custom-component/61246) which has a yet unsolved issue with Tradri bulbs. Apparently the lightbulbs cannot handle a change in brightness and temperature at the same time. Since Zigbee2mqtt 1.2.* this started breaking my zigbee devices.

Maybe these reports are related.

tunip, 2019-03-25:
About coverage... Please have in mind, that Zigbee channel 11 is not on WLAN 2,4GHz channel 11.
Zigbee Channel 11 (which is the default channel, if you not change it in your configuration.yaml) is very close to WLAN 2,4 GHz channel 1. My experience shows a clear influence if zigbee and wifi channel overlaps.

copystring, 2019-03-25:
Yes ekaats, this sound very familiar that my problem.

copystring, 2019-03-25:
That might be tunip. My wifi channel for 2.4ghz is 1. I guess I'll have to find out which channel Zigbee is running on.

tunip, 2019-03-25:
The default channel of Zigbee2mqtt is "11". So it overlaps in your case with WiFi channel "1". Changing Zigbee channel needs to pair your devices again. I changed my WiFi to a other less frequent channel then 1 (e.g. 6 or 11; used a WiFi scanner). If you have only a few Zigbee devices you could also change the Zigbee channel in you config and pair all devices again (do not use zigbee channel 26, as not all devices support it).

[image]

copystring, 2019-03-25:
I changed the wifi channel to 11. I'll report back if it works.

copystring, 2019-03-25:
Right.
At the moment I cannot switch the light outside at all and my lights in the kitchen react very slow, if at all.
For the light outside I do get this when trying to switch: `3/25/2019, 9:35:36 PM - error: Zigbee publish to device '0x7cb03eaa00ae8d43', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: AF data request fails, status code: 183. APS no ack.`

Just to be sure I have reflashed (to reset) the CC2531 router and re-paired it.

copystring, 2019-03-25:
From the lights in the kitchen (Müller Licht) I do get this a lot: `warn: No converter available for '404000/404005/404012' with cid 'haDiagnostic', type 'devChange' and data '{"cid":"haDiagnostic","data":{"averageMacRetryPerApsMessageSent":2,"lastMessageLqi":192,"lastMessageRssi":-52}}'
3/25/2019, 9:58:05 PM - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
`

mhelff, 2019-03-25:
Is the reporting feature turned on? If yes, this caused a lot of zigbee traffic from the Müller lights for me. Slow reacting devices are from my experience most times caused by too much traffic (too many commands or reports send in a short period of time)
Also take into consideration that turning lights in and off with a classic wall switch means powering a router on and off, which means a constant change in the mesh.
Additionally most of the 15 slots on your coordinator are "blocked" by the xiaomi sensors, but they don't route traffic.
Overall its hard to tell why exactly your network is so unreliable. Here are the things i did to stabilize my 48 devices network:
- Power all routers (for me mostly bulbs) permanently
- Add a Cc2530 router (with antenna) to increase range and routing capabilities
- Power on and pair all routers first to make sure they get a direct connection to the coordinator
- Then pair all other battery powered devices like the xiaomi sensors

After these steps i have a (99%) fast responding network with only rare hickups.
The only thing i still could not solve is a total "loss" of they communication with the whole zigbee network, which i can only solve by plugging off and on the cc2531 or rebooting the machine running zigbee2mqtt. This happens every 1-2 weeks, no clue why.

copystring, 2019-03-28:
Right. I have reset all my devices including the router and coordinator. Then went and re-paired all devices. The light seems to be able to switch all of the time now. Until I remove power from the lights in the kitchen which are, apart from one wall plug the closest routers to the outside lamp, the closest to the outside light.
Isn't the mesh sort of dynamic and finds a different route when one isn't available?
The way I understand it, it should use the router through the wall plug when the lights in the kitchen are not powered, right?

mhelff, 2019-03-28:
Yes, the mesh is dynamic, but its a kind of a black box how and when devices mesh. As seen on the xiaomi sensors some vendors seem also to ignore the meshing capabilities. 
Your wall plug in the kitchen (WeihnachtsbeleuchtungKueche?) also seems to have quite weak signal quality, that might be a reason why its not meshing immediately.

copystring, 2019-03-28:
WeihnachtsbeleuchtungKueche having a weak signal might make sense since it is the furthest away. only the light outside is further.
I did wait for it to mesh for 12 hours though. Surely at the point it should mesh?

mhelff, 2019-03-28:
I would expect a few minutes. Maybe your kitchenlight is really the only router able to get a proper connection to the outside gardenpole?

copystring, 2019-03-29:
Maybe but why? I assume when I can switch the wall plug on/off then it's part of the system and able to mesh. The wall plug is also closer to the garden pole than any other router. Or does the wall plug from Osram not support meshing?

mhelff, 2019-03-29:
Hard to tell, according to your network map the plug supports meshing. Maybe the garden poles are more picky about re-meshing. Do you see now any connections to the gardenpole on your network map?

copystring, 2019-03-29:
Yes. There is one connection to one of the 5 lights in the kitchen.

copystring, 2019-04-03:
I left the system alone for a few days. Looking at it now I noticed the CC2231 router is offline. This happened a lot before. It never comes back until I restart the coordinator.
The light outside works but is also marked as offline a lot.
In the past I was not always able to switch the light outside and I guess it did not work when it was "offline".
On the network map the router is however connected to other devices.
Looking at the log it shows a lot of:
`4/3/2019, 7:25:28 AM - error: Failed to ping 0x00124b0019369f63
4/3/2019, 7:25:38 AM - info: MQTT publish: topic 'zigbee2mqtt/ZigbeeRouter/availability', payload 'offline'`

0x00124b0019369f63 is the router.

mhelff, 2019-04-03:
On my network map, all routers (CC2530+all lights) are marked as "Offline", the battery powered devices are "Online". I suppose it is a bug. As long as your light is working now, dont give too much attention to the network map

copystring, 2019-04-03:
Yes. I know. If it works it's fine. However I assume the router isn't working since it doesn't respond to ping and is marked as offline in the log and map.

copystring, 2019-04-05:
So I restartet the raspberry pi this morning to bring back the CC2531 router and turned on the debug log. It was online for one ping and then offline for about 12 hours. However it only came back because server restarted at that time. (First log file)
https://gist.github.com/copystring/63e303257abeeadeb8dcb377e3518034

It was there for one ping and went back to being offline. (Second log file)
https://gist.github.com/copystring/74eac414ddd80fd5fd2f0c96fb414dcd

Surely there is something wrong. I'm not even talking about the network map. Just the log file.

Stefan300381, 2019-05-21:
any news here? facing similar issues but I have no clue how to debug log this. All I can say is that sometimes devices become "invisible" and don't report back their state.

closed 2019-07-27 by stale[bot]

copystring, 2021-01-01:
It has been a while but I actually have an update and solution. I have had 2 Osram Smart plugs and replaced them with Philips Hue plugs. Everything works now for weeks.

osnwt, 2023-01-21:
This thread is stale, but anyway... I started to see issues with network after I added few https://www.zigbee2mqtt.io/devices/TS011F_plug_1.html plugs. They are routers, and other devices started using them. But today again one of mains-powered switches (no neutral, not a router) showed as online and sent ZigBee messages, but did not respond. On the map it was dash line connected to two such routers at once, but neither link worked. After some time it finally decided to join just one, and network recovered. At the same time one of  such devices https://www.zigbee2mqtt.io/devices/BRT-100-TRV.html was shown as disconnected and also did not respond. It recovered as well.

All issues have come after addition of those routers. I don't need them as routers, just zigbee switches with monitoring, but can't disable routing feature. So all I can imagine is to add another one network just for them. Seems to be a stupid solution, maybe other exist?