Koenkk/zigbee2mqtt #1320: Color_Mode 2

Schnittcher, 2019-03-25:
Hello,

is color_mode 2 always for x and y?

greetings,
kai

Koenkk, 2019-03-26:
What do you mean with *color_mode 2*?

Schnittcher, 2019-03-27:
For example this is a log from a Ikea RGB LED:

Mär 24 17:40:43 PI3WiFi npm[498]:   zigbee2mqtt:info 2019-3-24 17:40:43 MQTT publish: topic 'zigbee2mqtt/LampeColor01', payload '{"state":"ON","linkquality":99,"brightness":25,"color_mode":2,"color":{"x":0.458,"y":0.41}}'

Is the color_mode 2 always for this color values?

greetings,
kai

Koenkk, 2019-03-27:
I think it depends on the device, further documentation on the color mode can be found here: http://www.zigbee.org/wp-content/uploads/2014/11/docs-07-5123-04-zigbee-cluster-library-specification.pdf (5.2.2.2.1.9 ColorMode Attribute)

closed 2019-06-02 by stale[bot]