Koenkk/zigbee2mqtt #1323: Failed to start

Btje1977, 2019-03-25:
I'm on the latest stick firmware and using hassio with the plugin. Coming from hassbian where it worked flawless i'm having nothing but trouble with hassio. First was the switching of the usb ports. I fixed it for my zwave stich and hope for a fast addition of connecting by ID in the plugin.

Now I cant get the plugin to run ok. The dev branche gives the same error. reboot of everything doesn't work. This is the logging:

[Info] Configuration file found: /share/zigbee2mqtt/configuration.yaml
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-03-25T21:48:11: PM2 log: Launching in no daemon mode
2019-03-25T21:48:11: PM2 log: App [npm:0] starting in -fork mode-
2019-03-25T21:48:11: PM2 log: App [npm:0] online
> zigbee2mqtt@1.2.1 start /zigbee2mqtt-1.2.1
> node index.js
  zigbee2mqtt:info 3/25/2019, 9:48:18 PM Logging to directory: '/share/zigbee2mqtt/log/2019-03-25.21-48-18'
  zigbee2mqtt:info 3/25/2019, 9:48:21 PM Starting zigbee2mqtt version 1.2.1 (commit #unknown)
  zigbee2mqtt:info 3/25/2019, 9:48:21 PM Starting zigbee-shepherd
  zigbee2mqtt:info 3/25/2019, 9:48:23 PM Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
  zigbee2mqtt:info 3/25/2019, 9:49:23 PM Starting zigbee-shepherd
  zigbee2mqtt:error 3/25/2019, 9:49:24 PM Error while starting zigbee-shepherd!
  zigbee2mqtt:error 3/25/2019, 9:49:24 PM Press the reset button on the stick (the one closest to the USB) and start again
  zigbee2mqtt:error 3/25/2019, 9:49:24 PM Failed to start
	{"message":"request timeout","stack":"Error: request timeout\n    at CcZnp.<anonymous> (/zigbee2mqtt-1.2.1/node_modules/cc-znp/lib/ccznp.js:261:22)\n    at Object.onceWrapper (events.js:273:13)\n    at CcZnp.emit (events.js:182:13)\n    at Timeout.<anonymous> (/zigbee2mqtt-1.2.1/node_modules/cc-znp/lib/ccznp.js:240:18)\n    at ontimeout (timers.js:436:11)\n    at tryOnTimeout (timers.js:300:5)\n    at listOnTimeout (timers.js:263:5)\n    at Timer.processTimers (timers.js:223:10)"}
  zigbee2mqtt:error 3/25/2019, 9:49:24 PM Exiting...
npm

pums12, 2019-03-25:
I was having exactly the same problem tonight. I took the stick out whilst re-flashing and let hassio boot without it.

On booting without the stick I noticed this entry in the system log:

_19-03-25 22:38:20 ERROR (SyncWorker_3) [hassio.docker] Can't start addon_a0d7b954_nodered: 500 Server Error: Internal Server Error ("linux runtime spec devices: error gathering device information while adding custom device "/dev/ttyACM0": no such file or directory")_

After reflashing the device is now working but could have been that Node Red was blocking access to the stick all along? Do you also have NodeRed running?

andreapede, 2019-04-02:
Same error. New installation.
node -v && npm -v
v10.15.3       6.4.1
Pasting the log

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.2.1~prestart: zigbee2mqtt@1.2.1
6 info lifecycle zigbee2mqtt@1.2.1~start: zigbee2mqtt@1.2.1
7 verbose lifecycle zigbee2mqtt@1.2.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.2.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
9 verbose lifecycle zigbee2mqtt@1.2.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.2.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.2.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.2.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.2.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.15.0-46-generic
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.2.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.2.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

nikolaykasyanov, 2019-04-09:
Something like this happened to me yesterday while I was messing with my router and other network devices (plugging/unplugging cables etc.) although I don't think it was related. The log looked exactly the same as the one from @Btje1977 (only the version was different as I was (and still am) on 1.3.0) after I had restarted zigbee2mqtt. The log of the session where it started happening is gone, unfortunately.

I had to reflash the stick to get it working again.

Koenkk, 2019-04-10:
@nikolaykasyanov I would recommend the max stability firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

nikolaykasyanov, 2019-04-11:
@Koenkk thanks, I'll give it a try if it happens again.

I'm curious though, what might be happening with the stick? Is it overwriting the firmware somehow?

Koenkk, 2019-04-12:
@nikolaykasyanov yes, it's writing to locations it shouldn't do and basically destroying the firmware.

timdonovanuk, 2019-04-23:
I'm running CC2531ZNP-Prod_20190223 and I've just had the same thing when I had to move and unplug a bunch of my routers. I'm running (well was running!) maybe 10 devices max at the moment. 

> I would recommend the max stability firmware
>  it's writing to locations it shouldn't do and basically destroying the firmware.

Bit confused, the default firmware has a bug? Or will that always happen once the limit is reached?

I'll try the max stability firmware when I get chance. How do I ensure that once the sticks device limit is reached that new devices use another router and not the coordinator directly when added (or will that happen automatically)?

Koenkk, 2019-04-23:
@timdonovanuk when the firmware runs out of memory, it will corrupt it self (this is separate from a device limit itself, it has nothing to do with it). 

The device will automatically pair with the router.

andreapede, 2019-04-24:
> Same error. New installation. No hassio.
> Only zigbee2mqtt installed
> 
> node -v v10.15.3
> npm -v 6.4.1
> 
> /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X10CEA9DEA311FFFF-if00 -> ../../ttyACM0
> 
> Pasting the log
> 
> 0 info it worked if it ends with ok
> 1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
> 2 info using npm@6.4.1
> 3 info using node@v10.15.3
> 4 verbose run-script [ 'prestart', 'start', 'poststart' ]
> 5 info lifecycle zigbee2mqtt@1.2.1prestart: zigbee2mqtt@1.2.1
> 6 info lifecycle zigbee2mqtt@1.2.1start: zigbee2mqtt@1.2.1
> 7 verbose lifecycle zigbee2mqtt@1.2.1start: unsafe-perm in lifecycle true
> 8 verbose lifecycle zigbee2mqtt@1.2.1start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin
> 9 verbose lifecycle zigbee2mqtt@1.2.1start: CWD: /opt/zigbee2mqtt
> 10 silly lifecycle zigbee2mqtt@1.2.1start: Args: [ '-c', 'node index.js' ]
> 11 silly lifecycle zigbee2mqtt@1.2.1start: Returned: code: 1 signal: null
> 12 info lifecycle zigbee2mqtt@1.2.1start: Failed to exec start script
> 13 verbose stack Error: zigbee2mqtt@1.2.1 start: `node index.js`
> 13 verbose stack Exit status 1
> 13 verbose stack at EventEmitter. (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
> 13 verbose stack at EventEmitter.emit (events.js:189:13)
> 13 verbose stack at ChildProcess. (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
> 13 verbose stack at ChildProcess.emit (events.js:189:13)
> 13 verbose stack at maybeClose (internal/child_process.js:970:16)
> 13 verbose stack at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
> 14 verbose pkgid zigbee2mqtt@1.2.1
> 15 verbose cwd /opt/zigbee2mqtt
> 16 verbose Linux 4.15.0-46-generic
> 17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
> 18 verbose node v10.15.3
> 19 verbose npm v6.4.1
> 20 error code ELIFECYCLE
> 21 error errno 1
> 22 error zigbee2mqtt@1.2.1 start: `node index.js`
> 22 error Exit status 1
> 23 error Failed at the zigbee2mqtt@1.2.1 start script.
> 23 error This is probably not a problem with npm. There is likely additional logging output above.
> 24 verbose exit [ 1, true ]

I found the origin of the problem at least in my case. I was using a cc2540 (Bluetooth) pen. The programmer works without problem and I realize just now. After start using the cc2531 everything works perfectly.

timdonovanuk, 2019-04-26:
Sorry, this is so confusing. So there is the DEFAULT firmware that eats away at itself. Nobody has acknowledge if this is a bug or intended behaviour and nobody has indicated if it's been fixed or in what version. 

(as a side note: you have firmware with a filename using "20190223" indicating it was released 62 days ago and github showing there was change 19 days ago. Then people tell you "switch to dev" but never indicate if that's dev zigbee2mqtt or dev firmware. Then you switch to dev branch of the firmware and come across a completely different folder structure to master. I truly greatly appreciate the work that's gone in to all of this, but its becoming *so* difficult to follow these issue threads!)

Koenkk, 2019-04-26:
@timdonovanuk 

- it is of course not intended behavior that the firmware get's corrupted
- this has been fixed in the max stability firmware
- currently I'm working on unifying all firmware's, all the different firmwares give too much confusion. (this is what you see in the dev branch).

olomarco, 2019-06-15:
I get this one:

Log
[Info] Configuration written to /share/zigbee2mqtt/configuration.yaml
2019-06-14T21:37:53: PM2 log: Launching in no daemon mode
2019-06-14T21:37:54: PM2 log: App [npm:0] starting in -fork mode-
2019-06-14T21:37:54: PM2 log: App [npm:0] online
> zigbee2mqtt@1.4.0 start /zigbee2mqtt-1.4.0
> node index.js
  zigbee2mqtt:info 6/14/2019, 9:37:58 PM Logging to directory: '/share/zigbee2mqtt/log/2019-06-14.21-37-58'
  zigbee2mqtt:info 6/14/2019, 9:37:59 PM Starting zigbee2mqtt version 1.4.0 (commit #unknown)
  zigbee2mqtt:info 6/14/2019, 9:37:59 PM Starting zigbee-shepherd
<--- Last few GCs --->
[40:0x2ee1ab0]   676416 ms: Scavenge 694.3 (720.6) -> 692.7 (721.1) MB, 190.0 / 0.0 ms  (average mu = 0.202, current mu = 0.050) allocation failure 
[40:0x2ee1ab0]   676706 ms: Scavenge 694.5 (721.1) -> 692.9 (721.6) MB, 58.6 / 0.0 ms  (average mu = 0.202, current mu = 0.050) allocation failure 
[40:0x2ee1ab0]   677224 ms: Scavenge 694.6 (721.6) -> 693.1 (726.1) MB, 101.2 / 0.0 ms  (average mu = 0.202, current mu = 0.050) allocation failure 
<--- JS stacktrace --->
FATAL ERROR: Ineffective mark-compacts near heap limit Allocation failed - JavaScript heap out of memory

and then starts again,do you guys know what it could be?
Thank you

closed 2019-08-22 by stale[bot]