Koenkk/zigbee2mqtt #1324: IKEA E1524 without battery information?

Maik-100, 2019-03-26:
In payload of IKEA E1524 are information about pressing the buttons.
Other sensors sends information about the battery state. In the MQTT log in the payload area, I am able to find infos for linkquality, but not for battery, why?

Koenkk, 2019-03-26:
This should be reported after +- 1 day of uptime

Maik-100, 2019-03-26:
What you mean with uptime?
My E1524 is peered for days per zigbee2mqtt to the MQTT-Brocker, but no battery state information.
Or is it importend that to the MQTT-Brocker is an application like Domotizz (with a plugin) is connected? The plugin gets support for E1524 today. So I had to update the plugin and to wait one or two days.

Koenkk, 2019-03-27:
I've just checked, this has been recently added and is not in the release version yet. In case you want battery information you should switch to the dev branch (edge for hassio).

closed 2019-03-27 by Koenkk

jlo88, 2019-04-06:
I'm running edge and I see the battery level in the zigbee2mqtt logs but not in home assistant. I just switched to edge because I like this feature. Should I wait for one day of uptime of zigbee2mqtt or home assistant?