Koenkk/zigbee2mqtt #1325: IKEA E1524 with no converter?

Maik-100, 2019-03-26:
IKEA E1524 is well supported. All five buttons give messages in MQTT log.

But if I push the big button in the middel for a long time, I got this log:

2019-3-26 08:46:05 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"action":"toggle","linkquality":0}'
2019-3-26 08:46:09 - warn: No converter available for 'E1524' with cid 'genLevelCtrl', type 'cmdMoveToLevelWithOnOff' and data '{"cid":"genLevelCtrl","data":{"level":254,"transtime":0}}'
2019-3-26 08:46:09 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
2019-3-26 08:46:09 - info: MQTT publish: topic 'zigbee2mqtt/0x000d6ffffe18e00a', payload '{"action":"arrow_right_click","linkquality":0}'

ninalovegood, 2019-03-26:
Hi,
I'm also currently trying to pair the ikea remote control.
Did you also have the issue that remote looses it's functionality (turning lights on/off, dimming,...) when you connected it to zigbee2mqtt? Is this the intended behaviour?!
he remote was either paired with the wireless receiver, able to control the lights but not visible in the zigbee2mqtt logs or the other way round...
I would have expected to get logs from clicking the buttons while the lights are controlled accordingly..
Any help is appreciated :)

Maik-100, 2019-03-26:
Hallo ninalovegood,

i am very new in the zigbee and  IKEA E1524.

But I think, that one sensor (IKEA E1524) is capable either to connect 
to one cental point (e.g. Raspberry with zigbee2mqtt or IKEA system with 
internet and cloud [wireless receiver]) or this sensor is peered with an 
actor (e. g. a lamp) (you can buy this bundel/set in the IKEA shop).

Koenkk, 2019-03-27:
Thanks, fixed in latest dev!

closed 2019-03-27 by Koenkk