Koenkk/zigbee2mqtt #1326: WXKG03LM multi click

ericsboro, 2019-03-26:
I just got new version on wall remote switch, but it looks like multiclick (like double, triple, etc.) not supported.
Are there plans to add this functionality?

many thanks in advance

Koenkk, 2019-03-26:
Could you publish your `log_level: debug` log (http://www.zigbee2mqtt.io/configuration/configuration.html) with clear annotations what zigbee message corresponds to what click event.

Zagitta, 2019-03-26:
As I noted here: https://github.com/Koenkk/zigbee-shepherd-converters/pull/211 the double click and long click seem to be a mistake in the documentation.
You'll have to implement it in your automation system to emulate multi-click unfortunately.

ericsboro, 2019-03-27:
understood. 
many thanks for info

closed 2019-03-27 by ericsboro