Koenkk/zigbee2mqtt #1328: Neighbors Hue Bulb appeared on my network map

roygbiv856, 2019-03-26:
I don't own any Hue bulbs. Today, while setting up a Xiaomi wireless double switch, I checked my network map and found a Hue bulb acting as a router. https://imgur.com/a/x9cZVXL Seems like it's definitely routing zigbee traffic.

I added the bulb to home assistant. I can add it as a light, but I cant control it. I haven't paired any new devices in weeks, so I am not sure how it joined the network. Anybody know how to proceed? I am thinking I can ban it from my network, but I'd like to prevent any future foreign devices from joining my network!

Koenkk, 2019-03-26:
Was `permit_join` set to true in `configuration.yaml`?

roygbiv856, 2019-03-26:
No, it is set to false.

sapkra, 2019-03-27:
Maybe your neighbor is using the same default zigbee network encryption key.
But I have to say that I don't have much knowledge how zigbee works so this is just an idea.

Koenkk, 2019-03-27:
This shouldn't happen anymore in the dev branch, probably fixed by #816

closed 2019-06-02 by stale[bot]