Koenkk/zigbee2mqtt #1330: Segmentation Fault when starting zigbee2mqtt

ninalovegood, 2019-03-26:
Hi,
I followed the instructions to install zigbee2mqtt on the raspi.
I only changed pi to openhabian in the config file and my credentials for the broker obviously.

I'm now getting the following error when I do the npm start in the zigbee2mqtt directory:

```
zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
node index.js

  zigbee2mqtt:info 2019-3-26 18:58:14 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-03-26.18-58-14'
  zigbee2mqtt:info 2019-3-26 18:58:15 Starting zigbee2mqtt version 1.2.1 (commit #4048cb8)
  zigbee2mqtt:info 2019-3-26 18:58:15 Starting zigbee-shepherd
Segmentation fault
npm ERR! code ELIFECYCLE
npm ERR! errno 139
npm ERR! zigbee2mqtt@1.2.1 start: `node index.js`
npm ERR! Exit status 139
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.2.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/openhabian/.npm/_logs/2019-03-26T17_58_15_529Z-debug.log
```
Any help is highly appreciated!

heiderich, 2019-03-26:
Maybe a duplicate of #1273 ?

ninalovegood, 2019-03-26:
Great, thanks for the hint!
A removal of the cloned folder /opt/zigbee2mqtt and restarting from a fresh clone did the job!

closed 2019-03-26 by ninalovegood