Koenkk/zigbee2mqtt #1331: 3D Printed case for CC2530 router from wiki?

hgelpke, 2019-03-27:
I see a lot of minimal cases but has anybody designed a case to function in place of the aliexpress project box in the wiki?

JLFN, 2019-03-27:
check the forum

JLFN, 2019-03-27:
check the forum

hgelpke, 2019-03-27:
I see like 4 posts in the forum.

Am I missing something?

jjrh, 2019-03-28:
Little confused what you're after - 
https://www.zigbee2mqtt.io/information/3d_cases.html 

What's this aliexpress project box?

hgelpke, 2019-03-28:
https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

jjrh, 2019-03-28:
Ah missed that. It's a pretty standard box.  Look on digikey/mouser or whatever your electronics distributor is if you don't want to go the aliexpress route.

hgelpke, 2019-03-28:
I was hoping somebody had designed a 3d printed version already laid out.

closed 2019-06-03 by stale[bot]