Koenkk/zigbee2mqtt #1333: How to reduce latency ?

Bastiencc, 2019-03-27:
Hi guys

My config : 
- zigbee2mqtt version 1.2.1 (commit #9a15e8c)
- Coordinator firmware version: '20190223'
- I'm using zgbee2mqtt/Mosquitto and HA for automations.

Sometimes when I press my Hue Dimmer Switch I'm experiencing bad latency (4/5 sec). From HA toggle card everything is very responsive.

Is there a way to reduce this latency ? Maybe there is a problem with my networkmap ?
https://imgur.com/a/rrpqIqx

There is the log (you can see datetimes): 
```
Mar 27 15:01:19 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:19 PM MQTT publish: topic 'zigbee2mqtt/cuisine_switch', payload '{"battery":100,"linkquality":68,"duration":0,"counter":1,"brightness":255,"action":"on-press"}'
Mar 27 15:01:27 raspberrypi npm[4485]:   zigbee2mqtt:error 3/27/2019, 3:01:27 PM Failed to ping 0x00158d0002d6f569
Mar 27 15:01:27 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:27 PM Zigbee publish to device '0x00158d0002d6f569', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 27 15:01:27 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:27 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":55,"brightness":255,"color_temp":338,"color":{"x":0.442,"y":0.41},"color_mode":1}'
Mar 27 15:01:27 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:27 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":55,"brightness":25,"color_temp":338,"color":{"x":0.442,"y":0.41},"color_mode":1}'
Mar 27 15:01:27 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:27 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":55,"brightness":25,"color_temp":338,"color":{"x":0.442,"y":0.41},"color_mode":1}'
```

And sometime there is no latency (using the same dimmer switch/bulb) : 
```
Mar 27 15:01:54 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:54 PM MQTT publish: topic 'zigbee2mqtt/cuisine_switch', payload '{"battery":100,"linkquality":68,"duration":0,"counter":1,"brightness":255,"action":"on-press"}'
Mar 27 15:01:54 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:54 PM Zigbee publish to device '0x00158d0002d6f569', genLevelCtrl - moveToLevelWithOnOff - {"level":255,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - null
Mar 27 15:01:55 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:55 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":55,"brightness":255,"color_temp":338,"color":{"x":0.442,"y":0.41},"color_mode":1}'
Mar 27 15:01:55 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:55 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":55,"brightness":30,"color_temp":338,"color":{"x":0.442,"y":0.41},"color_mode":1}'
Mar 27 15:01:55 raspberrypi npm[4485]:   zigbee2mqtt:info 3/27/2019, 3:01:55 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":55,"brightness":30,"color_temp":338,"color":{"x":0.442,"y":0.41},"color_mode":1}'
```

Koenkk, 2019-03-27:
Your linkquality seems to be low for some devices (1). Which bulb is the problematic one, `wc_plafond`?

Koenkk, 2019-03-27:
Also based on the first log, I see the latency is not coming from zigbee2mqtt. For the first log (where you have a delay) could you provide the `log_level: debug` log? (http://www.zigbee2mqtt.io/configuration/configuration.html)

Bastiencc, 2019-03-28:
Hi @Koenkk 

In the first log it's `cuisine_switch` , linkquality is 65 which is not so bad.
When I got latency on a bulb, the latency is also existing in others bulbs (more or less), it seems to impact all the network...
About linkquality, I think there is a bug in the networkmap picture, the linkquality of `wc_plafond` seems to be better (around 60) : 
```Mar 28 06:52:45 raspberrypi npm[1243]:   zigbee2mqtt:info 3/28/2019, 6:52:45 AM MQTT publish: topic 'zigbee2mqtt/wc_plafond', payload '{"state":"ON","linkquality":63,"brightness":7,"color_temp":338}'```

This morning everything works great, I will provide you a debug log as soon as possible.

thanks !

koen20, 2019-04-01:
Hello,
When I turn on a light I also have a delay of a few seconds. The problem started after updating from 1.1.1 to 1.2.0. I'm not using the latetst firmware (20190223).

Bastiencc, 2019-04-05:
Hi @Koenkk 

Sorry for the delay, here is the debug log :

```
Apr 05 12:59:37 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:37 PM Received zigbee message of type 'cmdHueNotification' with data '{"cid":"manuSpecificPhilips","data":{"button":1,"unknown1":3145728,"type":0,"unknown2":0,"time":0}}' of device 'RWL021' (0x0017880104af512e)
Apr 05 12:59:37 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:37 PM Received zigbee message of type 'cmdOn' with data '{"cid":"genOnOff","data":{}}' of device 'RWL021' (0x0017880104af512e)
Apr 05 12:59:37 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:37 PM Received zigbee message of type 'cmdHueNotification' with data '{"cid":"manuSpecificPhilips","data":{"button":1,"unknown1":3145728,"type":2,"unknown2":0,"time":1}}' of device 'RWL021' (0x0017880104af512e)
Apr 05 12:59:37 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:37 PM MQTT publish: topic 'zigbee2mqtt/cuisine_switch', payload '{"battery":89,"linkquality":63,"duration":0,"counter":1,"brightness":255,"action":"on-press"}'
Apr 05 12:59:37 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:37 PM Received MQTT message on 'zigbee2mqtt/cuisine_plafond/set' with data '{"state": "OFF"}'
Apr 05 12:59:50 raspberrypi npm[2428]:   zigbee2mqtt:error 4/5/2019, 12:59:50 PM Cannot get the Node Descriptor of the Device: 0x001788010408cd64 (Error: Timed out after 10000 ms)
Apr 05 12:59:50 raspberrypi npm[2428]:   zigbee2mqtt:error 4/5/2019, 12:59:50 PM Zigbee publish to device '0x00158d0002d6f569', lightingColorCtrl - moveToColorTemp - {"colortemp":338,"transtime":10} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: Timed out after 30000 ms
Apr 05 12:59:50 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:50 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: Timed out after 30000 ms","meta":{"entity":{"ID":"0x00158d0002d6f569","type":"device","friendlyName":"cuisine_plafond"},"message":"{\"state\": \"ON\", \"color_temp\": 338, \"brightness\": 255, \"transition\": 1}"}}'
Apr 05 12:59:50 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:50 PM Zigbee publish to device '0x00158d0002d6f569', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - null
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"genLevelCtrl","data":{"currentLevel":255}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":255}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:51 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":36,"brightness":255}'
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Received zigbee message of type 'devChange' with data '{"cid":"genLevelCtrl","data":{"currentLevel":255}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:51 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"ON","linkquality":36,"brightness":255}'
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:51 PM Zigbee publish to device '0x00158d0002d6f569', lightingColorCtrl - read - [{"attrId":7}] - {"manufSpec":0,"disDefaultRsp":0} - null
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"lightingColorCtrl","data":{"colorTemperature":338}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"zclVersion":1}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Successfully pinged 0x00158d0002d6f569
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:51 PM Zigbee publish to device '0x00158d0002d6f569', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:info 4/5/2019, 12:59:51 PM MQTT publish: topic 'zigbee2mqtt/cuisine_plafond', payload '{"state":"OFF","linkquality":36,"brightness":255}'
Apr 05 12:59:51 raspberrypi npm[2428]:   zigbee2mqtt:debug 4/5/2019, 12:59:51 PM Received zigbee message of type 'readRsp' with data '{"cid":"genOnOff","data":{"onOff":1,"onTime":0,"offWaitTime":0}}' of device 'lumi.light.aqcn02' (0x00158d0002d6f569)
Apr 05 12:59:53 raspberrypi npm[2428]:   zigbee2mqtt:error 4/5/2019, 12:59:53 PM Failed to ping 0x00158d0002d6f577
```

I have lot of issue with Philips Hue dimmer switch (latency, stop working, need to re-pair), I think someting is wrong with those remotes.

Koenkk, 2019-04-07:
The delay indeed comes from zigbee2mqtt, could you try the max stability firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

Bastiencc, 2019-04-07:
ok, I try this firmware and I'll came back to you asap. thanks @Koenkk

Bastiencc, 2019-04-19:
I tried the max stability firmware yesterday, and remove/repair all devices.  I thought there will be a problem with the limit of 5 devices (I have 11 devices including 3 routers) but it seems ok. Let's see !
I get back to you asap

Bastiencc, 2019-05-02:
I change my wifi channel to 11 (and zigbee still to channel 11), and now it seems to be better...

Koenkk, 2019-05-02:
@Bastiencc thanks, could you also try with this firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator/Z-Stack_Home_1.2/bin ?

Bastiencc, 2019-05-02:
If I experency latency again, I will test this version, thank you @Koenkk !

closed 2019-07-08 by stale[bot]