Koenkk/zigbee2mqtt #1334: Failed to find device

Bruceforce, 2019-03-27:
Hi,

today morning suddenly all my devices became unavailable. I updated my CC2531 to the latest "max stability" firmware to increase stability 😄. I had to re-pair the devices since they wouldn't automatically connect to the coordinator.

A few hours later I paired an Aqara switch and all devices are again unavailable.

The log says
```
zigbee2mqtt@1.2.1 start /app
node index.js
zigbee2mqtt:info 3/27/2019, 7:28:49 PM Logging to directory: '/app/data/log/2019-03-27.19-28-49'
zigbee2mqtt:error 3/27/2019, 7:40:04 PM Failed to find device with ieeAddr: '....'
...
```

for all devices. I'm using the latest dev docker image btw.

Koenkk, 2019-03-28:
Can you share your database.db file?

Bruceforce, 2019-03-28:
Yes. https://gist.github.com/Bruceforce/4833285cb6d20bf9a2041e20fc6654ca

Note: I have tried to re-pair the devices, but gave up on one of the Ikea outlets since it's hard to reach physically. 

So this one here is still failing
```
zigbee2mqtt       |   zigbee2mqtt:debug 3/28/2019, 5:08:20 PM Received MQTT message on 'zigbee2mqtt/tradfri_switch_01/set' with data 'OFF'
zigbee2mqtt       |   zigbee2mqtt:error 3/28/2019, 5:08:20 PM Failed to find device with ieeAddr: '0x000d6ffffedd8d32'
zigbee2mqtt       |   zigbee2mqtt:info 3/28/2019, 5:08:20 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"0x000d6ffffedd8d32"}'
```

Koenkk, 2019-03-28:
Devices are somehow missing from your database.db, e.g.`0x000d6ffffedd8d32` is not in the database.db, can you try to re-pair these devices.

Bruceforce, 2019-03-29:
@Koenkk I changed the channel yesterday to 25 - so I had to re-pair all devices. The same issue happened again after a few hours.

```
zigbee2mqtt       |   zigbee2mqtt:error 3/29/2019, 6:14:27 AM Failed to find device with ieeAddr: '0xd0cf5efffed828c3'
zigbee2mqtt       |   zigbee2mqtt:error 3/29/2019, 6:14:27 AM Failed to find device with ieeAddr: '0x90fd9ffffe32015e'
zigbee2mqtt       |   zigbee2mqtt:error 3/29/2019, 6:14:27 AM Failed to find device with ieeAddr: '0x000b57fffe30065f'
```

Here is my new database https://gist.github.com/Bruceforce/128b838186b7db7aa6eabc3c74a6a615
This time the failing devices are in there.

Koenkk, 2019-03-29:
Yes, but as you see they are marked as deleted by `{"$$deleted":true,"_id":"TzDJDMozsj0jRDlH"}`. Did you use the new ban functionality or remove any devices through http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove ?

Bruceforce, 2019-03-29:
@Koenkk This is really strange. I did indeed remove ONE device `0x000b57fffe15051e` but that isn't even listed as deleted. Maybe mosquitto is doing something strange?

Koenkk, 2019-03-29:
I think it's a bug in the dev branch, we need to figure out how to reproduce this.

Btje1977, 2019-03-29:
How did you remove them, I removed devices trough the HomeAssistant interface so not via mqtt message. I also deleted the latest dev branch and reinstalled everything. I also deleted everything in the share dir. I would think everything is clean then but when I want to add a device it doesn't recognize it.

Bruceforce, 2019-03-29:
I removed them via mqtt message (once per home assistant and in another try via a Windows client).

Koenkk, 2019-03-29:
@Btje1977 for this device, the pairing procedure has not been completed, you should restart zigbee2mqtt, remove all entries of `0x90fd9ffffe2a2de4 ` from `database.db` and pair it again.

Koenkk, 2019-03-29:
@Bruceforce can you provide the complete debug loggin? (`log_level: debug`) http://www.zigbee2mqtt.io/configuration/configuration.html

Btje1977, 2019-03-29:
> @Btje1977 for this device, the pairing procedure has not been completed, you should restart zigbee2mqtt, remove all entries of `0x90fd9ffffe2a2de4 ` from `database.db` and pair it again.

Could it be that HA puts things back in the database. I see 4 devices already installed even when I remove and reinstall zigbee2mqtt

Koenkk, 2019-03-29:
@Btje1977 it automatically add them if the devices rejoin.

Koenkk, 2019-03-29:
I've probably found the issue, please update to the latest dev branch and re-pair any uncontrollable devices.

Bruceforce, 2019-03-29:
@Koenkk I just collected logs with the old version. I'll upload it in case it would be useful ([Logs1.zip](https://github.com/Koenkk/zigbee2mqtt/files/3024684/Logs1.zip) - the last ones are the failing ones)

I just updated the docker image but now I can't pair any device at all. I tried re-plugging the stick, pressing the button on the stick and restarting zigbee2mqtt several times (see [Logs2.zip](https://github.com/Koenkk/zigbee2mqtt/files/3024683/Logs2.zip))

Koenkk, 2019-03-30:
Are the unavailable devices in the startup log?

```js
3/29/2019, 10:25:53 PM - info: Currently 7 devices are joined:
3/29/2019, 10:25:53 PM - info: stairs_light (0x000b57fffe30065f): LED1623G12 - IKEA TRADFRI LED bulb E27 1000 lumen, dimmable, opal white (Router)
3/29/2019, 10:25:53 PM - info: living_room_ceiling_mid_light (0x90fd9ffffe32015e): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
3/29/2019, 10:25:53 PM - info: living_room_ceiling_right_light (0xd0cf5efffe72bffc): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
3/29/2019, 10:25:53 PM - info: living_room_ceiling_left_light (0xd0cf5efffed828c3): LED1650R5 - IKEA TRADFRI LED bulb GU10 400 lumen, dimmable (Router)
3/29/2019, 10:25:53 PM - info: tradfri_switch_02 (0x000d6ffffe9c414a): E1603 - IKEA TRADFRI control outlet (Router)
3/29/2019, 10:25:53 PM - info: mija_switch_01 (0x00158d0002e27365): WXKG01LM - Xiaomi MiJia wireless switch (EndDevice)
3/29/2019, 10:25:53 PM - info: tradfri_dimmer_01 (0x000b57fffe15051e): ICTC-G-1 - IKEA TRADFRI wireless dimmer (EndDevice)
```

Bruceforce, 2019-03-30:
@Koenkk The devices seem to stay connected now, so the original issue seems to be solved. However since I re-paired everything the `availability_timeout` is not working anymore. My advanced settings looks like this:
```
advanced:
  panid: 6754
  channel: 25
  network_key: ...
  log_level: debug
  report: true
  availability_timeout: 60
```

Before the re-pairing all devices showed as `unavailable` in home assistant when they weren't connected to a power source. So when I cut the power of the tradfri bulbs they switched state to `unavailable` but now nothing happens and they always show the last state before I cut the power.

In the following log I started zigbee2mqtt with all 4 tradfri bulbs turned on. After 5 minutes I cut the power of all 4 bulbs and the state didn't switch to `unavailable`.

https://gist.github.com/Bruceforce/b3ca8571dc7b6033e3e896ad9ee3d07d

I noted from the logs that only `0x000d6ffffe9c414a` is being pinged regularly. Should I open a new issue for this?

Koenkk, 2019-04-01:
@Bruceforce can you share your `data/database.db` and `configuration.yaml`?

Bruceforce, 2019-04-01:
database: https://gist.github.com/Bruceforce/59451879ec7cfdc7e9454007da41515b

configuration: https://gist.github.com/Bruceforce/440fc653f2250e86f642cb90724f667f

Koenkk, 2019-04-04:
Can you set  `log_level: debug` and provide the logs again.

closed 2019-04-04 by Koenkk