Koenkk/zigbee2mqtt #1337: Crash and / or reboot every day

SargonofAssyria, 2019-03-28:
I am on dev branch. Using the latest router firmware (20190223)
Everything was working fine until commit zigbee2mqtt version 1.2.1 (commit #661f79c) from 15-03-2019
On 23-03 I upgraded to commit #008fda4. From that day on the system crashes and /or reboot every day. Until now I only have 11 devices in the zigbee network.
From that day I upgraded every day but the problem remains.
I have 2 problems.
1. The system hits a loop from which it does not recover.
`3/24/2019, 9:24:43 AM - error: Error while starting zigbee-shepherd!
3/24/2019, 9:24:46 AM - info: Logging to directory: '/var/log/zigbee'
3/24/2019, 9:24:47 AM - info: Starting zigbee2mqtt version 1.2.1 (commit #008fda4)
3/24/2019, 9:24:47 AM - info: Starting zigbee-shepherd
3/24/2019, 9:24:47 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
`
2. It gets into a loop of ping errors.
`3/28/2019, 8:25:44 AM - error: Failed to ping 0x00158d0002161ef0
3/28/2019, 8:25:45 AM - error: Failed to ping 0x00158d0002b7d689
3/28/2019, 8:26:45 AM - error: Failed to ping 0x00158d0002161ef0
3/28/2019, 8:26:45 AM - error: Failed to ping 0x00158d0002b7d689
`
Those are both routers. 1 is a xiaomi plug and 1 is a xiaomi lamp and sometimes a Osram powerplug has the same ping error.

Anybody any idea how this can be solved?

joseska, 2019-03-28:
same issue here or very similar. Failed to ping to 1 xiaomi plug and 1 xiaomi bulb and always crash after some hours or one or two days.....
https://pastebin.com/fEzqNge2
https://pastebin.com/BDYNSevt

I have a issue open here (https://github.com/Koenkk/zigbee2mqtt/issues/1294) but for now no resolution
maybe with and upgrade........

Koenkk, 2019-03-28:
can you try with the max stability firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

SargonofAssyria, 2019-03-28:
That seems to do the trick. Except for my Osram Powerplug it looks (fingers crossed) like everything is working fine. It gave this error:
`3/28/2019, 7:05:22 PM - warn: No converter available for 'AB3257001NJ' with cid 'haElectricalMeasurement', .....`  I am on (commit #7e6dc67)
 I had to pull my Router (CC2531) from power to get it working. The router was sending messages every few seconds insteed of every minute. Is there a new firmware for the router too?

SargonofAssyria, 2019-03-28:
Sorry to early. After some 30 minutes it reboots and hits the (takes 60 seconds) loop again.

Koenkk, 2019-03-28:
Can you provide the log of the crash?

SargonofAssyria, 2019-03-28:
This time it was an unwanted reboot and the log is deleted.
I will post the log if it happens again, I will set logging to debug.

SargonofAssyria, 2019-03-29:
This is what happened in the middle of the night.
`3/29/2019, 4:07:08 AM - info: MQTT publish: topic 'zigbee2mqtt/Powerplug', payload '{"state":"ON","linkquality":81,"device":{"ieeeAddr":"0x84182600000f0057","friendlyName":"Powerplug","type":"Router","nwkAddr":1839,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","hwVersion":1,"swBuildId":"V1.04.12","status":"offline"}}'
3/29/2019, 4:08:13 AM - error: Failed to ping 0x00158d0002161ef0
3/29/2019, 4:08:48 AM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":81,"device":{"ieeeAddr":"0x00158d0002161ef0","friendlyName":"UPS","type":"Router","nwkAddr":11686,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug","hwVersion":"unknown","swBuildId":"unknown","status":"online"},"power":0,"voltage":224.4,"consumption":26.21,"temperature":29}'
3/29/2019, 4:12:03 AM - info: MQTT publish: topic 'zigbee2mqtt/Powerplug', payload '{"state":"ON","linkquality":81,"device":{"ieeeAddr":"0x84182600000f0057","friendlyName":"Powerplug","type":"Router","nwkAddr":1839,"manufId":48042,"manufName":"OSRAM","powerSource":"Mains (single phase)","modelId":"Plug 01","hwVersion":1,"swBuildId":"V1.04.12","status":"offline"}}'
3/29/2019, 4:12:26 AM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAddr":"0x00158d0002b7d689","friendlyName":"Overlooplamp","type":"Router","nwkAddr":21178,"manufId":4447,"manufName":"LUMI","powerSource":"DC Source","modelId":"lumi.light.aqcn02","hwVersion":1,"swBuildId":"unknown","status":"online"},"linkquality":70,"brightness":127,"color_temp":250,"color_mode":1,"color":{"x":0.383,"y":0.382}}'
3/29/2019, 4:15:02 AM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":84,"device":{"ieeeAddr":"0x00158d0002161ef0","friendlyName":"UPS","type":"Router","nwkAddr":11686,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.plug","hwVersion":"unknown","swBuildId":"unknown","status":"online"},"power":0,"voltage":224.4,"consumption":26.21,"temperature":29}'
3/29/2019, 4:15:59 AM - info: Logging to directory: '/var/log/zigbee'
3/29/2019, 4:16:07 AM - info: Starting zigbee2mqtt version 1.2.1 (commit #7e6dc67)
3/29/2019, 4:16:07 AM - info: Starting zigbee-shepherd
3/29/2019, 4:16:07 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
3/29/2019, 4:17:07 AM - info: Starting zigbee-shepherd
3/29/2019, 4:17:07 AM - error: Error while starting zigbee-shepherd!
`
It did not recover from this error, and goes on until now.

Koenkk, 2019-03-29:
Why did zigbee2mqtt restart in the first place?

SargonofAssyria, 2019-03-29:
Let me rephrase the title: I did mean a reboot of the zigbee2mqtt service and not of the system.
Restart might be a better word.
I have no idea why zigbee2mqtt service did restart, but it does after some (random) time.
There is no logic to find for the why it happened in the middle of the night when I was at sleep.

Koenkk, 2019-03-29:
Could you provide the log of the session before the restart? (should be in the `data/log` directory).

SargonofAssyria, 2019-03-29:
There is nothing special, zigbee2mqtt service worked fine until the restartbut here it is:
`3/28/2019, 7:19:16 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAd$
3/28/2019, 7:19:16 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAd$
3/28/2019, 7:19:17 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAd$
3/28/2019, 7:19:20 PM - info: Zigbee publish to device '0x00158d0002b7d689', genOnOff - on - {} - {"manufSpec":0,"disDe$
3/28/2019, 7:19:20 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"ON","device":{"ieeeAdd$
3/28/2019, 7:19:20 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"ON","device":{"ieeeAdd$
3/28/2019, 7:19:20 PM - info: Zigbee publish to device '0x00158d0002b7d689', genLevelCtrl - read - [{"attrId":0}] - {"m$
3/28/2019, 7:19:20 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"ON","device":{"ieeeAdd$
3/28/2019, 7:19:20 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"ON","device":{"ieeeAdd$
3/28/2019, 7:20:28 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":false,"linkquality":89$
3/28/2019, 7:21:59 PM - info: MQTT publish: topic 'zigbee2mqtt/Powerplug', payload '{"state":"ON","linkquality":68,"dev$
3/28/2019, 7:23:03 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":false,"linkquality":84$
3/28/2019, 7:23:27 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":68,"device":$
3/28/2019, 7:23:43 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":true,"linkquality":92,$
3/28/2019, 7:23:50 PM - info: Zigbee publish to device '0x00158d0002161ef0', genOnOff - on - {} - {"manufSpec":0,"disDe$
3/28/2019, 7:23:50 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"ON","linkquality":68,"device":{$
3/28/2019, 7:23:50 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"ON","linkquality":68,"device":{$
3/28/2019, 7:23:50 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"ON","linkquality":68,"device":{$
3/28/2019, 7:23:53 PM - info: Zigbee publish to device '0x00158d0002161ef0', genOnOff - off - {} - {"manufSpec":0,"disD$
3/28/2019, 7:23:53 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":68,"device":$
3/28/2019, 7:23:53 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":68,"device":$
3/28/2019, 7:23:54 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":68,"device":$
3/28/2019, 7:24:59 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":true,"linkquality":89,$
3/28/2019, 7:25:03 PM - info: MQTT publish: topic 'zigbee2mqtt/Powerplug', payload '{"state":"OFF","linkquality":55,"de$
3/28/2019, 7:25:19 PM - info: MQTT publish: topic 'zigbee2mqtt/Powerplug', payload '{"state":"ON","linkquality":44,"dev$
3/28/2019, 7:25:22 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"ON","device":{"ieeeAdd$
3/28/2019, 7:26:29 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":false,"linkquality":89$
3/28/2019, 7:28:47 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":70,"device":$
3/28/2019, 7:31:24 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":true,"linkquality":92,$
3/28/2019, 7:32:54 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":false,"linkquality":92$
3/28/2019, 7:33:01 PM - info: MQTT publish: topic 'zigbee2mqtt/SlaapkamerPIR', payload '{"illuminance":3,"linkquality":$
3/28/2019, 7:33:42 PM - info: MQTT publish: topic 'zigbee2mqtt/DoubleSwitch', payload '{"battery":100,"voltage":3005,"l$
3/28/2019, 7:34:57 PM - info: MQTT publish: topic 'zigbee2mqtt/UPS', payload '{"state":"OFF","linkquality":70,"device":$
3/28/2019, 7:35:11 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"ON","device":{"ieeeAdd$
3/28/2019, 7:35:11 PM - info: Zigbee publish to device '0x00158d0002b7d689', genOnOff - off - {} - {"manufSpec":0,"disD$
3/28/2019, 7:35:11 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAd$
3/28/2019, 7:35:11 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAd$
3/28/2019, 7:35:12 PM - info: MQTT publish: topic 'zigbee2mqtt/Overlooplamp', payload '{"state":"OFF","device":{"ieeeAd$
3/28/2019, 7:35:14 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":true,"linkquality":92,$
3/28/2019, 7:36:44 PM - info: MQTT publish: topic 'zigbee2mqtt/ZolderPIR', payload '{"occupancy":false,"linkquality":92$
3/28/2019, 7:37:48 PM - info: Logging to directory: '/var/log/zigbee'
3/28/2019, 7:37:49 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #7e6dc67)
3/28/2019, 7:37:49 PM - info: Starting zigbee-shepherd
3/28/2019, 7:37:50 PM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
3/28/2019, 7:38:50 PM - info: Starting zigbee-shepherd
3/28/2019, 7:38:52 PM - error: Error while starting zigbee-shepherd!
3/28/2019, 7:38:55 PM - info: Logging to directory: '/var/log/zigbee'
[... cut ...]
I am using the domoticz zigbee2mqtt plugin version 0.0.18
In the domoticz.log after the the restart there is only 1 line logged every 10 sec.
Unfortunately no hint which device.
`2019-03-29 06:24:11.586  (Zigbee) MqttClient::ping
2019-03-29 06:24:21.626  (Zigbee) MqttClient::ping
2019-03-29 06:24:31.600  (Zigbee) MqttClient::ping
2019-03-29 06:24:41.622  (Zigbee) MqttClient::ping
2019-03-29 06:24:51.605  (Zigbee) MqttClient::ping
2019-03-29 06:25:01.592  (Zigbee) MqttClient::ping
2019-03-29 06:25:11.643  (Zigbee) MqttClient::ping
`

Koenkk, 2019-03-29:
This is the log after the restart, do you have the one that caused the restart it self? (one log before this one).

SargonofAssyria, 2019-03-30:
No this is the log of the restart.
Just out of the blue (debug on) there is a 'restart' starting with:
`3/30/2019, 9:41:58 AM - info: Logging to directory: '/var/log/zigbee'
3/30/2019, 9:41:59 AM - info: Starting zigbee2mqtt version 1.2.1 (commit #7e6dc67)
3/30/2019, 9:41:59 AM - info: Starting zigbee-shepherd
3/30/2019, 9:41:59 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
`
So essentially this is not a compleet (re)start of the service.
This part of the logging starts out of the blue after normal operation, and the keeps on going for ever, until I stop the service manually. Then normal (re) start is only done manually by pulling the plug from the USB and start zigbee2mqtt manually.
The manual restart show normal start. Which looks like.
`3/28/2019, 7:03:42 PM - info: Logging to directory: '/var/log/zigbee'
3/28/2019, 7:03:43 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #7e6dc67)
3/28/2019, 7:03:43 PM - info: Starting zigbee-shepherd
3/28/2019, 7:03:44 PM - info: zigbee-shepherd started
3/28/2019, 7:03:44 PM - info: Coordinator firmware version: '20190315'
3/28/2019, 7:03:44 PM - info: Currently 11 devices are joined:
3/28/2019, 7:03:44 PM - info: Router (0x00124b0014d9de22): CC2530.ROUTER - Custom devices (DiY) `

I found this (meaningfull?) part in the syslog
`Mar 30 09:47:15 domoticz systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 30 09:47:15 domoticz systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Mar 30 09:47:15 domoticz systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 30 09:47:15 domoticz systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mar 30 09:47:15 domoticz systemd[1]: Stopped zigbee2mqtt.
Mar 30 09:47:15 domoticz systemd[1]: Started zigbee2mqtt.
Mar 30 09:48:19 domoticz systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 30 09:48:19 domoticz systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Mar 30 09:48:19 domoticz systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 30 09:48:19 domoticz systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mar 30 09:48:19 domoticz systemd[1]: Stopped zigbee2mqtt.
`

SargonofAssyria, 2019-03-30:
What also happened is that my python plugin created all my zigbee devices again, so they are all double now with one small differance, and that is the Unit number as shown below. The rest is exact duplicate.
(Except for the name of cource, which I manually altered.)
` | 846 | Zigbee | 0x00158d0002b7d689_light | 2 | Zigbee - Overlooplamp | Color Switch | WW | Off 
  | 819 | Zigbee | 0x00158d0002b7d689_light | 46 | Overloop lamp | Color Switch | WW | Off
`

SargonofAssyria, 2019-03-30:
This is how the zigbee2mqtt normally starts
`3/30/2019, 10:12:20 AM - info: Logging to directory: '/var/log/zigbee'
3/30/2019, 10:12:21 AM - info: Starting zigbee2mqtt version 1.2.1 (commit #91bc56a)
3/30/2019, 10:12:21 AM - info: Starting zigbee-shepherd
3/30/2019, 10:12:23 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
3/30/2019, 10:13:23 AM - info: Starting zigbee-shepherd
3/30/2019, 10:13:25 AM - info: zigbee-shepherd started
3/30/2019, 10:13:25 AM - info: Coordinator firmware version: '20190315'
3/30/2019, 10:13:25 AM - info: Currently 11 devices are joined:
3/30/2019, 10:13:25 AM - info: Router (0x00124b0014d9de22): CC2530.ROUTER - Custom devices (DiY) .... etc ...
3/30/2019, 10:13:25 AM - info: Zigbee: disabling joining new devices.
3/30/2019, 10:13:25 AM - info: Connecting to MQTT server at mqtt://localhost
3/30/2019, 10:13:25 AM - info: zigbee-shepherd ready
3/30/2019, 10:13:25 AM - info: Connected to MQTT server
`

Koenkk, 2019-03-30:
So, if I understand you correctly, when starting it manually everything works as expected, in the service it doesn't?

SargonofAssyria, 2019-03-30:
I always start with sudo systemctl start zigbee2mqtt.service because I stop first, pull the coordinator from USB and start again. Then it works again, that is what I mean with manually.
Now I am on
`3/30/2019, 4:18:48 PM - info: Logging to directory: '/var/log/zigbee'
3/30/2019, 4:18:49 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #91bc56a)
3/30/2019, 4:18:49 PM - info: Starting zigbee-shepherd
3/30/2019, 4:18:51 PM - info: zigbee-shepherd started
3/30/2019, 4:18:51 PM - info: Coordinator firmware version: '20190315'
3/30/2019, 4:18:51 PM - info: Currently 11 devices are joined:
`
Let hope for the best....

SargonofAssyria, 2019-03-31:
Unfortunately NOT solved.
Zigbee2mqtt logging
`3/31/2019, 7:32:00 AM - info: MQTT publish: topic 'zigbee2mqtt/Router', payload '{"state":false,"linkquality":23,"devic$
3/31/2019, 7:33:11 AM - info: Logging to directory: '/var/log/zigbee'
3/31/2019, 7:33:20 AM - info: Starting zigbee2mqtt version 1.2.1 (commit #91bc56a)
3/31/2019, 7:33:20 AM - info: Starting zigbee-shepherd
3/31/2019, 7:33:20 AM - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
3/31/2019, 7:34:20 AM - info: Starting zigbee-shepherd
3/31/2019, 7:34:20 AM - error: Error while starting zigbee-shepherd!
`
Syslog logging
`Mar 31 07:32:10 domoticz kernel: [425271.192857] usb 1-1.2.1-port3: disabled by hub (EMI?), re-enabling...
Mar 31 07:32:10 domoticz kernel: [425271.193244] usb 1-1.2.1.3: USB disconnect, device number 27
Mar 31 07:32:10 domoticz kernel: [425271.193799] cdc_acm 1-1.2.1.3:1.0: failed to set dtr/rts
Mar 31 07:32:11 domoticz kernel: [425271.489975] usb 1-1.2.1.3: new full-speed USB device number 28 using dwc_otg
Mar 31 07:32:11 domoticz kernel: [425271.648661] usb 1-1.2.1.3: New USB device found, idVendor=0451, idProduct=16a8
Mar 31 07:32:11 domoticz kernel: [425271.648678] usb 1-1.2.1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 31 07:32:11 domoticz kernel: [425271.648687] usb 1-1.2.1.3: Product: TI CC2531 USB CDC
Mar 31 07:32:11 domoticz kernel: [425271.648696] usb 1-1.2.1.3: Manufacturer: Texas Instruments
Mar 31 07:32:11 domoticz kernel: [425271.648704] usb 1-1.2.1.3: SerialNumber: __0X00124B0012023155
Mar 31 07:32:11 domoticz kernel: [425271.651478] cdc_acm 1-1.2.1.3:1.0: ttyACM2: USB ACM device
Mar 31 07:32:17 domoticz mtp-probe: checking bus 1, device 28: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2/1$
Mar 31 07:32:17 domoticz mtp-probe: bus: 1, device: 28 was not an MTP device
Mar 31 07:32:54 domoticz systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Mar 31 07:32:54 domoticz systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Mar 31 07:32:54 domoticz systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Mar 31 07:32:54 domoticz systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Mar 31 07:32:54 domoticz systemd[1]: Stopped zigbee2mqtt.
`
This now shows show USB hub problem, I will investigate further.

Koenkk, 2019-03-31:
@SargonofAssyria you should use the serial id instead: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#1-determine-location-of-cc2531-usb-sniffer-and-checking-user-permissions

SargonofAssyria, 2019-03-31:
One done that. But still this was a one time only (so far) that EMI message pops up.
If have take some action to prevent EMI, although that was never an issue.
I have a 1 mtr cable running from the hub to the coordinator (as adviced), and that cable had already some ferrite coil attached.

SargonofAssyria, 2019-04-01:
Knock on wood! Until know (more then 24 hours) everything is working fine. Now connected directly on the USB hub. I had it working for more then 6 months with a extra USB cable of 1 mtr. and had no problem until approx. 1 week ago.

closed 2019-04-01 by SargonofAssyria