Koenkk/zigbee2mqtt #1339: Trouble with QBKG04LM

mayrp2001, 2019-03-28:
Hi,

I have had my QBKG04LM working fine for a few months now, but from this morning I can't seem to get it working. On the log file for zigbee2mqtt I get the following. 

` zigbee2mqtt:error 3/28/2019, 5:47:59 PM Zigbee publish to device '0x00158d0002ec227d', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 2 failed with error Error: AF data request fails, status code: 240. MAC transaction expired.
  zigbee2mqtt:info 3/28/2019, 5:47:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: AF data request fails, status code: 240. MAC transaction expired.","meta":{"entity":{"ID":"0x00158d0002ec227d","type":"device","friendlyName":"Living_Room_Light_Main"},"message":"ON"}}'
  zigbee2mqtt:info 3/28/2019, 5:48:28 PM MQTT publish: topic 'zigbee2mqtt/Temperature_Sensor', payload '{"temperature":21.09,"linkquality":78,"humidity":41.67,"battery":100,"voltage":3025}'
  zigbee2mqtt:error 3/28/2019, 5:48:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:49:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:50:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:51:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:52:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:53:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:54:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:55:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:56:53 PM Failed to ping 0x00158d0002ec227d
  zigbee2mqtt:error 3/28/2019, 5:57:53 PM Failed to ping 0x00158d0002ec227d`

Any help would be great.

Thanks

Ryan

Koenkk, 2019-03-29:
Can you try with the latest dev branch? This should improve the pining issues you are having.

mayrp2001, 2019-03-29:
Sorry, I'm pretty new to this. How do I go about this?

Koenkk, 2019-03-29:
http://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html, let's continue in #943

closed 2019-03-29 by Koenkk