Koenkk/zigbee2mqtt #1340: Tradfri Signal repeater

MeisterQ, 2019-03-28:
Hello,

i just addet my first Tradfri Signal-Repeater (not the Plug) to my CC2531.

Firmware is something CC2531ZNP-Prod_20190109.hex (I bought this stick with this firmware from ebay).

I also updated zigbee2mqtt yesterday evening.

So now i got this messages:

https://pastebin.com/3SmqwUjq

Seems like the repeater is connected, but there are some sorts of errors in the log after the pairing.

Well. I put the repeater somewhere between the stick and a place where i do not get any signal from the stick. Waited afew minutes, tried to get some contact. No reponse. 

Well then i tried to pair my Mijia DWS close to the repeater and got some weird errors. 

I pressed the button for >5s till the led is flashing 3 times. Then i start pressing the button shortly each second.

Then the LED is flashing sometimes very short, sometimes its not flashing, and sometimes the LED stays on for about >1second.

After this behaiviour i cannot get any payload from the sensor anymore nor i can pair the sensor anymore to the stick untill i disconnect it from the USB-Port and connect it again, and reboot the Raspberry.

Anyone got some idea?

djchen, 2019-03-29:
It looks like the repeater is paired but Zigbee2mqtt doesn't support the "haDiagnostic" message coming from that device.

My guess is a "fz.ignore_diagnostic_change" needs to be added [here](https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/devices.js#L706) so Zigbee2mqtt ignores these diagnostic messages.

MeisterQ, 2019-03-29:
> It looks like the repeater is paired but Zigbee2mqtt doesn't support the "haDiagnostic" message coming from that device.
>
> My guess is a "fz.ignore_diagnostic_change" needs to be added [...]

So?

What about the weird behaivor if i try to pair devices? 

I figured out, that my Sensors does not find their way across the repeater to the stick,

I need to unplug and plug, and reboot and remove devices and pair them close to the repeater to get any payloads from it...

Also then, the sensors only works with the repeater and not with the stick anymore.

djchen, 2019-03-29:
> So?

You could try making that modification to devices.js and test it locally to see if it removes those messages.

> What about the weird behaivor if i try to pair devices?

I've found its much better to pair devices right next (literally like 1 inch away) to the coordinator. Try that.

After pairing, it shouldn't matter if the sensors are connected to the repeater (router) or the coordinator (stick), it should still work.

MeisterQ, 2019-03-30:
> I've found its much better to pair devices right next (literally like 1 inch away) to the coordinator. Try that.
>
> After pairing, it shouldn't matter if the sensors are connected to the repeater [...]

Well. I tried, but after i unplugged my Repeater, i couldnt get any signal from my sensor anymore. Even not if it was in range to my Stick.

After i plugged in i got signals again, but only in range to the repeater. Thats not a "Mesh" behaivor.

Are there still any problems with the Tradfri repeater? I orderd a Osram Plug to see if i got the same behaivor with this device.

closed 2019-06-05 by stale[bot]