Koenkk/zigbee2mqtt #1341: aqara cube paired but getting no converter available

csloz, 2019-03-29:
Having issues getting the aqara cube to pass on messages. 
Successfully paired the device with the cc2531 (flashed with latest firmware)
Detected as a hex id, have edited configuration.yaml to map it to the right device in devices.js (lumi.sensor_cube)

devices:  
  '0x00158d000275dc3c':
    friendly_name: 'lumi.sensor_cube'
    retain: true

Logs giving me errors about no converter available.
If I check the converters I can see that it should be supported.
i.e. in devices.js it uses the following converters for lumi.sensor_cube

fromZigbee: [
            fz.xiaomi_battery_3v, fz.MFKZQ01LM_action_multistate, fz.MFKZQ01LM_action_analog,
            fz.ignore_analog_change, fz.ignore_multistate_change, fz.ignore_basic_change,
        ],

In 
https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js
I can see that  fz.MFKZQ01LM_action_multistate should match for this - 

 zigbee2mqtt:debug 3/27/2019, 4:01:40 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)

...but it isn't.

Any suggestions?

Logs - 

3/27/2019, 3:46:58 PM - info: Starting zigbee2mqtt version 1.2.1 (commit #e5ca977)                                                                                  3/27/2019, 3:46:58 PM - info: Starting zigbee-shepherd
3/27/2019, 3:46:58 PM - info: zigbee-shepherd started                                                                                                               3/27/2019, 3:46:58 PM - info: Coordinator firmware version: '20190223'                                                                                              3/27/2019, 3:46:58 PM - info: Currently 1 devices are joined:
3/27/2019, 3:46:58 PM - info: 0x00158d000275dc3c (0x00158d000275dc3c): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)                                      3/27/2019, 3:46:58 PM - warn: `permit_join` set to  `true` in configuration.yaml.                                                                                   3/27/2019, 3:46:58 PM - warn: Allowing new devices to join.
3/27/2019, 3:46:58 PM - warn: Set `permit_join` to `false` once you joined all devices.                                                                             3/27/2019, 3:46:58 PM - info: Zigbee: allowing new devices to join.                                                                                                 3/27/2019, 3:46:58 PM - info: Connecting to MQTT server at mqtt://localhost
3/27/2019, 3:46:58 PM - info: zigbee-shepherd ready                                                                                                                 3/27/2019, 3:46:58 PM - info: Connected to MQTT server                                                                                                              3/27/2019, 3:46:58 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
3/27/2019, 3:46:59 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'                                    3/27/2019, 3:47:24 PM - warn: No converter available for 'MFKZQ01LM' with cid 'genAnalogInput', type 'attReport' and data '{"cid":"genAnalogInput","data":{"65285":500,"presentValue":11.470000267028809}}'
...

zigbee2mqtt:info 3/27/2019, 4:01:11 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"debug","permit_join":true}'
  zigbee2mqtt:debug 3/27/2019, 4:01:40 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":2}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:warn 3/27/2019, 4:01:40 PM No converter available for 'MFKZQ01LM' with cid 'genMultistateInput', type 'attReport' and data '{"cid":"genMultistateInput","data":{"presentValue":2}}'
  zigbee2mqtt:warn 3/27/2019, 4:01:40 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 3/27/2019, 4:01:41 PM Received zigbee message of type 'attReport' with data '{"cid":"genMultistateInput","data":{"presentValue":133}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:warn 3/27/2019, 4:01:41 PM No converter available for 'MFKZQ01LM' with cid 'genMultistateInput', type 'attReport' and data '{"cid":"genMultistateInput","data":{"presentValue":133}}'
  zigbee2mqtt:warn 3/27/2019, 4:01:41 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 3/27/2019, 4:01:41 PM Received zigbee message of type 'devChange' with data '{"cid":"genMultistateInput","data":{"presentValue":133}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:debug 3/27/2019, 4:01:58 PM Saving state to file /app/data/state.json
  zigbee2mqtt:debug 3/27/2019, 4:02:10 PM Received zigbee message of type 'attReport' with data '{"cid":"genAnalogInput","data":{"65285":500,"presentValue":15.029999732971191}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:warn 3/27/2019, 4:02:10 PM No converter available for 'MFKZQ01LM' with cid 'genAnalogInput', type 'attReport' and data '{"cid":"genAnalogInput","data":{"65285":500,"presentValue":15.029999732971191}}'
  zigbee2mqtt:warn 3/27/2019, 4:02:10 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
  zigbee2mqtt:debug 3/27/2019, 4:02:10 PM Received zigbee message of type 'devChange' with data '{"cid":"genAnalogInput","data":{"presentValue":15.029999732971191}}' of device 'lumi.sensor_cube' (0x00158d000275dc3c)
  zigbee2mqtt:debug 3/27/2019, 4:06:58 PM Saving state to file /app/data/state.json

8xiom, 2019-03-29:
The Aqara cube is supported "out of the box" which worked fine for me. After successful pairing the device (which indeed took me several attempts) everything worked fine (using node red with zigbe2mqtt and MQTT), no need for adjustments. A user-friendly name can be assigned as usual via MQTT directly. Are you sure that the detection and pairing process was really successful?

Koenkk, 2019-03-29:
This should indeed be covered by the `MFKZQ01LM_action_multistate` converter (https://github.com/Koenkk/zigbee-shepherd-converters/blob/master/converters/fromZigbee.js#L440)

How are you running zigbee2mqtt? Did you made any modifications to `devices.js`?

csloz, 2019-03-29:
running via docker.
mapped 2 volumes

 volumes:
   - /opt/zigbee2mqtt/data:/app/data
      - /opt/zigbee2mqtt/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters

git pulled the latest clone https://github.com/Koenkk/zigbee-shepherd-converters.git in /opt/zigbee2mqtt/data

If i check the folder in docker, its definitely mapped to that folder.
I've tried removing the docker file, and recreating, same problem.

Device is definitely joined

grep  joined * -R
2019-03-28.12-44-55/log.txt:3/28/2019, 12:44:56 PM - info: Currently 1 devices are joined:
2019-03-28.12-44-55/log.txt-3/28/2019, 12:44:56 PM - info: lumi.sensor_cube (0x00158d000275dc3c): MFKZQ01LM - Xiaomi Mi/Aqara smart home cube (EndDevice)
2019-03-28.12-44-55/log.txt-3/28/2019, 12:44:56 PM - warn: `permit_join` set to  `true` in configuration.yaml.
2019-03-28.12-44-55/log.txt-3/28/2019, 12:44:56 PM - warn: Allowing new devices to join.

grep  MFKZQ01LM * -R
2019-03-28.12-44-55/log.txt:3/29/2019, 8:47:07 AM - warn: No converter available for 'MFKZQ01LM' with cid 'genMultistateInput', type 'attReport' and data '{"cid":"genMultistateInput","data":{"presentValue":2}}'
2019-03-28.12-44-55/log.txt:3/29/2019, 9:20:52 AM - warn: No converter available for 'MFKZQ01LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"1":2985,"3":21,"4":5032,"5":218,"6":[0,2],"10":0,"151":0,"152":500,"153":451,"154":4}}}'
2019-03-28.12-44-55/log.txt:3/29/2019, 10:19:04 AM - warn: No converter available for 'MFKZQ01LM' with cid 'genBasic', type 'attReport' and data '{"cid":"genBasic","data":{"65281":{"1":2985,"3":21,"4":5032,"5":218,"6":[0,1],"10":0,"151":0,"152":527,"153":478,"154":4}}}'

Koenkk, 2019-03-29:
Please remove this volume, it's not necessary if you don't want to support new devices:       - /opt/zigbee2mqtt/data/zigbee-shepherd-converters:/app/node_modules/zigbee-shepherd-converters

pathob, 2019-04-06:
I had the same configuration with the mentioned volume and exactly the same issue after updating the Docker image and my clone of the zigbee-shepherd-converters repo yesterday. Removing the volume solved this problem for me. Thank you!

closed 2019-04-06 by Koenkk