Koenkk/zigbee2mqtt #1345: Most routers offline no pairing

mullischlumpf, 2019-03-29:
Hi i have the problem, that i could not integrate my new Xiaomi Aqara Motion Sensor and Door Sensors to me zigbee network. I have 26 devices with many routers so i think there should be no problem to add more devices. In the network map i saw, that many of my routers are marked as offline?!? But they work without problems. Is there a way to get them online or what is the problem?

Here is my network map.

  "0x0017880102e4536f" -> "0x0017880102e45f36" [label="172"]                                                                                                    
  "0x0017880102e4536f" -> "0x000b57fffe9e4f52" [label="100"]                                                                                                    
  "0x00178801023933ca" [style="rounded", label="{flurlicht2|Router|Philips Hue white ambiance GU10 (8718696598283)|offline}"];                                  
  "0x00178801023933ca" -> "0x00124b001936ada8" [label="77"]                                                                                                     
  "0x00178801023933ca" -> "0x84182600000db28a" [label="48"]                                                                                                     
  "0x00178801023933ca" -> "0x0017880102e45f36" [label="205"]                                                                                                    
  "0x0017880102e45f36" [style="rounded", label="{flurlicht3|Router|Philips Hue white ambiance GU10 (8718696598283)|offline}"];                                  
  "0x0017880102e45f36" -> "0x00124b001936ada8" [label="46"]                                                                                                     
  "0x0017880102e45f36" -> "0x84182600000db28a" [label="60"]                                                                                                     
  "0x0017880102e45f36" -> "0x000b57fffe9e4f52" [label="109"]                                                                                                    
  "0x90fd9ffffe7171ad" [style="rounded", label="{terrassenlicht|Router|IKEA TRADFRI LED bulb E26/E27 950 lumen, dimmable, white spectrum, clear (LED1546G12)|off
line}"];                                                                                                                                                        
  "0x90fd9ffffe7171ad" -> "0x00124b001936ada8" [label="3"]                                                                                                      
  "0x90fd9ffffe7171ad" -> "0x84182600000db28a" [label="26"]                                                                                                     
  "0x90fd9ffffe7171ad" -> "0x0017880102e45f36" [label="28"]                                                                                                     
  "0x90fd9ffffe7171ad" -> "0x000b57fffe9e4f52" [label="130"]                                                                                                    
  "0x7cb03eaa00a9faa5" [style="rounded", label="{terrassenleiste1|Router|OSRAM Outdoor Flex RGBW (4058075036185)|offline}"];                                    
  "0x7cb03eaa00a9faa5" -> "0x84182600000db28a" [label="24"]                                                                                                     
  "0x7cb03eaa00a9faa5" -> "0x0017880102e45f36" [label="4"]                                                                                                      
  "0x7cb03eaa00a9faa5" -> "0x000b57fffe9e4f52" [label="96"]                                                                                                     
  "0x0017880102e23eaf" [style="rounded, dashed", label="{schlafzimmerschalter2|EndDevice|Philips Hue dimmer switch (324131092621)|online}"];                    
  "0x8418260000dabc86" [style="rounded", label="{küchenleiste1|Router|OSRAM Flex RGBW (4052899926110)|offline}"];                                               
  "0x8418260000dabc86" -> "0x00124b001936ada8" [label="22"]                                                                                                     
  "0x8418260000dabc86" -> "0x84182600000db28a" [label="62"]                                                                                                     
  "0x8418260000dabc86" -> "0x0017880102e45f36" [label="142"]                                                                                                    
  "0x8418260000dabc86" -> "0x000b57fffe9e4f52" [label="49"]                                                                                                     
  "0x8418260000dabaa9" [style="rounded", label="{küchenleiste2|Router|OSRAM Flex RGBW (4052899926110)|offline}"];                                               
  "0x8418260000dabaa9" -> "0x00124b001936ada8" [label="1"]                                                                                                      
  "0x8418260000dabaa9" -> "0x84182600000db28a" [label="24"]                                                                                                     
  "0x8418260000dabaa9" -> "0x0017880102e45f36" [label="130"]                                                                                                    
  "0x8418260000dabaa9" -> "0x000b57fffe9e4f52" [label="41"]                                                                                                     
  "0x84182600000db28a" [style="rounded", label="{ankleideleiste|Router|OSRAM Smart+ plug (AB3257001NJ)|online}"];                                               
  "0x84182600000db28a" -> "0x00124b001936ada8" [label="57"]                                                                                                     
  "0x84182600000db28a" -> "0x0017880102e45f36" [label="195"]                                                                                                    
  "0x84182600000db28a" -> "0x000b57fffe9e4f52" [label="86"]                                                                                                     
}'

djchen, 2019-03-29:
Thats perfectly normal for the routers to show offline. Do you see any other devices connected to them?

Pairing the Xiaomi sensors is extremely finicky. You need to pick them right up against your coordinator and make sure to follow the tips here very precisely (https://www.zigbee2mqtt.io/getting_started/pairing_devices.html). It took me quite a few tries for the door/window sensors.

mullischlumpf, 2019-03-30:
i still had noch success with the xiaomi sensor :-( I had a message "device incoming" but then nothing happens. I then tried to add another ikea bulb but also with no success. Nothing happens in the log when i try to join devices...

mullischlumpf, 2019-04-08:
It seems, that the maximum of devices is reached in my zigbee cc2531. I reset the stick and readd all devices. I could add 29 devices (including xiaomi sensor) without any problems. Now no new device could join the network. I thought the limit is higher with many routers connected?!? 
But it seems not to be a bug so you could close the issue.

closed 2019-04-08 by Koenkk

tyjtyj, 2020-05-19:
> It seems, that the maximum of devices is reached in my zigbee cc2531. I reset the stick and readd all devices. I could add 29 devices (including xiaomi sensor) without any problems. Now no new [...]

Sorry to bring this 1 year old tread up. I reach device limit with 3 router and 20 xiaomi sensor. 
I believe the 20 xiaomi sensor is the limit regardless how many router you have you. 

Do you have work around on your issue ? Does source routing firmware allow beyond 20 xiaomi sensor ?