Koenkk/zigbee2mqtt #1346: zigbee2mqtt suddenly stopped working after some months

Strind39, 2019-03-29:
Hi,

Until earlier today everything seems to work fine for several months. Suddenly out of nothing no updates from my Zigbee devices anymore. With a restart (npm start) I got the following error message:

Error: Cannot find module './common'
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:582:15)
    at Function.Module._load (internal/modules/cjs/loader.js:508:25)
    at Module.require (internal/modules/cjs/loader.js:637:17)
    at require (internal/modules/cjs/helpers.js:22:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-shepherd-converters/converters/fromZigbee.js:4:16)
    at Module._compile (internal/modules/cjs/loader.js:701:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
    at Module.load (internal/modules/cjs/loader.js:600:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
    at Function.Module._load (internal/modules/cjs/loader.js:531:3)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.0.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.0.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2019-03-29T21_43_34_917Z-debug.log

but also with the information in the logfile I am not able to understand the issue as I did not change anything:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.4.1
3 info using node@v10.15.3
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.0.1~prestart: zigbee2mqtt@1.0.1
6 info lifecycle zigbee2mqtt@1.0.1~start: zigbee2mqtt@1.0.1
7 verbose lifecycle zigbee2mqtt@1.0.1~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.0.1~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/opt/zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.0.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.0.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.0.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.0.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.0.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:301:16)
13 verbose stack     at EventEmitter.emit (events.js:189:13)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:189:13)
13 verbose stack     at maybeClose (internal/child_process.js:970:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:259:5)
14 verbose pkgid zigbee2mqtt@1.0.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 4.14.98-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v10.15.3
19 verbose npm  v6.4.1
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.0.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.0.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

can anybody help?

Thanks
Ray

Koenkk, 2019-03-30:
In the zigbee2mqtt folder, can you do:

```bash
sudo npm install -g npm 
rm -rf node_modules
npm install
```

Fabiancrg, 2019-03-30:
I had a similar issue a few weeks ago, finally I had to flash again my CC2531, hopefully with latest firmware I didn't lost anything.

Strind39, 2019-03-31:
@Koenkk, thanks it works. Still got an error message on npm start but all devices are active again. Thanks

closed 2019-03-31 by Strind39