Koenkk/zigbee2mqtt #1347: Adding support for Nue/3a Temp and Humidity Sensor

CamHaug, 2019-03-30:
Hi, trying to add support for this sensor....
Have added this: 
    {
        zigbeeModel: ['FB56-THM12HM1.4'],
        model: 'FB56-THM12HM1.4',
        description: 'Smart Temperature and Humidity Detector',
        supports: 'Temperature and Humidity',
        toZigbee: [],
        fromZigbee: [
            fz.generic_temperature,
        ],
    },

But need to add the converter for humidity. (I think it also reports battery voltage as well
I am pretty new to all of this to some pointers would be cool.

Device is returning:

  zigbee2mqtt:info 3/30/2019, 9:54:28 AM MQTT publish: topic 'zigbee2mqtt/0x00124b0014ce4833', payload '{"temperature":25.39,"linkquality":84}'
  zigbee2mqtt:warn 3/30/2019, 9:54:28 AM No converter available for 'FB56-THM12HM1.4' with cid 'msRelativeHumidity', type 'attReport' and data '{"cid":"msRelativeHumidity","data":{"measuredValue":4063}}'

  zigbee2mqtt:info 3/30/2019, 9:54:29 AM MQTT publish: topic 'zigbee2mqtt/0x00124b0014ce4833', payload '{"temperature":25.39,"linkquality":84}'

I assume this means the temp bit is working OK. And that relative humidity is 40.63%....maybe?

Once i get this one done i would like to attempt the IR blaster they have but have no idea where to start on that one :) Baby Steps.

Koenkk, 2019-03-30:
I expect that you can use the `fz.xiaomi_humidity` converter.

CamHaug, 2019-03-31:
Cool, humidity seems ok
8:20:32 AM MQTT publish: topic 'zigbee2mqtt/0x00124b0014ce4833', payload '{"temperature":24.27,"linkquality":97,"humidity":40.32}'

Now also getting:
No converter available for 'FB56-THM12HM1.4' with cid 'msTemperatureMeasurement', type 'devChange' and data '{"cid":"msTemperatureMeasurement","data":{"measuredValue":2427}}'

I assume measuredValue is battery level..... maybe the other one is for temperature change?

CamHaug, 2019-03-31:
also this :\

No converter available for 'FB56-THM12HM1.4' with cid 'genBasic', type 'devChange' and data '{"cid":"genBasic","data":{"manufacturerName":"Feibit,co.ltd   ","modelId":"FB56-THM12HM1.4","powerSource":3}}'

Koenkk, 2019-04-01:
These messages can be ignored, use: `fz.ignore_basic_change` and `fz.ignore_temperature_change`

Koenkk, 2019-04-09:
To make it supported, you need to make a pr here: https://github.com/koenkk/zigbee-shepherd-converters

Afterwards I can do the home assistant integration

CamHaug, 2019-04-20:
I have no idea if i did it correctly....let me know.... Ii think the devices.js I used is pretty old...

CamHaug, 2019-05-11:
I wonder if there is a way to slow down the polling. it seems to report once every 5-10 seconds (and flattens the battery) Have any ideas?

Koenkk, 2019-05-14:
Can you share the log of that?

closed 2019-07-20 by stale[bot]