Koenkk/zigbee2mqtt #1350: Xiaomi Aqara Temperture and humidity sensor gives error: Device doesn't have "model" attribute, skipped

Wamor, 2019-03-30:
Hello,

Today I have received my Xiaomi Aqara temperature and humidity sensor (WSDCGQ11LM) and I tried to connect it to my Raspberry Pi with CC2530 + CC2591.
The device is detected but I see in the log-file the following message:
```
2019-03-30 20:23:37.865 (Home) MQTT message: zigbee2mqtt/bridge/log {'type': 'devices', 'message': [{'type': 'Router', 'ieeeAddr': '0xd0cf5efffe7b5415', 'friendly_name': '0xd0cf5efffe7b5415', 'model': 'E1603'}, {'type': 'EndDevice', 'ieeeAddr': '0xd0cf5efffe7ccd2c', 'friendly_name': '0xd0cf5efffe7ccd2c', 'model': 'ICTC-G-1'}, {'type': 'Router', 'ieeeAddr': '0xd0cf5efffef571a5', 'friendly_name': '0xd0cf5efffef571a5', 'model': 'LED1623G12'}, {'type': 'Router', 'ieeeAddr': '0xd0cf5efffecb6ed5', 'friendly_name': '0xd0cf5efffecb6ed5', 'model': 'E1603'}, {'type': 'Router', 'ieeeAddr': '0x90fd9ffffe0f75c7', 'friendly_name': '0x90fd9ffffe0f75c7'}]} 
2019-03-30 20:23:37.865 (Home) Received available devices list from bridge 
2019-03-30 20:23:37.865 (Home) Device E1603 0xd0cf5efffe7b5415 (0xd0cf5efffe7b5415) 
2019-03-30 20:23:37.866 (Home) Device ICTC-G-1 0xd0cf5efffe7ccd2c (0xd0cf5efffe7ccd2c) 
2019-03-30 20:23:37.866 (Home) Device LED1623G12 0xd0cf5efffef571a5 (0xd0cf5efffef571a5) 
2019-03-30 20:23:37.867 (Home) Device E1603 0xd0cf5efffecb6ed5 (0xd0cf5efffecb6ed5) 
2019-03-30 20:23:37.867 (Home) Device 0x90fd9ffffe0f75c7 (0x90fd9ffffe0f75c7) doesn't have "model" attribute, skipped
```
Does anybody have some idea what is happening here? Could it be the firmware of the Xiaomi device?

Best regards,

Wamor

Koenkk, 2019-04-01:
Probably the pairing procedure has not fully been completed.

I would recommend removing the device (http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove) and re-pair the device using: http://www.zigbee2mqtt.io/getting_started/pairing_devices.html#xiaomi

Wamor, 2019-04-02:
Hello Koenkk,

Please forgive me this (beginner) question but:
How do I send a ZigBee message through zigbee2mqtt to remove the device?
Can this be done in Domoticz or from the command-line in Linux?

Regards,

Wamor

Koenkk, 2019-04-03:
You can use MQTT.fx https://mqttfx.jensd.de

tommies1, 2019-04-03:
I had some problems with some of my xiaomi aqara temperature.
begin by pushing the reset button for 5 seconds, release AND don't forget to push again each  second the button during the pairing process.

Wamor, 2019-04-15:
Thank you tommies1 and Koenkk for your help.
I followed the instructions removing the device and re-paired again with the instructions suggested by tommies1. Everything works fine now.

Regards,

Wamor

closed 2019-04-15 by Wamor

Rudi1952, 2019-07-01:
I have the same issue with a XIAOMI body sensor, but i can't unpair this product.
following is the log-output: Zigbee2mqtt is running on a RPI3B with domoticz.
Maybe someone can help me solving this issue

2019-06-30 08:05:38.587  Status: (Zigbee2mqtt) Started.
2019-06-30 08:05:39.676  Status: (Zigbee2mqtt) Entering work loop.
2019-06-30 08:05:39.677  Status: (Zigbee2mqtt) Initialized version 0.0.17
2019-06-30 08:05:39.848  (Zigbee2mqtt) Zigbee2mqtt bridge is online
2019-06-30 08:05:39.848  (Zigbee2mqtt) Joining new devices is enabled on the zigbee bridge
2019-06-30 08:05:39.899  (Zigbee2mqtt) Received available devices list from bridge
2019-06-30 08:05:39.900  (Zigbee2mqtt) Joining new devices is enabled on the zigbee bridge
2019-06-30 08:25:24.271  (Zigbee2mqtt) Received available devices list from bridge
2019-06-30 08:25:24.272  (Zigbee2mqtt) Device WSDCGQ11LM 0x00158d00025f037b (vochtigheidssensor douche)
2019-06-30 08:25:24.272  (Zigbee2mqtt) Device ZNLDP12LM 0x00158d0002e8f7a8 (0x00158d0002e8f7a8)
2019-06-30 08:25:24.273  (Zigbee2mqtt) Device CC2530.ROUTER 0x00124b001bc746da (0x00124b001bc746da)
2019-06-30 08:25:24.273  (Zigbee2mqtt) Device 0x00158d00031e13ed (0x00158d00031e13ed) doesn't have "model" attribute, skipped