Koenkk/zigbee2mqtt #1351: Group state based on individual devices states

Bojan023, 2019-03-30:
I've updated today to the latest firmware version for my CC2531 coordinator since I wanted support for groups. I'm using Homeassistant to control 12 Ikea bulbs plus 8 Aqara sensors and 3 wall switches. So far I can conclude it increases reliability enormously, my compliments! 

The thing which really bothers me is that state changes of devices (originating from outside a group) are not linked to/reported as group state afterwards. For instance: turn on all lights from group X with group command --> shut off all lamps within that group without group command --> group state still on (at least, in Homeassistant). It would be a pain in the ass to make automations for all Zigbee groups to check whenever group states should be changed if devices with a group are changing states. 

Is a group state report functionality something that should be build into zigbee2mqtt or should this be implemented elsewhere?

xoxys, 2019-04-28:
Would be awesome to have this in zigbee2mqtt

Koenkk, 2019-04-28:
I'm currently working on this and it's almost ready, expect this soon (#1464)

xoxys, 2019-04-28:
Good news

xoxys, 2019-04-28:
@Koenkk Groups lost his state after restart home assistant is this also a known issue?

Koenkk, 2019-04-29:
@xoxys that depends on how you configured groups in home assistant. Correct configuration can be found here: http://www.zigbee2mqtt.io/integration/home_assistant.html#groups

Koenkk, 2019-04-29:
Implemented, for more info see: https://github.com/Koenkk/zigbee2mqtt/issues/764#issuecomment-487696964

closed 2019-04-29 by Koenkk