Koenkk/zigbee2mqtt #1352: Filter markdown in console and networkmap

sapkra, 2019-03-30:
In my network map and device listing (console at startup) I can see the markdown provided in the description of the devices.
My suggestion is to remove the markdown or the description from these views or get the text of the markdown link and show it without the link. I think it could be possible with some regex.

Console:
```
CC2530.ROUTER - Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (Router)

KS-SM001 - Ksentry Electronics [Zigbee OnOff Controller](http://ksentry.manufacturer.globalsources.com/si/6008837134660/pdtl/ZigBee-module/1162731630/zigbee-on-off-controller-modules.htm) (Router)
```

Network Map:
```
Custom devices (DiY) [CC2530 router](http://ptvo.info/cc2530-based-zigbee-coordinator-and-router-112/) (CC2530.ROUTER)

Ksentry Electronics [Zigbee OnOff Controller](http://ksentry.manufacturer.globalsources.com/si/6008837134660/pdtl/ZigBee-module/1162731630/zigbee-on-off-controller-modules.htm) (KS-SM001)
```

Koenkk, 2019-04-01:
Yes, this would indeed be nice :smile:

Could you make a PR for this?

sapkra, 2019-04-02:
I definitely can't find any time to to this within the next months. Maybe someone in the community will find the time or I can do it later this year.

sapkra, 2019-06-01:
I think this should not be marked as stale.

closed 2019-08-07 by stale[bot]