Koenkk/zigbee2mqtt #1355: CC2531 - CC2530 / Range Performance ((rx)/tx-power)

tb-killa, 2019-03-31:
Now that I've been working on the development and integration of Zigbee devices in the last months, I've noticed the following summary.
- CC2531 is an excellent simple coordinator, because we have the plug and play method thanks to the USB connection.
- CC2531 was never meant to be a full coordinator by TI, the layout as well as the performance adjustments on the part of the controller were only planned for sniffer use.

- The CC2530 is better suited as a coordinator with a higher range, but is more complex due to its circuitry, as additional accessories are required.
- In the context of its work as a fully-fledged coordinator, the CC2530 can also have a very long range.

In the course of my further research I came across an interesting hint within the TI Forum, where an "undocumented" interface of the ZNP is pointed out:
https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/443496/1593040#1593040

```
If you use ZNP, you can use the undocumented MT_SYS_STACK_TUNE command to the ZNP with the data for setting the Tx transmit power. It is an SREQ with 2 bytes of data, so

02 21 0F 00 XX where:

02 = 2 bytes of data

21 = SREQ + MT_RPC_SYS_SYS

0F = MT_SYS_STACK_TUNE

00 = STK_TX_PWR

XX = a valid power setting from ZMAC.h:

typedef enum
{
TX_PWR_MINUS_22 = -22,
TX_PWR_MINUS_21,
TX_PWR_MINUS_20,
TX_PWR_MINUS_19,
TX_PWR_MINUS_18,
TX_PWR_MINUS_17,
TX_PWR_MINUS_16,
TX_PWR_MINUS_15,
TX_PWR_MINUS_14,
TX_PWR_MINUS_13,
TX_PWR_MINUS_12,
TX_PWR_MINUS_11,
TX_PWR_MINUS_10,
TX_PWR_MINUS_9,
TX_PWR_MINUS_8,
TX_PWR_MINUS_7,
TX_PWR_MINUS_6,
TX_PWR_MINUS_5,
TX_PWR_MINUS_4,
TX_PWR_MINUS_3,
TX_PWR_MINUS_2,
TX_PWR_MINUS_1,
TX_PWR_ZERO,
TX_PWR_PLUS_1,
TX_PWR_PLUS_2,
TX_PWR_PLUS_3,
TX_PWR_PLUS_4,
TX_PWR_PLUS_5,
TX_PWR_PLUS_6,
TX_PWR_PLUS_7,
TX_PWR_PLUS_8,
TX_PWR_PLUS_9,
TX_PWR_PLUS_10,
TX_PWR_PLUS_11,
TX_PWR_PLUS_12,
TX_PWR_PLUS_13,
TX_PWR_PLUS_14,
TX_PWR_PLUS_15,
TX_PWR_PLUS_16,
TX_PWR_PLUS_17,
TX_PWR_PLUS_18,
TX_PWR_PLUS_19
} ZMacTransmitPower_t; // The transmit power in units of -1 dBm.
```
Since I can't estimate whether these changes could affect the CC2531 as well (possibly only with a modified antenna), I would like to do various tests with other people who like to experiment too.

Maybe @Koenkk could do some implementation after the results ?

lolorc, 2019-03-31:
has anybody ever tried to compile for cc2538+cc2592 ? I have one at home, would love to use it to replace my cc2530+cc2591

jesperldk, 2019-04-01:
> * The CC2530 is better suited as a coordinator with a higher range, but is more complex due to its circuitry, as additional accessories are required.

Is this statement widely accepted? I am running a testsetup with a CC2530 as coordinator, but was planning to go production with a CC2531 when I get it in af few days. I am also getting a second CC2530, so I could choose between:
1) A CC2531 as coordinator and two CC2530 as router, or
2) A CC2530 as coordinator and another CC2530 as router

I am pretty sure both setups will have enough range. I am running with a CC2530 right now just because my CC2531 was dead. There seems to be many here with problems with the CC2531 and few with problems with CC2530, but that may of course be just because most use CC2531. 
Personally I find that it is simple enough to set up a CC2530 ad coordinator, so that does not realy influence my choice.

lolorc, 2019-04-01:
it's also easier to find a cc2530 + frontend (cc2591/cc2592) than a cc2531 with frontend, that's why I'm using a cc2530+frontend as coordinator.

Koenkk, 2019-04-03:
I've received a free CC2652R from Texas Instruments, this device is much more capable of being a coordinator (and would also bring us Zigbee 3.0).

Ultimately I see:
- Users with bigger network could use the CC2652R
- Smaller networks can still benefit of the cheapness and simplicity of the CC2531/CC2530

lolorc, 2019-04-03:
if cc2538 is a tad more expensible than cc2530, cc2652R is a lot more :-) but it would also bring BLE.
I'll try to find a way (and time) to compile a firmware for cc2538, it's supposed to be compatible with cc-znp.

fredrikgk, 2019-04-04:
> CC2531 was never meant to be a full coordinator by TI, the layout as well as the performance adjustments on the part of the controller were only planned for sniffer use.
>
> The CC2530 is better suited [...]

I work with TI´s low power radio devices, including the CC253x family, and this is not correct. The only difference between the CC2530 and the CC2531 is the USB peripheral on the latter device. Both devices are equally suited for Zigbee operation and both have the same RF performance.

fredrikgk, 2019-04-04:
For anybody looking into Zigbee 3.0, I would highly recommend the CC2652R instead of the CC2538 since it is a newer and better supported device. The CC26x2R LaunchPad development kit could potentially also interface directly with a Raspberry Pi (or any machine running Zigbee2MQTT) through the onboard debugger which also acts as serial interface over USB. 

Aditionally, the CC1352P device could be used which will enable up to 20 dBm output power.

closed 2019-04-07 by tb-killa

Koenkk, 2019-04-08:
@fredrikgk the CC2652R should now work with the latest dev branch, will publish firmware and guides soon.

matlab22, 2019-08-19:
Any plans to support the [CC1352P](http://www.ti.com/product/CC1352P) with the integrated PA?

Koenkk, 2019-08-19:
@matlab22 yes, I already own two of them, but first need to fix the stability problems with the CC2652R (which probably also the CC1352P has).

BigDi, 2019-09-27:
@tb-killa I know it is closed. But I'm interessed in power setting of cc2530/31.
Is the setting already implemented in the firmware?