Koenkk/zigbee2mqtt #1359: Ikea bulb does not connect to cc2530 router

frank3523, 2019-03-31:
Because my ikea bulb is too far from the coordinator cc2531 I installed a cc2530 router. The router joins the zigbee network. But after joining the ikea lamp with the coordinator and then installing it in its final place the Ikea bulb is out of reach. I would expect that the router would pick up the ikea bulb but this is not the case. So how can I connect the ikea bulb with the router.

ekaats, 2019-04-03:
In my experience it can take awhile for Tradfri bulbs to connect via other routers. It can help to turn them of and on as well. Furthermore, connections sometimes take awhile to establish. Currently I see in my network map that even battery powered Xiaomi devices are able to connect via routers (Tradfri bulbs as well, in my case).
I have no experience with using a cc2530 as router though, but I suspect it to work as well as any other powered zigbee device. Maybe the cc2530 has less coverage, and is not able to setup a strong two way connection.

frank3523, 2019-04-05:
Yes thnnx for the answer. I had to wait a while before it connected.
Now all works fine.

closed 2019-04-05 by frank3523