Koenkk/zigbee2mqtt #1360: NOT ISSUE - New Aqara accessory

GerSant, 2019-04-01:
Only for you information

Aqara has shipped a new relay module LLKZMK11LM

https://homekitnews.com/2019/03/27/aqara-wireless-relay-controller/

Regards

lolorc, 2019-04-01:
just bought one :-)

fmolero, 2019-04-01:
I think I receive it this week

Koenkk, 2019-04-01:
Already supported in the dev branch!

closed 2019-04-01 by Koenkk

fmolero, 2019-04-09:
Hi, I am testing this device with zigbee2mqtt and everything works fine except that the publication in mqtt does not work to trigger a relay or another. In my case, if I publish

zigbee2mqtt/[Device ID]/set  {"state":"ON"} or {"state":"OFF"} works fine, but only activates or deactivates the L1 relay. 

If I publish in zigbee2mqtt/[Device ID]/l1 or l2/set  {"state":"ON"} or {"state":"OFF"} I receive the following message

abr 09 16:11:31 JeedomServer npm[12775]:   zigbee2mqtt:error 2019-4-9 16:11:31 Failed to find device with ieeAddr: 'XiaomiSwitch1/l1'
abr 09 16:11:31 JeedomServer npm[12775]:   zigbee2mqtt:info 2019-4-9 16:11:31 MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"entity_not_found","message":"XiaomiSwitch1/l1"}'
abr 09 16:11:31 JeedomServer npm[12775]:   zigbee2mqtt:warn 2019-4-9 16:11:31 Cannot handle MQTT message on 'zigbee2mqtt/XiaomiSwitch1/l1/set' with data '{"state":"OFF"}'

How can I activate a relay or another?

fmolero, 2019-04-11:
After investigating, I have noticed that the definition of this product is not well done at all and there is a small bug. The endpoints are defined in device.js as l1 and l2 but nevertheless if we use zigbee2mqtt / DEVICE_ID / l1 / set it does not work, however if it works if we use zigbee2mqtt / DEVICE_ID / left / set, for example or right. So I have changed in device.js l1 and l2 by left and right and it is already working correctly. Should we define l1 and l2 somewhere else, apart from device.js?

Koenkk, 2019-04-11:
@fmolero yes, l1 and l2 need to be added to this list: https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/util/utils.js#L31 can you make a pr?

fmolero, 2019-04-11:
Do I do the pr in the master or dev branch?

Koenkk, 2019-04-11:
Dev branch.

fmolero, 2019-04-11:
Ok, done.

Koenkk, 2019-04-11:
Can you provide a link to the pr? I dont see it.

fmolero, 2019-04-12:
Sorry, something had not done well. I just created the pr again, here is the link.
https://github.com/Koenkk/zigbee2mqtt/pull/1411