Koenkk/zigbee2mqtt #1362: DL-22 module (CC2530 + CC2591) support

kirbah, 2019-04-02:
I bought 12$ USD module (looks like great price) and trying to use it https://www.aliexpress.com/item/2-4G-Zigbee-Wireless-Transceiver-Module-Long-Distance-Wireless-Serial-Transceiver-Module/32703502764.html

There is some stock firmware in this module that should be usable to transfer any bytes over serial to zigbee. But stock is not interesting for sure.
Here is product page http://www.hexin-technology.com/1000m_TTL_to_ZigBee_Module-Product-566.html
There is also PDF with some wiring details http://www.hexin-technology.com/shared/manual/Diagram_DL-22.pdf

I was able to connect CC debugger and flush router firmware from https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/CC2530_CC2591 but it is not possible to pair it. There is blinking led once per second and that is all that I see.
I was able to get following log in the serial after connecting CH340G adapter to the module and trying to pair with CC2531 coordinator with all the recent firmwares.

> ЧH─IEEE:  0024B000CBDD4FE Ч!H─TexasInstruments SW5: Help       Н
> ЧH─IEEE:  0124B000CBDD4FE Ч!H─TexasInstruments SW5: Help       Н
> Ч!H─TexasInstruments SW5: Help       Н
> ЧH─IEEE:  00124B000BDD4FE Ч!H─TexasInstruments SW5: Help       Н
> Ч!H─TexasInstruments SW5: Help       Н
> ЧH─IEEE:  0024B000CBDD4FE Ч!H─TexasInstruments SW5: Help       Н
> Ч!H─TexasInstruments SW5: Help       Н
> ЧH─IEEE:  00124B000CBDD4FE Ч!H─TexasInstruments SW5: Help       Н
> Ч!H─TexasInstruments SW5: Help       Н
> 

My next attept was coordinator firmware and it is almost working with following config settings in configuration.yaml
```
port: /dev/ttyUSB0
rtscts: false
```

I simply replaced my CC2531 coordingator with the new one that also has CH340G, updated config and restarted zigbee2mqtt.
I was able to get some commands from aqara cube without any repairing but it was not very reliable and cube was located close to the CC2530 coordinator. Just few commands were transfered during my attempts. Bunch of other zigbee devices are not visible at all. My guess that CC2591 is not turned on or there might be some hardware issue with my module. http://www.hexin-technology.com/shared/manual/Diagram_DL-22.pdf has some details about pins that should be used to control CC2591 and probably firmware use different pins.
I am going to try to flash one more module to double check.
Here is wiring to flash module with CC debugger. I just hold wires in the 3 pins for DD/DC/Reset and it was enough without soldering them.

[image]

Diggen85, 2019-04-06:
Can confirm that this works with [DL-20 from Hexin](http://www.hexin-technology.com/250m_TTL_to_ZigBee_Module-Product-565.html),[ Schematic.](http://www.hexin-technology.com/shared/manual/Diagram_DL-20.pdf). Its the CC2350 only version of Hexins wireless UART series. Also available at Aliexpress.

Had a little bit struggle first, because the Debug-Header has **reversed** layout. But flashing as coordinator with CC2530_MAX_STABILITY_20190315 firmware worked flawlessly.
The reception is almost the same as a CC2531, maybe a little bit less.

[image] Sadly my only pic

rowdy15, 2019-04-15:
I also bought the DL-22 and have tried router and coordinator firmwares with no success.

The router firmware as suggested by kirbah just blinks.  With the default coordinator firmware I am able get zigbee2mqtt up and running, however I wasn't able to pair any devices.

I also inspected the pcb and I could be wrong, but it seems as though the pin connections are different to the ones in http://www.ti.com/lit/an/swra308a/swra308a.pdf.  Page 17.

SHOULD BE:
**cc2530   -    cc2591**
P1_1   -    PAEN
P1_4   -    EN
P0_7   -    HGM          

BUT ACTUALLY SEEMS TO BE:
**cc2530   -    cc2591**
P1_4   -    PAEN
P1_5   -    EN
P0_7   -    HGM         

I really could be wrong about this, but I was wondering if it's possible to change the pin assignments and create a new .hex file for the DL-22?

BoBeRzE, 2019-04-24:
Any News? I have the same Problem with my DL-22

ptvoinfo, 2019-06-06:
I can try to make a router firmware, if you are interesting.

BoBeRzE, 2019-06-06:
> I can try to make a router firmware, if you are interesting.

Hi ptvoinfo,

this would be great.

ptvoinfo, 2019-06-06:
@BoBeRzE 
Please, try the following:
http://ptvo.info/wp-content/uploads/2018/09/cc2530_cc2591_hexin_dl22_1.2.2a.44539_firmware.zip

BoBeRzE, 2019-06-06:
Hi ptvinfo,

thank you very much. I will give it a try when i'm back at home in the evening.

rowdy15, 2019-06-06:
Thanks heaps.  Gonna test In the next day or so too.

Sam Thompson

AndrewLinden, 2019-06-06:
@ptvoinfo 
Seems to work, thanks a lot!

BoBeRzE, 2019-06-06:
@ptvoinfo

my test was also successful. One question: Is it possible to use the onboard button for e.g activate pairing-mode?

Thank a lot

ptvoinfo, 2019-06-07:
@BoBeRzE 
Currently, the router does not support this button.

BoBeRzE, 2019-06-07:
@ptvoinfo
is it possible to support such a feature? The board also has 4 different LEDs. Can you start something useful with that?
If I can help (test firmwares) please let me know

Many Thanks

rowdy15, 2019-06-07:
I flashed the diag hex firmware and it seems to be showing on the zigbee network.  I have it running right next to the usb 2531 coordinator.  one xiaomi sensor has doubled its linkquality.  another has been lost.  I tried rejoining it (press and hold the button?) but no luck.  I'm not too familiar with the process of setting up a coordainator with routers network (if there is one), so will do some reading over the next week or so to see if I'm doing something wrong and if the DL-22 is actually working satisfactorily.

P.S. @ptvoinfo I read the read me that came with the firmware you uploaded. And I am surprised (impressed with myself) that my inspection of the pinouts are correct (maybe). :-)

Anyway, thankyou for your efforts and hopefully I can give some more meaningful feedback over the next week or so.

rowdy15, 2019-06-07:
Also just wondering if I should have:

rtscts: false

in the configuration.yaml file?  Thanks.

ptvoinfo, 2019-06-07:
@rowdy15 This is the router firmware. So, configuration.yaml does not matter at all.

rowdy15, 2019-06-07:
Ok thanks.

Sam Thompson

duylnk68, 2019-06-08:
@ptvoinfo: can you please to provide a coordinate firmware for DL-22? (Or please help me how to patch the source code to change the pin assignments)

Thanks very much :)

duylnk68, 2019-06-09:
@ptvoinfo: I've patched the source code to change pin assignments and my DL-22 works successfully.

But I still cannot understand that my DL-22 has "linkquality" weaker than my USB stick CC2531 which comes without amplifier CC2519.

**DL-22:**
```
'{"contact":false,"linkquality":46,"battery":100,"voltage":3025}'
```

**USB Stick CC2531:**
```
'{"contact":false,"linkquality":102,"battery":100,"voltage":3025}'
```

**This is my patch:**
```
diff --git a/Z-Stack Home 1.2.2a.44539/Components/mac/low_level/srf04/single_chip/mac_radio_defs.c b/Z-Stack Home 1.2.2a.44539/Components/mac/low_level/srf04/single_chip/mac_radio_defs.c
index 15058db..d0e73ee 100644
--- a/Z-Stack Home 1.2.2a.44539/Components/mac/low_level/srf04/single_chip/mac_radio_defs.c	
+++ b/Z-Stack Home 1.2.2a.44539/Components/mac/low_level/srf04/single_chip/mac_radio_defs.c	
@@ -566,13 +566,13 @@ MAC_INTERNAL_API void macRadioTurnOnPower(void)
     }  
     else 
     {   
-      /* P1_1 -> PAEN */
+      /* P1_4 -> PAEN */
       RFC_OBS_CTRL0 = RFC_OBS_CTRL_PA_PD_INV;
-      OBSSEL1       = OBSSEL_OBS_CTRL0;
+      OBSSEL4       = OBSSEL_OBS_CTRL0;
       
-      /* P1_4 -> EN (LNA control) */
+      /* P1_5 -> EN (LNA control) */
       RFC_OBS_CTRL1 = RFC_OBS_CTRL_LNAMIX_PD_INV;
-      OBSSEL4       = OBSSEL_OBS_CTRL1;
+      OBSSEL5       = OBSSEL_OBS_CTRL1;
     }
     
     /* For any RX, change CCA settings for CC2591 compression workaround.
@@ -628,11 +628,11 @@ MAC_INTERNAL_API void macRadioTurnOffPower(void)
     {   
       if (paLnaChip == PA_LNA_CC2591  ||  paLnaChip == PA_LNA_CC2590)
       {
-        /* Set direction of P1_4 to output and pulled down to prevent any leakage
+        /* Set direction of P1_5 to output and pulled down to prevent any leakage
          * when used to drive PA LNA		
 		 */
-        P1DIR |= BV(4);
-        P1_4 = 0;
+        P1DIR |= BV(5);
+        P1_5 = 0;
       }
     }
   } 
```

kirbah, 2019-06-09:
@ptvoinfo Thanks a lot for the router firmware. Linkquality is better than CC2531. Finally I am able to replace weak router and get stable link to outdoor sensor.

[image]
[image]
[image]
[image]

ptvoinfo, 2019-06-10:
@duylnk68 I'm sorry, but my coordinator firmware is obsolete. Koenkk made many changes. Here is my changes in mac_radio_defs.c

```
#if defined PA_LNA_CC2591_HEXIN_DL22 //!!! ptvoinfo
      /* P1_4 -> PAEN */
      RFC_OBS_CTRL0 = RFC_OBS_CTRL_PA_PD_INV;
      OBSSEL4       = OBSSEL_OBS_CTRL0;
      
      /* P1_5 -> EN (LNA control) */
      RFC_OBS_CTRL1 = RFC_OBS_CTRL_LNAMIX_PD_INV;
      OBSSEL5       = OBSSEL_OBS_CTRL1;

#else
      /* P1_1 -> PAEN */
      RFC_OBS_CTRL0 = RFC_OBS_CTRL_PA_PD_INV;
      OBSSEL1       = OBSSEL_OBS_CTRL0;
      
      /* P1_4 -> EN (LNA control) */
      RFC_OBS_CTRL1 = RFC_OBS_CTRL_LNAMIX_PD_INV;
      OBSSEL4       = OBSSEL_OBS_CTRL1;
#endif      
```

duylnk68, 2019-06-30:
Hi @ptvoinfo, :)

According to http://www.hexin-technology.com/shared/manual/Diagram_DL-22.pdf
The HGM pin has been changed from P0_7 to P1_6, I think this is the reason that cause weak RX signal

[image]

AndrewLinden, 2019-06-30:
> I'm sorry, but my coordinator firmware is obsolete. Koenkk made many changes. 

@Koenkk Please add support for DL-22 module with correct pins in your firmwares! Thank you!

Koenkk, 2019-07-01:
Could somebody make a PR for https://github.com/koenkk/z-stack-firmware?

rowdy15, 2019-07-15:
Just trying to follow the issue thread above.  

@ptvoinfo have you changed the HGM pin from P0_7 to P1_6 (in mac_radio_defs??).  Because after looking at the schematic from @duylnk68 and then looking at the board again, I believe I was wrong in an earlier post in sayin that the HGM pin is P0_7.  I think it is actually P1_6 as stated in the schematic.  If you have not changed the HGM pin from P0_7 to P1_6 then that might be why the TX and RX are not as good as you would expect given it has the CC2591 amplifier chip (sorry... I know I'm stating the obvious). I just wanted to check to see if your router firmware has made the HGM pin change?

Also, is there a link to the source code for the router firmware.  I see the COMPILE.MD for the coordinator firmware, so I'm just wondering if there is the same thing for the router firmware.  I have IAR installed and running and can compile and run the coordinator firmware but would like to work on the router sourcecode for the DL-22 if possible.

ptvoinfo, 2019-07-15:
@rowdy15, Yes, of course.

Sorry, but I didn't upload the sources (patches) for the router firmware.

closed 2019-09-20 by stale[bot]

blakadder, 2020-03-26:
Resurrected!

Did the modification talked about here get merged in Z-Stack firmware in the end?

AndrewLinden, 2022-01-17:
Just leaving a note here for people who have the DL-22 laying around and searching for firmware:

Currently there is no support for DL-22 board in https://github.com/Koenkk/Z-Stack-firmware or [zigbee-configurable-firmware](https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/62)

But the DL-22 board is supported by @ptvoinfo router only firmware for CC2530:
Info page: https://ptvo.info/cc2530-cc2590-cc2591-cc2592-rfx2401-routers-190/
Firmware: https://ptvo.info/download/cc2530_1.2.2a.44539_firmware.zip