Koenkk/zigbee2mqtt #1363: Philips Hue Dimmer Switch stop working

Bastiencc, 2019-04-02:
Hi guys

My config :
-  zigbee2mqtt version 1.2.1 (commit #9a15e8c)
-  Coordinator firmware version: '20190223'

Some of my Philips Hue Dimmer Switch (RWL021 / 324131092621) work for a moment (few days, or hours, or minutes ...) and then stop working. I have to reset them (by pressing the 4 buttons on the switch) to work again. 
Sometimes I have to remove them from Z2M for pair again.

it's very strange...

the log when I pair (I think something is broken there) : 

```
Apr 02 07:19:09 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:09 AM Zigbee: allowing new devices to join.
Apr 02 07:19:09 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:09 AM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.2.1","commit":"9a15e8c","coordinator_firmware":20190223,"log_level":"info","permit_join":true}'
Apr 02 07:19:20 raspberrypi npm[8247]:   zigbee2mqtt:warn 4/2/2019, 7:19:20 AM Device with modelID 'undefined' is not supported.
Apr 02 07:19:20 raspberrypi npm[8247]:   zigbee2mqtt:warn 4/2/2019, 7:19:20 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 02 07:19:21 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:21 AM MQTT publish: topic 'homeassistant/sensor/0x001788010408cda6/action/config', payload '{"icon":"mdi:gesture-double-tap","value_template":"{{ value_json.action }}","expire_after":1,"state_topic":"zigbee2mqtt/salon_switch_canape","json_attributes_topic":"zigbee2mqtt/salon_switch_canape","name":"salon_switch_canape_action","unique_id":"0x001788010408cda6_action_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010408cda6","name":"salon_switch_canape","sw_version":"Zigbee2mqtt 1.2.1","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 02 07:19:21 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:21 AM MQTT publish: topic 'homeassistant/sensor/0x001788010408cda6/battery/config', payload '{"unit_of_measurement":"%","device_class":"battery","value_template":"{{ value_json.battery }}","state_topic":"zigbee2mqtt/salon_switch_canape","json_attributes_topic":"zigbee2mqtt/salon_switch_canape","name":"salon_switch_canape_battery","unique_id":"0x001788010408cda6_battery_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010408cda6","name":"salon_switch_canape","sw_version":"Zigbee2mqtt 1.2.1","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 02 07:19:21 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:21 AM MQTT publish: topic 'homeassistant/sensor/0x001788010408cda6/linkquality/config', payload '{"unit_of_measurement":"-","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/salon_switch_canape","json_attributes_topic":"zigbee2mqtt/salon_switch_canape","name":"salon_switch_canape_linkquality","unique_id":"0x001788010408cda6_linkquality_zigbee2mqtt","device":{"identifiers":"zigbee2mqtt_0x001788010408cda6","name":"salon_switch_canape","sw_version":"Zigbee2mqtt 1.2.1","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"},"availability_topic":"zigbee2mqtt/bridge/state"}'
Apr 02 07:19:25 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:25 AM Successfully configured salon_switch_canape (0x001788010408cda6)
Apr 02 07:19:27 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:27 AM MQTT publish: topic 'zigbee2mqtt/salon_switch_canape', payload '{"battery":31,"linkquality":55,"duration":0,"counter":1}'
Apr 02 07:19:29 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:29 AM Device incoming...
Apr 02 07:19:29 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:29 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
Apr 02 07:19:30 raspberrypi npm[8247]:   zigbee2mqtt:info 4/2/2019, 7:19:30 AM Zigbee: disabling joining new devices.
```

Koenkk, 2019-04-02:
Duplicate of #531

closed 2019-04-02 by Koenkk

Bastiencc, 2019-04-02:
Sorry, I did not see the #531 , thanks @Koenkk  !

marothe, 2019-09-16:
Having the same issue - the hue dimmer switch randomly loses connection and you have to re-pair them to Z2M. Did anybody of you guys find a solution for this?