Koenkk/zigbee2mqtt #1364: NPM errors after running on Raspberry - failed to start

Kratax, 2019-04-02:
Running fresh install, following the instructions, end up with the following log... Latest firmware and version of zigbee2mqtt.

Tried to restart on the stick and also raspberry. 

Thanks a lot for help!

https://pastebin.com/9frBq15A

joostjdj, 2019-04-02:
I actually have the exact same issue. Installed today as well.

jonnycastaway, 2019-04-03:
You have to install Nodejs 10. With Raspbain an older Nodejs is installed.

joostjdj, 2019-04-03:
For me 10 is already installed now, and still issues

Koenkk, 2019-04-03:
Please provide the `npm start` log

Kratax, 2019-04-03:
Hi,
thanks for suggestion, I have tried to update Nodejs, still not working, log attached below...

https://pastebin.com/r5MUmxD9

Koenkk, 2019-04-04:
This is not the kind of log I'm expecting, you can find the correct logs in `data/log`

Kratax, 2019-04-04:
Hi, 
Here you go

https://pastebin.com/LfmcW76A

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Koenkk, 2019-04-04:
What is the output of `ls /dev` and whats in your configuration.yaml?

Kratax, 2019-04-04:
Ls /dev 
https://pastebin.com/0CpA4PQA

Config
https://pastebin.com/CN9WN4wi

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

Koenkk, 2019-04-05:
Can you both follow: http://www.zigbee2mqtt.io/information/zigbee2mqtt_fails_to_start.html

Kratax, 2019-04-09:
Hi, I have followed the steps... unfortunatelly no success, details from terminal attached...

https://pastebin.com/y3JB29AN

Koenkk, 2019-04-10:
Unfortunately, everything looks OK, somehow the communication with the CC2531 fails. Can you try this on a different system? My other guess is that the CC2531 is broken somehow.

didiht, 2019-04-10:
Did you reset the Zigbee Stick?

closed 2019-06-16 by stale[bot]