Koenkk/zigbee2mqtt #1365: Networkmap online orphans

sascha777, 2019-04-02:
Hi folks,
I have a coordinator with the latest default image as well as two routers (also latest default image) running. 
I just have rearraged some devices from the coordinator to the routers to keep the coordinator handle <= 13 direct children.
Whilst two hours ago the networkmap showed the direct children connected to the coordinator/router, now it shows lots of end devices not connected anymore to the coordinator in the map. 
Anyway they are still shown as online end devices and also status changes are reported properly.
How can I check the node that these devices are connected to if not shows in the map?
Thanks

SargonofAssyria, 2019-04-03:
I experiance exactly the same. The network show a lot (not all) EndDevices NOT connected. So there is no info on linkquality. The devices it self work fine. I also see in the zigbee logging of the router that some of those devices are connected to the router with a linkquality and rssi present. But in the output of the network map they are not connected.

Koenkk, 2019-04-03:
End devices are not connected because they are sleeping during the network scan.

sascha777, 2019-04-03:
But I see some end devices connected all the time, even on the next days network scan. 

Is there another place to find the parent of these devices? Without this information I can‘t check how many devices are connected directly to the coordinator for not exceeding the device limit. 

Saying that it would be fantastic if you just can define that manually (e.g. in the device.db). Would that be possible?

Koenkk, 2019-04-03:
@sascha777 no, zigbee2mqtt doesn't decide the parent, the nodes in the network figure this out.

SargonofAssyria, 2019-04-03:
In the past it was different, and everything was connected with a linkquality.
Could not at least the linkquality be included in the networkmap? The info itself is present.

sascha777, 2019-04-07:
Agree, also for the parent node information to stay inside the device limit. 
I ran several times into the issue of having too many childs (without knowing) and had to reflash my coordinator.

stepanov1975, 2019-06-11:
Is it possible to add more information about networkmap? 
When exactly a network scan performed? Immediately after zigbee2mqtt/bridge/networkmap command or at some other time?  
All my end devices, except one router, are battery powered, so actually I will be unable to get any information from the map because all devices always sleeping during the scan? Is it any workaround if I still want to have a network map?
If only active devices should be connected in the map, why I still see some battery powered devices in my network map?

Koenkk, 2019-06-11:
Please continue this discussion in #652

closed 2019-06-11 by Koenkk