Koenkk/zigbee2mqtt #1366: SEGV starting zigbee-shepherd

larsimmisch, 2019-04-03:
New install. I successfully flashed my (knock-off) CC2351 with the latest firmware (the hex inside https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/CC2531/bin/CC2531ZNP-Prod_20190223.zip)

But I get a SEGV, in what looks like the initial handshake with the board:

```
  zigbee2mqtt:info 4/3/2019, 6:37:12 PM Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-03.18-37-12'
  zigbee2mqtt:debug 4/3/2019, 6:37:12 PM Removing old log directory '/opt/zigbee2mqtt/data/log/2019-04-03.15-01-58'
  serialport/bindings loading LinuxBinding +0ms
  zigbee2mqtt:debug 4/3/2019, 6:37:13 PM Using zigbee-shepherd with settings: '{"net":{"panId":6754,"extPanId":[221,221,221,221,221,221,221,221],"channelList":[11],"precfgkey":"HIDDEN"},"dbPath":"/opt/zigbee2mqtt/data/database.db","sp":{"baudRate":115200,"rtscts":true}}'
  zigbee2mqtt:debug 4/3/2019, 6:37:13 PM Loaded state from file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:debug 4/3/2019, 6:37:13 PM Saving state to file /opt/zigbee2mqtt/data/state.json
  zigbee2mqtt:info 4/3/2019, 6:37:13 PM Starting zigbee2mqtt version 1.2.1 (commit #4048cb8)
  zigbee2mqtt:info 4/3/2019, 6:37:13 PM Starting zigbee-shepherd
  zigbee-shepherd:init zigbee-shepherd booting... +0ms
  cc-znp Initialize, opening serial port +0ms
  serialport/stream opening path: /dev/ttyACM0 +0ms
  serialport/binding-abstract open +0ms
  serialport/stream _read queueing _read for after open +5ms
Segmentation fault
```

Koenkk, 2019-04-04:
What system and os are you running zigbee2mqtt on?

larsimmisch, 2019-04-04:
I’m running zigbee2mqtt on a raspberry pi 3 with a contemporary raspbian image. I think stretch. Compiling the serial adapter with npm needed a npm audit fix, but flashing the firmware went fine.

Koenkk, 2019-04-05:
Can you try without the `npm audit fix`? The problem is caused because of the serial adapter library.

larsimmisch, 2019-04-05:
Good point. It's just that `npm install` can't build the serial adapter before the `npm audit fix`.

This should probably be reported as an issue here (https://github.com/node-serialport/node-serialport/issues), correct?

```
make: Entering directory '/opt/zigbee2mqtt/node_modules/cc-znp/node_modules/serialport/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:41:75: warning: 'v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)' is deprecated: Use Isolate version [-Wdeprecated-declarations]
   v8::String::Utf8Value path(Nan::To<v8::String>(info[0]).ToLocalChecked());
                                                                           ^
In file included from /home/lars/.node-gyp/10.15.3/include/node/v8.h:26:0,
                 from /home/lars/.node-gyp/10.15.3/include/node/node.h:63,
                 from ../../../../../nan/nan.h:53,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/lars/.node-gyp/10.15.3/include/node/v8.h:2892:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^
/home/lars/.node-gyp/10.15.3/include/node/v8config.h:324:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/opt/zigbee2mqtt/node_modules/cc-znp/node_modules/serialport/node_modules/@serialport/bindings/build'

> @serialport/bindings@2.0.5 install /opt/zigbee2mqtt/node_modules/cc-znp/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=10.15.3 runtime=node arch=arm platform=linux)
make: Entering directory '/opt/zigbee2mqtt/node_modules/cc-znp/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':
../src/serialport.cpp:41:75: warning: 'v8::String::Utf8Value::Utf8Value(v8::Local<v8::Value>)' is deprecated: Use Isolate version [-Wdeprecated-declarations]
   v8::String::Utf8Value path(Nan::To<v8::String>(info[0]).ToLocalChecked());
                                                                           ^
In file included from /home/lars/.node-gyp/10.15.3/include/node/v8.h:26:0,
                 from /home/lars/.node-gyp/10.15.3/include/node/node.h:63,
                 from ../../../nan/nan.h:53,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/lars/.node-gyp/10.15.3/include/node/v8.h:2892:28: note: declared here
                   explicit Utf8Value(Local<v8::Value> obj));
                            ^
/home/lars/.node-gyp/10.15.3/include/node/v8config.h:324:3: note: in definition of macro 'V8_DEPRECATED'
   declarator __attribute__((deprecated(message)))
   ^~~~~~~~~~
  CXX(target) Release/obj.target/bindings/src/serialport_unix.o
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
```

Koenkk, 2019-04-05:
These are only warnings, no errors, does it work now?

larsimmisch, 2019-04-05:
(Doh! Slaps forehead. Sorry)

Yes, it does.

closed 2019-04-05 by larsimmisch