Koenkk/zigbee2mqtt #1370: CC2530 dies without meaningful info

antst, 2019-04-04:
I run dev branch (pulled today latest, didn't change) and latest FW.

Same problem appears with two coordinators: CC2530 and CC2530+CC2591 USB stick.

It used to work as a charm with my initial test-setup (one Xiaomi bulb and couple of Xiaomi wall switches, wired and wireless).
But once I added more devices (8 ikea bulbs, 4 of GU10 and 4 of E14), now after some certain time (or number of actions) CC2530 dies silently.

all I see in the log (with debug level) at transition point is bellow. Till `4/4/2019, 7:33:55 PM` it used to work OK, and then not anymore. All attempts to communicate with all devices ends up with `null failed with error Error: request timeout`. Nothing else.
What I can suspect here is that I am trying to turn on/off group of lights, so 4 messages. Could it be that this is because CTS/RTS are disabled in FW and C2530 can not keep up with communication on serial port?

```
4/4/2019, 7:33:52 PM - info: MQTT publish: topic 'zigbee2mqtt/stairs_light2', payload '{"state":"ON","linkquality":13,"brightness":190}'
4/4/2019, 7:33:52 PM - info: MQTT publish: topic 'zigbee2mqtt/stairs_light2', payload '{"state":"ON","linkquality":13,"brightness":190}'
4/4/2019, 7:33:52 PM - info: MQTT publish: topic 'zigbee2mqtt/upstairs_light2', payload '{"state":"ON","linkquality":0,"brightness":1}'
4/4/2019, 7:33:52 PM - debug: Received zigbee message of type 'attReport' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI bulb GU10 W 400lm' (0x000b57fffeefa1b3) of endpoint 1
4/4/2019, 7:33:52 PM - info: MQTT publish: topic 'zigbee2mqtt/upstairs_light2', payload '{"state":"ON","linkquality":13,"brightness":1}'
4/4/2019, 7:33:52 PM - debug: Received zigbee message of type 'devChange' with data '{"cid":"genOnOff","data":{"onOff":1}}' of device 'TRADFRI bulb GU10 W 400lm' (0x000b57fffeefa1b3) of endpoint 1
4/4/2019, 7:33:52 PM - info: MQTT publish: topic 'zigbee2mqtt/upstairs_light2', payload '{"state":"ON","linkquality":13,"brightness":1}'
4/4/2019, 7:33:55 PM - debug: Received MQTT message on 'zigbee2mqtt/stairs_light1/set' with data '{"state": "OFF"}'
4/4/2019, 7:33:55 PM - debug: Received MQTT message on 'zigbee2mqtt/upstairs_light2/set' with data '{"state": "OFF"}'
4/4/2019, 7:33:55 PM - info: Zigbee publish to device '0x000d6ffffee7fc69', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:33:55 PM - debug: Received MQTT message on 'zigbee2mqtt/upstairs_light1/set' with data '{"state": "OFF"}'
4/4/2019, 7:33:55 PM - debug: Received MQTT message on 'zigbee2mqtt/stairs_light2/set' with data '{"state": "OFF"}'
4/4/2019, 7:33:55 PM - info: Zigbee publish to device '0x000b57fffeefa1b3', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:33:55 PM - info: Zigbee publish to device '0x000d6ffffea7e904', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:33:56 PM - info: Zigbee publish to device '0x000d6ffffec939dc', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:33:56 PM - error: Zigbee publish to device '0x000d6ffffee7fc69', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
4/4/2019, 7:33:56 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x000d6ffffee7fc69","type":"device","friendlyName":"stairs_light1"},"message":"{\"state\": \"OFF\"}"}}'
4/4/2019, 7:33:58 PM - error: Zigbee publish to device '0x000b57fffeefa1b3', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
4/4/2019, 7:33:58 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x000b57fffeefa1b3","type":"device","friendlyName":"upstairs_light2"},"message":"{\"state\": \"OFF\"}"}}'
4/4/2019, 7:33:59 PM - error: Zigbee publish to device '0x000d6ffffea7e904', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
4/4/2019, 7:33:59 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x000d6ffffea7e904","type":"device","friendlyName":"upstairs_light1"},"message":"{\"state\": \"OFF\"}"}}'
4/4/2019, 7:34:01 PM - error: Zigbee publish to device '0x000d6ffffec939dc', genOnOff - off - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
4/4/2019, 7:34:01 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x000d6ffffec939dc","type":"device","friendlyName":"stairs_light2"},"message":"{\"state\": \"OFF\"}"}}'
4/4/2019, 7:34:09 PM - debug: Received MQTT message on 'zigbee2mqtt/stairs_light1/set' with data '{"state": "ON"}'
4/4/2019, 7:34:09 PM - debug: Received MQTT message on 'zigbee2mqtt/upstairs_light2/set' with data '{"state": "ON"}'
4/4/2019, 7:34:09 PM - debug: Received MQTT message on 'zigbee2mqtt/upstairs_light1/set' with data '{"state": "ON"}'
4/4/2019, 7:34:09 PM - debug: Received MQTT message on 'zigbee2mqtt/stairs_light2/set' with data '{"state": "ON"}'
4/4/2019, 7:34:09 PM - info: Zigbee publish to device '0x000d6ffffee7fc69', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:34:09 PM - info: Zigbee publish to device '0x000b57fffeefa1b3', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:34:10 PM - info: Zigbee publish to device '0x000d6ffffea7e904', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:34:10 PM - info: Zigbee publish to device '0x000d6ffffec939dc', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null
4/4/2019, 7:34:11 PM - error: Zigbee publish to device '0x000d6ffffee7fc69', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
4/4/2019, 7:34:11 PM - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Error: request timeout","meta":{"entity":{"ID":"0x000d6ffffee7fc69","type":"device","friendlyName":"stairs_light1"},"message":"{\"state\": \"ON\"}"}}'
4/4/2019, 7:34:12 PM - error: Zigbee publish to device '0x000b57fffeefa1b3', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - null failed with error Error: request timeout
```

antst, 2019-04-04:
I can detach/attach USB stick, restart zigbee2MQTT and it will work again.
Just restarting zigbee2MQTT will end up with infinite loop of "trying to recover" stuff.

GForce2010, 2019-04-04:
I've just had the same issue, just upgraded to latest firmware today. Everything was working fine before.

I had to unplug the USB adaptor earlier as well as pairing stopped as described in issue #1117

Koenkk, 2019-04-05:
In this case I recommend using the max stability firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/dev/coordinator

closed 2019-06-11 by stale[bot]