Koenkk/zigbee2mqtt #1372: How to send command to philips hue lamp

guitar9, 2019-04-04:
I have done all the things. I get this output 

```
 zigbee2mqtt:info 4/4/2019, 7:47:34 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
  zigbee2mqtt:info 4/4/2019, 7:48:03 PM MQTT publish: topic 'zigbee2mqtt/0x0017880104eb6195', payload '{"state":"ON","linkquality":89,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
  zigbee2mqtt:info 4/4/2019, 7:48:03 PM MQTT publish: topic 'zigbee2mqtt/0x0017880104eb6195', payload '{"state":"ON","linkquality":89,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
  zigbee2mqtt:info 4/4/2019, 7:48:03 PM MQTT publish: topic 'zigbee2mqtt/0x0017880104eb6195', payload '{"state":"ON","linkquality":89,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
  zigbee2mqtt:info 4/4/2019, 7:48:03 PM MQTT publish: topic 'zigbee2mqtt/0x0017880104eb6195', payload '{"state":"ON","linkquality":86,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
  zigbee2mqtt:info 4/4/2019, 7:48:03 PM MQTT publish: topic 'zigbee2mqtt/0x0017880104eb6195', payload '{"state":"ON","linkquality":89,"brightness":254,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":366,"color_mode":2}'
```
So this looks good. When i send to topic zigbee2mqtt/0x0017880104eb6195 this message through MQTT.fx
`{"state":"ON","linkquality":68,"brightness":100,"color":{"x":0.457,"y":0.41,"saturation":55.1,"hue":46.2},"color_temp":3,"color_mode":2}`, nothing happens. Somebody can help me?

jvdpas, 2019-04-04:
Publish to the "set" topic of the device.

guitar9, 2019-04-05:
thank you this works.
I can adjust the brightness but when i want to change the color nothing happens. 

{"state":"ON","linkquality":89,"brightness":255,"color":{"x":10.957,"y":0.21,"saturation":105.1,"hue":106.2},"color_temp":255,"color_mode":2}

closed 2019-04-07 by guitar9

dersimn, 2019-07-31:
Is your Hue Bulb always returning 

    {"state":"ON","linkquality":39,"brightness":255,"color":{"x":0.457,"y":0.41,"saturation":54.7,"hue":46.2},"color_temp":362,"color_mode":2}

as well?

No matter which color the bulb is set to, it always returns `{"x":0.457,"y":0.41,"saturation":54.7,"hue":46.2}` with `"color_mode":2`.

Color temperature works, Brightness works.