Koenkk/zigbee2mqtt #1374: One of my router/end-device work only with "permit_join" is set to "true"...!

falbalala, 2019-04-05:
After pairing device, swapping "_permit_join_" _true_ to _false_ & restart z2mqtt, all good.
But when i reset/OffOn the device, it's no longer accessible... For the device, the log is :

`AF data request fails, status code: 233. MAC no ack.`

The device appears **offline**.

What the problem...?

Thx in advance

> PS: I use the last release.

Bastiencc, 2019-04-05:
What device it is?
Everytime I get a error like `status code: 233. MAC no ack` I remove the device , restart Z2M, re-paire and it works again. Did you try that ?

falbalala, 2019-04-05:
It's a _DIY/custom_ device on :
- [x] cc2530/E18-MS1-PCB

It's right, but everytime that it's reset, the problem repeat.
You must then start the process again:

1. remove the device
2. restart Z2MQTT
3. re-paire
4. swapping "permit_join" to false

falbalala, 2019-04-17:
The update has fixed the problem.
Thx ;)

closed 2019-04-17 by falbalala