Koenkk/zigbee2mqtt #1375: Support Osram Smart+ plug

larsimmisch, 2019-04-05:
I got my zigbee2mqtt working. Yay!

But now I'm trying to control an Osram Smart+ switchable socket, and can't figure out how to configure that in homekit2mqtt. I did read the section about supporting new devices and I think I understand what https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html is trying to tell me, and when I press the button on the switch, I see:

'zigbee2mqtt:info 4/5/2019, 5:16:07 PM MQTT publish: topic 'zigbee2mqtt/0x84182600000a9077', payload '{"state":"OFF","linkquality":47}'`

So that looks pretty sensible.

But with this device configuration:
```
{
	"osram//buero/switch": {
		"name": "Buero",
		"manufacturer": "Osram",
		"model": "Smart+ plug (AB3257001NJ)",
		"services": [
			{
				"name": "Buero Steckdose",
				"service": "Switch",
				"topic": {
					"setOn": "zigbee2mqtt/0x84182600000a9077/set",
					"setOff": "zigbee2mqtt/0x84182600000a9077/set",
					"statusOn": "zigbee2mqtt/0x84182600000a9077/status",
					"statusOff": "zigbee2mqtt/0x84182600000a9077/status"
				},
				"payload": {
					"onTrue": 1,
					"onFalse": 0
				},
				"config": {},
				"props": {}
			}
		],
		"topicIdentify": "/set/lights/Bad/alert",
		"payloadIdentify": "select"
	}
}
```

The switch does not respond. Is there a good way to listen to other bridges to reverse engineer this? Or is there a good introduction to the MQTT message structure somewhere?

So far, this looks delightfully accessible, but the devil is in the details, I guess.

fred-c1, 2019-04-06:
My plug is working "OK".
I didn't have to look into "how_to_support_new_devices".
The only problem I have with Osram Smart+ plug is the range.
When the plug is more than about 7m away, the behavior is erratic.
For example it would turn ON and then turn OFF 3 seconds later.
I have this behavior with my 2 plugs.

closed 2019-06-12 by stale[bot]

juview, 2020-11-23:
Hi @larsimmisch , did you found a solution for this specific plug? Would be great if you can share it with as I'm struggeling to get it work.

larsimmisch, 2020-11-23:
No, I didn't. I gave up :(