Koenkk/zigbee2mqtt #1376: Aqara LED Bulb Not Discovered (ZNLDP12LM)

GreenVine, 2019-04-06:
Hi there,

Firstly thanks for such a gorgeous project, currently running the latest stable `zigbee2mqtt` on Docker and has 10 different Hue / Xiaomi devices connected without any problem.

I have purchased an Aqara LED Bulb (model name: ZNLDP12LM) and tried to pair it by switching on and off for 5 times. The bulb did blink twice, however nothing is showing in the log in terms of new incoming devices nor new entries in `configuration.yaml` and `database.db`. The bulb is around 4 meters away from the coordinator and 1 meter away from the nearest router. There is a Philips motion sensor just installed beneath the Aqara bulb and has been working perfectly.

```yaml
homeassistant: true
permit_join: true
mqtt: [redacted]
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  channel: 25
  log_level: debug
  last_seen: epoch
  elapsed: 15
```

Any ideas why it is not discovered by the coordinator?

Ref: https://github.com/Koenkk/zigbee2mqtt/issues/621

Perlovka, 2019-04-12:
I've succesfully paired one today. Manual says that you need turn ON|OFF 3 times, not 5. Then it starts blinking and pairs succesfully.

zigbee2mqtt version: 1.2.1

GreenVine, 2019-04-13:
> I've succesfully paired one today. Manual says that you need turn ON|OFF 3 times, not 5. Then it starts blinking and pairs succesfully.
> 
> zigbee2mqtt version: 1.2.1

Thanks, it looks like I have to pair it with the coordinator not the router.

closed 2019-06-19 by stale[bot]