Koenkk/zigbee2mqtt #1378: "undefined" model ID

mchacher, 2019-04-06:
Hi all,
I am using Zigbee2Mqtt for a while using "non supported" device. I am testing a new one and facing an unexpected issue.
I am not able to create a specific converter for this device as the "modelID" is not recognized and set to "undefined". See the log from Zigbee2Mqtt below. 
" zigbee2mqtt:warn 2019-4-6 13:49:56 Device with modelID 'undefined' is not supported."
One possible root cause could be that the model identifier attribute of the basic cluster is not set properly on this device. Do you confirm? Any advise to push further the investigation?
Thanks

tb-killa, 2019-04-06:
I see this behavior too.
What we could check if we do some sort of retry on https://github.com/Koenkk/zigbee-shepherd/blob/master/lib/components/event_handlers.js#L161 if our read failed.
@Koenkk what do you think? What would be the best test implementation?

GerSant, 2019-04-07:
The same for me when trying to pairing a CC2530 router

bizit524, 2019-04-07:
I have had the same issue. There are 2 solutions I have found. 1) Try repairing it multiple times. i've had to try and pair some devices more than 10 times. 2) Get a USB extension cable (if using the usb version) and move the cordinator/router away from all other electronics. This instantly helped me, I was having trouble pairing a few devices even though they were supported. As soon as I added the extension they started working instantly. It appears there was too much interference

he902, 2019-04-09:
In my case (device Yunmi Lock), the real modelID 'FNB56-DOR23FB1.3' comes just after the modelID 'undefined'.
After added modelID 'FNB56-DOR23FB1.3' into the device.js, the modelID 'undefined' is gone!
```
  zigbee2mqtt:info 4/5/2019, 6:01:09 PM Connected to MQTT server
  zigbee2mqtt:info 4/5/2019, 6:01:09 PM MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
  zigbee2mqtt:info 4/5/2019, 6:01:10 PM MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"log_level":"info","permit_join":true}'
  zigbee2mqtt:info 4/5/2019, 6:33:59 PM New device 'undefined' with address 0x00158d0002acxxxx connected!
  zigbee2mqtt:info 4/5/2019, 6:33:59 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002acxxxx","meta":{}}'
  zigbee2mqtt:warn 4/5/2019, 6:33:59 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 4/5/2019, 6:33:59 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info 4/5/2019, 6:34:02 PM Device incoming...
  zigbee2mqtt:info 4/5/2019, 6:34:02 PM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"device incoming"}'
  zigbee2mqtt:warn 4/5/2019, 6:34:02 PM Device with modelID 'FNB56-DOR23FB1.3' is not supported.
  zigbee2mqtt:warn 4/5/2019, 6:34:02 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

mchacher, 2019-04-09:
Hi all,
digging further in my issue, I have solved it while limiting the number of attributes requested to 2 (and not requesting  all of them ('zclVersion', 'appVersion', 'stackVersion', 'hwVersion', 'manufName', 'modelId', 'dateCode', 'powerSource','swBuildId'). 
So I wonder if the issue is not coming from the device itself being overloaded by too much requests at the same time ... 
Since we are mainly looking for the manufacturer name, I wonder if we should not split the requests to make this process more robust.
@tb-killa / @Koenkk : what's you view on this?

tb-killa, 2019-04-09:
@mchacher The best way would be some soft of: https://github.com/Koenkk/zigbee2mqtt/issues/1378?_pjax=%23js-repo-pjax-container#issuecomment-480524184

BTW: Please test with latest ``dev`` version and provide log file with ``DEBUG=zigbee* npm start``.
We include debug-log output for this.

mchacher, 2019-04-09:
@tb-killa 
Thanks. I will try to do it in the evening (Paris time).
Should I rebase all the projects, or only this file?
Thanks.

tb-killa, 2019-04-09:
Only this file from main base of zigbee shepherd because we query now all needing attr.

mchacher, 2019-04-09:
Hi @tb-killa 
I did the log very quickly. You can look at the attached file. Will all attributes the device is underfined. Just after decomissioning the device, the commissioning process started (line 6). I let you have a look. thanks for your support.
[log_mch.txt](https://github.com/Koenkk/zigbee2mqtt/files/3058633/log_mch.txt)

tb-killa, 2019-04-09:
hi @mchacher thanks for the fast answer.

Please modify this too and provide a new logfile with same debug parameters:
[https://github.com/Koenkk/zigbee-shepherd/pull/16/commits/b6f6e5a3dfac0eda734a1e49ee0cf77d1d9fec79](https://github.com/Koenkk/zigbee-shepherd/pull/16/commits/b6f6e5a3dfac0eda734a1e49ee0cf77d1d9fec79)

The genbasic interview only run if the device itself ``handlers.devIncoming``.
Maybe we should move this part to ``endDeviceAnnceHdlr `` too ??

@Koenkk we should do some more tests and build some rework basements.

mchacher, 2019-04-09:
hi @tb-killa 
thanks for your support again!
here you go with the log file and af.js file modified as requested with the latest version.
Let me know if it is ok.
[log_mch_02.txt](https://github.com/Koenkk/zigbee2mqtt/files/3060471/log_mch_02.txt)

JumpmanJunior, 2019-04-10:
At the moment I am not able to add an Ikea Tradfri remote to my network which was working in the past. I assume this is because of the same issue.

> 
> 2019-4-10 02:14:47 - debug: Received zigbee message of type 'endDeviceAnnce' with data '"0x000d6ffffe1d7895"' of device 'undefined' (0x000d6ffffe1d7895) of endpoint 1
> 2019-4-10 02:14:47 - warn: Device with modelID 'undefined' is not supported.
> 2019-4-10 02:14:47 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

bizziebis, 2019-04-10:
I'm unable to re-pair a Xiaomi Human Body sensor which didn't respond anymore. I thought the battery was not good anymore so I replaced it. Afterwards I spend 3 hours to try to get it paired again. Every single time this is the result:

```
2019-4-10 21:20:50 - warn: Failed to remove '0x00158d0001de730b', trying force remove...
2019-4-10 21:20:50 - info: Force removed 0x00158d0001de730b
2019-4-10 21:20:50 - info: Successfully removed 0x00158d0001de730b
2019-4-10 21:20:50 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d0001de730b"}'

2019-4-10 21:21:25 - error: Cannot get the Node Descriptor of the Device: 0x00158d0001de730b (Error: Timed out after 10000 ms)
2019-4-10 21:21:25 - error: Cannot get the Node Descriptor of the Device: 0x00158d0001de730b (Error: Timed out after 10000 ms)
2019-4-10 21:21:52 - info: New device 'undefined' with address 0x00158d0001de730b connected!
2019-4-10 21:21:52 - info: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0001de730b","meta":{}}'
2019-4-10 21:21:52 - warn: Device with modelID 'undefined' is not supported.
2019-4-10 21:21:52 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-10 21:21:52 - warn: Device with modelID 'undefined' is not supported.
2019-4-10 21:21:58 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-10 21:21:58 - warn: Device with modelID 'undefined' is not supported.
2019-4-10 21:21:58 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-10 21:22:04 - warn: Device with modelID 'undefined' is not supported.
2019-4-10 21:22:04 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-10 21:22:06 - warn: Device with modelID 'undefined' is not supported.
2019-4-10 21:22:06 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-10 21:22:06 - warn: Device with modelID 'undefined' is not supported.
2019-4-10 21:22:06 - warn: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
2019-4-10 21:22:12 - warn: Device with modelID 'undefined' is not supported.
```

Please note that I already use 5 of these sensors, so I know how to pair them. I know they can be a pain in the a** to pair, but this is getting anoying. 

I'm running 
**zigbee2mqtt version 1.3.0 (commit #da94252)**
**Coordinator firmware version: '20190223**

CodeFinder2, 2019-04-12:
I am facing the same problem with an Aqara door & window contact sensor (MCCGQ11LM) and I've already paired 6 of them more or less successfully (although a pain sometimes with many retries).

Log:
```
zigbee2mqtt:debug 4/12/2019, 1:54:30 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d0003166471"' of device 'undefined' (0x00158d0003166471) of endpoint 1
  zigbee2mqtt:warn 4/12/2019, 1:54:30 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 4/12/2019, 1:54:30 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 4/12/2019, 1:55:21 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d0003166471"' of device 'undefined' (0x00158d0003166471) of endpoint 1
  zigbee2mqtt:warn 4/12/2019, 1:55:21 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 4/12/2019, 1:55:21 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 4/12/2019, 1:56:35 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d0003166471"' of device 'undefined' (0x00158d0003166471) of endpoint 1
  zigbee2mqtt:warn 4/12/2019, 1:56:35 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 4/12/2019, 1:56:35 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:debug 4/12/2019, 1:56:53 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00158d0003166471"' of device 'undefined' (0x00158d0003166471) of endpoint 1
  zigbee2mqtt:warn 4/12/2019, 1:56:53 PM Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn 4/12/2019, 1:56:53 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

Perlovka, 2019-04-12:
Hi, I've paired today Aquara LED lamp.
Same issue, but device been added and works. Can control it via zigbee2mqtt/0x00158d0002d40d29 topic.

Here's the pairing log part:
```
  zigbee2mqtt:info Sat Apr 13 2019 01:08:00 GMT+0300 (IDT) New device 'undefined' with address 0x00158d0002d40d29 connected!
  zigbee2mqtt:info Sat Apr 13 2019 01:08:00 GMT+0300 (IDT) MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0002d40d29","meta":{}}'
  zigbee2mqtt:warn Sat Apr 13 2019 01:08:00 GMT+0300 (IDT) Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn Sat Apr 13 2019 01:08:00 GMT+0300 (IDT) Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:warn Sat Apr 13 2019 01:08:00 GMT+0300 (IDT) Device with modelID 'undefined' is not supported.
  zigbee2mqtt:warn Sat Apr 13 2019 01:08:00 GMT+0300 (IDT) Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
  zigbee2mqtt:info Sat Apr 13 2019 01:08:01 GMT+0300 (IDT) MQTT publish: topic 'zigbee2mqtt/0x00158d0002d40d29', payload '{"state":"ON","brightness":255,"color_temp":153,"linkquality":63}'
  zigbee2mqtt:info Sat Apr 13 2019 01:08:01 GMT+0300 (IDT) MQTT publish: topic 'zigbee2mqtt/0x00158d0002d40d29', payload '{"state":"ON","brightness":255,"color_temp":153,"linkquality":63}'
  zigbee2mqtt:info Sat Apr 13 2019 01:08:02 GMT+0300 (IDT) Device incoming...
```

CodeFinder2, 2019-04-13:
I just tried to pair multiple Aqara temperature/humidity sensors and it fails for the same reason ...

Is this a bug accidentally added in recent commits/releases? I cannot imagine having these massive problems previously ...

tb-killa, 2019-04-14:
@Koenkk Do some simple work at : https://github.com/tb-killa/zigbee-shepherd/commit/bf0747d197e10350816b6fce2bad403aae2e5c66
Please review. Maybe this would fix the Issues with the undefined.

Koenkk, 2019-04-14:
I found the actual cause of this issue, this is a bug introduced in 1.3.

@tb-killa surprisingly some end devices don't respond to a  read of the zclVersion, appVersion and stackVersion.

mchacher, 2019-04-14:
Hi all
This is consistent with one of my previous post. 
Why requesting so much info since only manufacturer code is the one we are looking for? 
Thanks

tb-killa, 2019-04-14:
@Koenkk that's why I build the code above do the requests in two steps. First do the minimal and after update we read the other attributes.
@mchacher we read them to get the needing value for sw and hw version info per device.

Koenkk, 2019-04-14:
@tb-killa can you make a PR?

tb-killa, 2019-04-14:
Will do need someone to first test.
Maybe you will fork and check them?

Koenkk, 2019-04-14:
@tb-killa yes I will test the PR, we can continue working further from there.

tb-killa, 2019-04-14:
https://github.com/Koenkk/zigbee-shepherd/pull/22

CodeFinder2, 2019-04-15:
@Koenkk Would you mind making this a minor v1.3.1 release so that the HassIO addon can be updated as well? This bug is quite severe and many people are affected by it.

Koenkk, 2019-04-16:
@CodeFinder2 yes, I will release a new version with this fix asap

Koenkk, 2019-04-16:
Fixed in 1.3.1

closed 2019-04-16 by Koenkk

tb-killa, 2019-04-16:
Downgrade fix issue ;)

Koenkk, 2019-04-16:
@tb-killa yes, I will do some further testing and integrate your pr in 1.4

@danielwelch can you update the hassio addon? this is quite a crucial issue.

mateuszwozniak, 2019-04-16:
Hi Guys, 
I've just tested 1.3.1 and unfortunately, I still have the issue when trying to use Aquara RTCGQ11LM human body sensor. 

I get this all the time:
```
Apr 16 23:05:24 openHABianPi npm[15407]:   zigbee2mqtt:warn 4/16/2019, 11:05:24 PM Device with modelID 'undefined' is not supported.
Apr 16 23:05:24 openHABianPi npm[15407]:   zigbee2mqtt:warn 4/16/2019, 11:05:24 PM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
```

What's interesting is that I tried to reset that sensor probably at least 50 times and it worked once, but after a couple of minutes it just stopped. After that I couldn't connect it anymore.

Let me know if you need help with some debugging, I can try to provide more info if needed.

Koenkk, 2019-04-17:
@mateuszwozniak before re-pairing you first need to remove it http://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

mateuszwozniak, 2019-04-17:
@Koenkk yes, I did that and it was successfully removed:
```
Apr 17 09:19:58 openHABianPi npm[16485]:   zigbee2mqtt:warn 4/17/2019, 9:19:58 AM Failed to remove '0x00158d0001e52c22', trying force remove...
Apr 17 09:19:58 openHABianPi npm[16485]:   zigbee2mqtt:info 4/17/2019, 9:19:58 AM Force removed 0x00158d0001e52c22
Apr 17 09:19:58 openHABianPi npm[16485]:   zigbee2mqtt:info 4/17/2019, 9:19:58 AM Successfully removed 0x00158d0001e52c22
Apr 17 09:19:58 openHABianPi npm[16485]:   zigbee2mqtt:info 4/17/2019, 9:19:58 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"0x00158d0001e52c22"}'
```
and the device was removed from `data/configuration.yml`

mateuszwozniak, 2019-04-17:
Ok, I've made a few more trials:
1. Removed the device via mqtt message
2. Stopped zigbee2mqtt
3. Started it again
4. Once it was ready I pressed reset button until it flashed 3 times (released the button 1s after 3rd flash)
5. Waited for a while (nothing showed in zigbee2mqtt logs for about 20s)
6. Then I saw this:
```
Apr 17 09:29:41 openHABianPi npm[4918]:   zigbee2mqtt:info 4/17/2019, 9:29:41 AM New device 'undefined' with address 0x00158d0001e52c22 connected!
Apr 17 09:29:41 openHABianPi npm[4918]:   zigbee2mqtt:info 4/17/2019, 9:29:41 AM MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":"0x00158d0001e52c22","meta":{}}'
Apr 17 09:29:41 openHABianPi npm[4918]:   zigbee2mqtt:warn 4/17/2019, 9:29:41 AM Device with modelID 'undefined' is not supported.
Apr 17 09:29:41 openHABianPi npm[4918]:   zigbee2mqtt:warn 4/17/2019, 9:29:41 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 17 09:29:42 openHABianPi npm[4918]:   zigbee2mqtt:warn 4/17/2019, 9:29:42 AM Device with modelID 'undefined' is not supported.
Apr 17 09:29:42 openHABianPi npm[4918]:   zigbee2mqtt:warn 4/17/2019, 9:29:42 AM Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Apr 17 09:29:43 openHABianPi npm[4918]:   zigbee2mqtt:info 4/17/2019, 9:29:43 AM MQTT publish: topic 'zigbee2mqtt/0x00158d0001e52c22', payload '{"illuminance":371,"linkquality":131}'
```
And it seems to be working correctly now.
I'm going to do a couple of tests in the evening like stopping/starting zigbee2mqtt and raspberry a few times to check if now keeps the device connected across restarts (as before it couldn't contact it after restart) and I'll let you know how it goes. 
Anyway, thanks for a great job on that!