Koenkk/zigbee2mqtt #1381: starting  Zigbee2mqtt as a service

bizit524, 2019-04-06:
Hello
This is not exactly a bug in the code. This is me being to dumb to properly resolve the issue. 
I have successfully flashed and setup zigbeee2mqtt with connected devices. 
When I run it using below command, it runs perfectly fine. I see output and see devices connecting and giving temps etc
cd /opt/zigbee2mqtt
npm start

The problem lies when I try to set it up as a service. I follow the guide which works fantastically on my pi. However I switched to a Vm running Ubuntu 18.04 (with the same install guides) it wont start as a service as it has a problem starting NPM. I have switched the user from pi to my own  Below is what mine looks like 
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=myuser

[Install]
WantedBy=multi-user.target

Using the systemctl status zigbee2mqtt.service
it gives this error. 

● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-04-06 18:25:36 UTC; 7s ago
  Process: 4764 ExecStart=/usr/bin/npm start (code=exited, status=1/FAILURE)
 Main PID: 4764 (code=exited, status=1/FAILURE)

Apr 06 18:25:36 ha systemd[1]: zigbee2mqtt.service: Main process exited, code=exited, status=1/FAILURE
Apr 06 18:25:36 ha systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 06 18:25:36 ha systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Apr 06 18:25:36 ha systemd[1]: zigbee2mqtt.service: Scheduled restart job, restart counter is at 5.
Apr 06 18:25:36 ha systemd[1]: Stopped zigbee2mqtt.
Apr 06 18:25:36 ha systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Apr 06 18:25:36 ha systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 06 18:25:36 ha systemd[1]: Failed to start zigbee2mqtt.

I have confirmed that /usr/bin/npm exists
I also tried /usr/lib/node_modules/npm/bin because I did a ls -R | grep npm which found that location as well but that did not work. 
Any suggestions? 

I will continue to troubleshoot while I await further advice

swissbyte, 2019-04-08:
try to start it with user=root

bizit524, 2019-04-08:
I will try this but I dont think I ever setup a specific root account. When I created the instance I created a different account. WHen I  run stuff as sudo is my account elevated to root permissions or does it run that command with the root user?
I will certainly try this. I understood as using SUDO would elevate the permissions of my current username but I could most certainly be wrong!
Many thanks for providing this.

Nephiel, 2019-04-08:
For what it's worth, I had a similar issue with a TMPFS `/var/log` mount, where `/var/log/zigbee2mqtt/` does not exist after a reboot, and needs to be created before starting zigbee2mqtt. Try something like this:

```
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/bin/npm --no-update-notifier start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=myuser
PermissionsStartOnly=true
ExecStartPre=-/bin/mkdir /var/log/zigbee2mqtt
ExecStartPre=-/bin/chown myuser:adm /var/log/zigbee2mqtt
ExecStartPre=-/bin/chmod 740 /var/log/zigbee2mqtt

[Install]
WantedBy=multi-user.target
```
With `PermissionsStartOnly` set to `true`, this will first execute the `ExecStartPre` commands as root, and then drop to user `myuser` to run the `ExecStart` command.
The `--no-update-notifier` flag is not related, it's just to prevent `npm` from creating `~/.config/configstore/update-notifier.npm.json`.

bizit524, 2019-04-08:
Many thanks for both of the suggestions. i have tried the root user and it works beautifully and now starts up automatically when restarted :)

closed 2019-04-08 by bizit524