Koenkk/zigbee2mqtt #1382: Tradfri Routers (E1746) don't work as replacements of Xiaomi Routers

omer1141, 2019-04-06:
I have tried to add 4 Ikea Tradfri Routers (E1746) on four different floors in my house (the CC2531 is on the first floor). I'm using version 1.2.1 with the suggested Firmware 20190223.
The router on the same floor as the CC2531 works without problems. The other three Ikea Routers I've 'learned in' close to the CC2531 and then I've moved them to the other floors. Unfortunately all of my Xiaomi end devices on the three upper floors  (presence sensors, movement sensor, cube, ...) don't really continue to work (I wanted to replace my four Xiaomi Wired-Switch routers with the smaller Ikea routers). Any help to get the Ikea routers to work would be appreciated.

See network diagram attached:
[image]

thissiac, 2019-04-09:
Maybe problems with WiFi interferences?

The small Xiaomi End Devices have no high range and are easily disturbed. Even at a distance of 3 meters they can be disturbed if there is a good Wifi source nearby.

omer1141, 2019-04-09:
I don't think that this can be the problem as I've put the Ikea routers into the same location where the Xiaomi wired switches (which I've used as routers before) were. Anybody else out therewith a bigger Ikea Tradfri  E1746 installation and no issues with Xiaomi end devices?

tunip, 2019-04-15:
We also have 4 floors and a few (~30) xiaomi end devices. Right now, I installed a CC2530 + CC2591 on every floor. But it is not easy to find a need case for it if you use them in the hallway. Thats why I am still looking for an alternative for the 2 hallways. Would buy 2 of this E1746 devices this week and test it.

omer1141, 2019-04-21:
Any luck with the E1746 devices on your side? It looks to me like the Ikea routers with direct connection to the CC2531 work, however if the Ikea router is further away and would need to be relayed through another Ikea router, things don't work anymore.

tunip, 2019-04-21:
Yes, they work fine here. Have 3 of them now. Coordinator (CC2530 + CC2591) is in the basement. And on every other floor a E1746 now.

omer1141, 2019-04-21:
Great. Do you also have indirect connections, i.e. E1746 <-> E1746 <-> CC2530+CC2591 ? I believe that the indirect connection does not work for me.

closed 2019-06-27 by stale[bot]