Koenkk/zigbee2mqtt #1383: Gledopto GL-C-008 state after power loss

raptr55, 2019-04-07:
Hi 

every time i have a power loss, the led strip connected on the gledopto GL-C-008, after the power goes back the led strip power ON but the state on Home assistant does not refresh

there's a way to refresh the state after a power loss and thers a way to specify a power on behaviour?

thanks

Koenkk, 2019-04-07:
You could check if reporting works for this device: http://www.zigbee2mqtt.io/information/report.html

raptr55, 2019-04-08:
Hi!

Report dont works... 😕

Koenkk, 2019-04-08:
Does a `zigbee2mqtt/[device_id]/get` `{"state": ""}` works?

raptr55, 2019-04-08:
> Does a `zigbee2mqtt/[device_id]/get` `{"state": ""}` works?

yup!

[image]

this is the log when the power goes off:

zigbee2mqtt:info 4/8/2019, 12:42:14 PM Zigbee publish to device '0x00124b001d3ac596', genLevelCtrl - read - [{"attrId":0}] - {"manufSpec":0,"disDefaultRsp":0} - 11

zigbee2mqtt:info 4/8/2019, 12:42:14 PM MQTT publish: topic 'zigbee2mqtt/Striscia Led letto', payload '{**"state":"OFF"**,"color_temp":500,"color":{"x":0.323,"y":0.329},"linkquality":55,"brightness":1}'

and this is what i get when i enable the report in the conf:

  zigbee2mqtt:info 4/8/2019, 12:48:46 PM MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe2bd139', payload '{"state":"OFF","linkquality":94,"brightness":254,"color_mode":2,"color":{"x":0.458,"y":0.41}}'

  zigbee2mqtt:info 4/8/2019, 12:50:45 PM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe51f715', payload '{"state":"OFF","linkquality":102,"brightness":68}'

  **zigbee2mqtt:info 4/8/2019, 12:52:14 PM MQTT publish: topic 'zigbee2mqtt/Striscia Led letto/availability', payload 'online'**

raptr55, 2019-04-20:
I can try anything else?

Koenkk, 2019-04-22:
If the reporting doesn't work, we can not now that the device has powered on again. (unless you see some other zigbee messages when the device comes online)

raptr55, 2019-04-27:
i set the log level to debug and when i restart the plugin i see this in the log:

  zigbee2mqtt:debug 4/27/2019, 2:44:11 PM Received MQTT message on 'zigbee2mqtt/0x00124b001d3ac596/set' with data '{"state": "ON"}'
  zigbee2mqtt:info 4/27/2019, 2:44:11 PM Zigbee publish to device '0x00124b001d3ac596', genOnOff - on - {} - {"manufSpec":0,"disDefaultRsp":0} - 11
  zigbee2mqtt:info 4/27/2019, 2:44:11 PM MQTT publish: topic 'zigbee2mqtt/Striscia Led letto', payload '{"state":"ON","color_temp":500,"color":{"x":0.323,"y":0.329},"linkquality":63,"brightness":1}'

and this after a power loss:

  zigbee2mqtt:debug 4/27/2019, 2:47:23 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00124b001d3ac596"' of device 'GL-C-007' (0x00124b001d3ac596)
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'readRsp' with data '{"cid":"genBasic","data":{"manufacturerName":"GLEDOPTO","modelId":"GL-C-007","powerSource":1}}' of device 'GL-C-007' (0x00124b001d3ac596)
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devChange' with data '{"cid":"genBasic","data":{"manufacturerName":"GLEDOPTO","modelId":"GL-C-007","powerSource":1}}' of device 'GL-C-007' (0x00124b001d3ac596)
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'
  zigbee2mqtt:debug 4/27/2019, 2:47:24 PM Received zigbee message of type 'devInterview' with data '"0x00124b001d3ac596"'

it can be usefull?

Koenkk, 2019-04-28:
Yes, I think you can already get this working by enabling the availability feature. (see `availability_timeout` http://www.zigbee2mqtt.io/configuration/configuration.html). After the device comes online, zigbee2mqtt polls the new state.

raptr55, 2019-04-28:
hi!
availability set on 30 seconds. this the log:

  zigbee2mqtt:debug 4/28/2019, 11:20:01 PM Ping 0x00124b001d3ac596
  zigbee2mqtt:debug 4/28/2019, 11:20:01 PM Successfully pinged 0x00124b001d3ac596
  zigbee2mqtt:info 4/28/2019, 11:20:01 PM MQTT publish: topic 'zigbee2mqtt/Striscia Led letto/availability', payload 'online'

but the state is still off:

[image]

Koenkk, 2019-04-29:
Did you remove the device from power? Otherwise I don't expect that it can ping it.

raptr55, 2019-04-29:
yeap , after the ping error message i replug the power but on the front-end the state remain off

4/29/2019, 8:28:38 AM - error: Failed to ping 0x00124b001d3ac596
4/29/2019, 8:28:38 AM - info: MQTT publish: topic 'zigbee2mqtt/Striscia Led letto/availability', payload 'offline'

Koenkk, 2019-04-29:
Can you provide the complete debug log containing (http://www.zigbee2mqtt.io/configuration/configuration.html):
- Device on
- Remove power from device
- Wait 2 mins
- Power device on

raptr55, 2019-04-29:
[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/3128539/log.txt)

there you go!

thanks :)

raptr55, 2019-04-30:
aaaaaaaaand the [configuration.txt](https://github.com/Koenkk/zigbee2mqtt/files/3132294/configuration.txt) that i forgot

Koenkk, 2019-05-01:
Should be fixed in latest dev branch, can you confirm?

raptr55, 2019-05-05:
> Should be fixed in latest dev branch, can you confirm?

Yeap!!! you made it.

set report true and availability timeout to 30 and after a power loss :
5/5/2019, 11:55:50 AM - error: Failed to ping 0x00124b001d3ac596

when i replug the power:
5/5/2019, 11:55:58 AM - info: MQTT publish: topic 'zigbee2mqtt/Striscia Led letto/availability', payload 'online'

[image]

this is the log just in case:
[log_2.txt](https://github.com/Koenkk/zigbee2mqtt/files/3145385/log_2.txt)

thanks!!!!

raptr55, 2019-05-05:
sorry, i've made another test

if i unplug the power and replug the power before the ping test the state in the front end still says the led strip is off:

[image]

this is the log when i unplug and replug the power:

  zigbee2mqtt:info 5/5/2019, 12:16:11 PM MQTT publish: topic 'zigbee2mqtt/0xd0cf5efffe2bd139/availability', payload 'online'
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Ping 0x000b57fffe51f715 (default)
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Successfully pinged 0x000b57fffe51f715
  zigbee2mqtt:info 5/5/2019, 12:16:24 PM MQTT publish: topic 'zigbee2mqtt/0x000b57fffe51f715/availability', payload 'online'
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Received zigbee message of type 'endDeviceAnnce' with data '"0x00124b001d3ac596"' of device 'GL-C-007' (0x00124b001d3ac596) of endpoint 11
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Setting up reporting for 0x00124b001d3ac596
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Checking if 0x00124b001d3ac596 is pingable
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM 0x00124b001d3ac596 is pingable (true) not router or battery powered
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Setup reporting for 0x00124b001d3ac596 - genOnOff - 1
  zigbee2mqtt:debug 5/5/2019, 12:16:24 PM Setup reporting for 0x00124b001d3ac596 - genLevelCtrl - 1
  zigbee2mqtt:debug 5/5/2019, 12:16:25 PM Setup reporting for 0x00124b001d3ac596 - lightingColorCtrl - 3
  zigbee2mqtt:debug 5/5/2019, 12:16:25 PM Successfully setup reporting for 0x00124b001d3ac596 - genOnOff - 1
  zigbee2mqtt:debug 5/5/2019, 12:16:25 PM Successfully setup reporting for 0x00124b001d3ac596 - genLevelCtrl - 1
  zigbee2mqtt:debug 5/5/2019, 12:16:26 PM Successfully setup reporting for 0x00124b001d3ac596 - lightingColorCtrl - 3

Koenkk, 2019-05-06:
Should be fixed in latest dev, can you confirm?

raptr55, 2019-05-07:
> Should be fixed in latest dev, can you confirm?

nope... it only works after a ping test error, you need the log? what i can do to make your job easier?

Koenkk, 2019-05-07:
Are you sure you are on the latest dev? If not, please post your debug log.

raptr55, 2019-05-07:
my bad... seems like i did not install the latest dev version... everything's ok!!!

closed 2019-05-07 by raptr55

jorgespneto, 2020-05-04:
I am having this same problem on several Gledopto devices, I would like to ask if you have any solution for this problem. Can be via webcore or smartapp.