Koenkk/zigbee2mqtt #1389: The CC2531 dongle reconnects once every minute and zigbee2mqtt fails to start.

alexisayenko, 2019-04-08:
# Bug Report
## What happened
In short:
The CC2531 dongle "stopped working" after a month. I tried to plug in on both Raspberry Pi and Windows machine. On Windows it gives sound like it disconnects every minute and the green LED blinks when it disconnects. I suspect it might be related to the issue that z2m cannot start.

Details:
I am running the system on Raspberry Pi 3 B+ for about a month. I flashed the CC2531 with the recommended firmware (I don't remember the version and don't know how to check it, would appreciate if you tell me). After a month I noticed that z2m does not start with message "request timeout" (error log is https://pastebin.com/FZEwvR5P). Also in the syslog there is the following line: 
   kernel: [33158.839756] cdc_acm 1-1.1.3:1.0: failed to set dtr/rts

I tried to restart RPi, tried pressing Reset button. Reinstalled npm. Nothing helped. Then I connected the CC2531 to my laptop running on Windows 10, installed (no npm, no z2m installed). And I noticed that it disconnects once every minute.

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug Info
zigbee2mqtt version: 1.1.1
CC253X firmware version: ?

Koenkk, 2019-04-09:
I would recommend switching to the max stability firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator

leonols, 2019-04-09:
I have the same problem for 2 days, from the moment of Raspbian update. I have last firmware.

alexisayenko, 2019-04-26:
Koenkk were right. I reflashed my CC2531 with the mentioned firmware and it is working now. Thank you

closed 2019-07-02 by stale[bot]