Koenkk/zigbee2mqtt #1390: Binding not possible with WXKG02LM - Xiaomi Aqara double key wireless wall switch?

diplix, 2019-04-08:
# Bug Report
when i send 
`zigbee2mqtt/bridge/bind/xiaomi_schalter_2/right`
with a payload of `xiaomi_schalter_3`
nothing happens. the log only acknoledges, that it received the message:
`4/8/2019, 8:43:22 PM - debug: Received MQTT message on 'zigbee2mqtt/bridge/bind/xiaomi_schalter_2/right' with data 'xiaomi_schalter_3'`
this also happens for 
`zigbee2mqtt/bridge/bind/xiaomi_schalter_2`

does the WXKG02LM not (yet) support binding?

## Debug Info
zigbee2mqtt version: 1.3.0 (commit #00ebd44)
CC253X firmware version: 20190223

Koenkk, 2019-04-09:
These probably dont support binding

aktentasche, 2019-04-09:
Not sure if it helps but I am binding my Xiaomi key switches (no relay, i.e. WXKG03LM and 02LM) by just setting permit_join to true, restart service and then long press the button. The two blue LEDs indicate the process (both blink, then both on, then one on, then the other on). Even though the zigbee2mqtt log states that something went wrong I can use the switches as expected:

`Apr 09 23:34:15 betty npm[28708]:   zigbee2mqtt:info 4/9/2019, 11:34:15 PM MQTT publish: topic 'zigbee2mqtt/xiaomi_double_click_1', payload '{"battery":100,"voltage":3045,"linkquality":60,"click":"left"}'`

config.yaml:
```
'0x00158d0002XXXXXX':
    friendly_name: xiaomi_double_click_1
    retain: false
```

LarsNorgaard, 2019-04-09:
> Not sure if it helps but I am binding my Xiaomi key switches (no relay, i.e. WXKG03LM and 02LM) by just setting permit_join to true, restart service and then long press the button until the LEDs start blinking (I think both in case of the double switch). The two blue LEDs indicate the process (both blink, then both on, then one on, then the other on). Even though the zigbee2mqtt log states that something went wrong I can use the switches as expected:
> 
> ```
> Apr 09 23:34:15 betty npm[28708]:   zigbee2mqtt:info 4/9/2019, 11:34:15 PM MQTT publish: topic 'zigbee2mqtt/xiaomi_double_click_1', payload '{"battery":100,"voltage":3045,"linkquality":60,"click":"left"}'
> ```
> config.yaml:
> 
> ```
> '0x00158d0002XXXXXX':
>     friendly_name: xiaomi_double_click_1
>     retain: false
> ```

Binding is when the switch connects directly to ie. zigbee bulb.

Not many devices support binding, but you can read more about it here:
http://www.zigbee2mqtt.io/information/binding.html

diplix, 2019-04-10:
@Koenkk that makes sense. i was hoping _binding_ was a general feature of zigbee devices. closing this, since it’s neither a bug nor a feature that can be implemented.

closed 2019-04-10 by diplix

derrij, 2025-11-04:
But binding was working then i use WXKG02LM with Xiaomi Mi Home Gateway 2 DGNWG02LM.
With the move to zigbee2mqtt, this option is no longer available.