Koenkk/zigbee2mqtt #1391: One of my Aqara door sensors won't pair correctly

andreasbrett, 2019-04-08:
I have one stubborn Aqara door sensor that just won't pair correctly. It worked some weeks ago though. I removed it via MQTT remove command, removed the entries from database.db and re-paired through resetting it.

However the sensor will be detected as "undefined" device. Checking my logs I see some strange behaviour. There are messages by the sensor that contain its correct modelId "lumi.sensor_magnet.aq2" but yet the log says device is "undefined".

`debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3045,"3":30,"4":424,"5":184,"6":[0,0],"10":12142,"100":1},"modelId":"lumi.sensor_magnet.aq2"}}' of device 'undefined' (0x00158d000252b225) of endpoint 1
`

Koenkk, 2019-04-10:
The modelId reported in the `genBasic` `attReport` is something that is Xiaomi specific and is currently not recognized to set the modelId in `database.db`  (should be done in the future to improve pairing of Xiaomi devices). 

I would recomend to remove the device again (also from `database.db`) and try re-pairing it.

andreasbrett, 2019-04-10:
Okay, will try for the x-th time with this stubborn sensor. If it's not working I'll just wait until your implementation is done.

https://github.com/Koenkk/zigbee2mqtt/issues/1401

closed 2019-04-11 by andreasbrett