Koenkk/zigbee2mqtt #1392: Please commit your changes or stash them before you merge.

z4rn0x, 2019-04-09:
Hello,
Trying to update to the new rev 1.3.0, I get an error when i enter "git pull"

Updating 4048cb8..00ebd44
error: Your local changes to the following files would be overwritten by merge:
        .eslintrc.json
        .github/ISSUE_TEMPLATE.md
        index.js
        lib/controller.js
        lib/extension/bridgeConfig.js
        lib/extension/deviceAvailability.js
        lib/extension/deviceBind.js
        lib/extension/deviceGroupMembership.js
        lib/extension/devicePublish.js
        lib/extension/deviceReceive.js
        lib/extension/deviceReport.js
        lib/extension/homeassistant.js
        lib/extension/livolo.js
        lib/extension/xiaomi.js
        lib/state.js
        lib/util/settings.js
        lib/util/utils.js
        lib/zigbee.js
        test/controller.test.js
        test/data.test.js
        test/deviceAvailability.test.js
        test/devicePublish.test.js
        test/deviceReceive.test.js
        test/homeassistant.test.js
        test/settings.test.js
        test/utils.js
Please commit your changes or stash them before you merge.
Aborting

Anything I should do?
Thanks for your help

andreasbrett, 2019-04-09:
Please just use the update.sh script that comes with zigbee2mqtt.

z4rn0x, 2019-04-11:
> Please just use the **update.sh** script that comes with zigbee2mqtt.
> 
> https://github.com/Koenkk/zigbee2mqtt/blob/master/update.sh

That didn't solve my problem...
The method 1 I was referring to didn't solve it either, I got the same error, so I tried this:
git stash
git stash drop
and problem solved!
Thanks anyway

closed 2019-04-11 by z4rn0x

blauhorn, 2024-10-13:
Just to complete the list:
I had the same error during today's update via update.sh
Unpleasant side effect: 
- the cancelled update creates a data-backup folder with the current configuration.
- A restart of the update.sh script now requires the deletion of the old data-backup, which means that there is now only a standard new configuration in the data directory.
- Fortunately, I had an additional backup before running update.sh so that I could restore my configuration.
So please remember when this error occurs: always restore the data manually after the cancelled update or save it to a safe place so that it can be restored after the update.