Koenkk/zigbee2mqtt #1393: Update Error from 1.2.1 to 1.3.0  error cb() never called!

mpuff, 2019-04-09:
Hello,

i have done an update from 1.2.1 to 1.3.0 and get an error.

last log lines:

> 9576 silly extract lodash@4.17.11 extracted to /opt/zigbee2mqtt/node_modules/.staging/lodash-d62e4f52 (42865ms)
9577 silly extract fsevents@1.2.7 extracted to /opt/zigbee2mqtt/node_modules/.staging/fsevents-c353231d (32109ms)
9578 silly extract rxjs@6.4.0 extracted to /opt/zigbee2mqtt/node_modules/.staging/rxjs-21bfd94b (95712ms)
9579 timing npm Completed in 231710ms
9580 error cb() never called!
9581 error This is an error with npm itself. Please report this error at:
9582 error <https://npm.community>

i had installed nodejs 10.5.0.
now i have updatet to 10.15.3 (latest), and tried to update again, but the same error appears.

the service logs tell me these infos:

> Apr 09 10:58:41 zigbee2mqttpi02 systemd[1]: Started zigbee2mqtt.
Apr 09 10:58:46 zigbee2mqttpi02 npm[2620]: > zigbee2mqtt@1.3.0 start /opt/zigbee2mqtt
Apr 09 10:58:46 zigbee2mqttpi02 npm[2620]: > node index.js
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: internal/modules/cjs/loader.js:584
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: throw err;
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: ^
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: Error: Cannot find module 'semver'
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Function.Module._resolveFilename (internal/modules/cjs/loader.js:582:15)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Function.Module._load (internal/modules/cjs/loader.js:508:25)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Module.require (internal/modules/cjs/loader.js:637:17)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at require (internal/modules/cjs/helpers.js:22:18)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Object.<anonymous> (/opt/zigbee2mqtt/index.js:1:78)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Module._compile (internal/modules/cjs/loader.js:701:30)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Object.Module._extensions..js (internal/modules/cjs/loader.js:712:10)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Module.load (internal/modules/cjs/loader.js:600:32)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at tryModuleLoad (internal/modules/cjs/loader.js:539:12)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: at Function.Module._load (internal/modules/cjs/loader.js:531:3)
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! code ELIFECYCLE
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! errno 1
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! zigbee2mqtt@1.3.0 start: `node index.js`
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! Exit status 1
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR!
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! Failed at the zigbee2mqtt@1.3.0 start script.
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR! A complete log of this run can be found in:
Apr 09 10:58:47 zigbee2mqttpi02 npm[2620]: npm ERR!     /home/pi/.npm/_logs/2019-04-09T08_58_47_490Z-debug.log
Apr 09 10:58:47 zigbee2mqttpi02 systemd[1]: zigbee2mqtt.service: main process exited, code=exited, status=1/FAILURE
Apr 09 10:58:47 zigbee2mqttpi02 systemd[1]: Unit zigbee2mqtt.service entered failed state.
Apr 09 10:58:47 zigbee2mqttpi02 systemd[1]: zigbee2mqtt.service holdoff time over, scheduling restart.
Apr 09 10:58:47 zigbee2mqttpi02 systemd[1]: Stopping zigbee2mqtt...

Koenkk, 2019-04-10:
After updating, did you do a `rm -rf node_modules && npm install` ?

mpuff, 2019-04-10:
i have done the update with the update scribt in the folder, i have not doe this with this command

Koenkk, 2019-04-10:
Can you run the command in the zigbee2mqtt folder?

mpuff, 2019-04-10:
thank you, this works

closed 2019-04-10 by mpuff