Koenkk/zigbee2mqtt #1394: aquara sensor shows unknown after HA update from 0.90.2 to 0.91.2

hajo62, 2019-04-09:
I did the update of Home Assistant (v0.90.2 to v0.91.2)
My aquara sensor shows unknown.
When I did the systemctl status zigbee2mqtt.service, I saw published values.

After downgrading the HA back to v0.90.2, the sensor worked well as before.

My sensors.yaml: https://community.home-assistant.io/t/how-to-declare-xiaomi-aquara-temperatur-sensor-with-zigbee2mqtt/106573/6

Koenkk, 2019-04-10:
Can you post the zigbee2mqtt log of the sensor publishing it values? For me HA 0.91.x works fine.

hajo62, 2019-04-10:
Koenkk, did the update to HA 0.91.3 and the sensors are online...

hajo62, 2019-04-11:
Hhm, today, I did a restart of zigbee2mqtt and the sensor isn't online. Even not after the reboot of the Pi.
Here the output:
```
systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2019-04-11 16:48:39 CEST; 7s ago
 Main PID: 1422 (npm)
   CGroup: /system.slice/zigbee2mqtt.service
           ├─1422 npm
           ├─1433 sh -c node index.js
           └─1434 node index.js

Apr 11 16:48:39 raspberrypi systemd[1]: Started zigbee2mqtt.
Apr 11 16:48:40 raspberrypi npm[1422]: > zigbee2mqtt@1.2.1 start /opt/zigbee2mqtt
Apr 11 16:48:40 raspberrypi npm[1422]: > node index.js
Apr 11 16:48:41 raspberrypi npm[1422]:   zigbee2mqtt:info 2019-4-11 16:48:41 Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-11.16-48-41'
Apr 11 16:48:42 raspberrypi npm[1422]:   zigbee2mqtt:info 2019-4-11 16:48:42 Starting zigbee2mqtt version 1.2.1 (commit #b97f5d1)
Apr 11 16:48:42 raspberrypi npm[1422]:   zigbee2mqtt:info 2019-4-11 16:48:42 Starting zigbee-shepherd
Apr 11 16:48:42 raspberrypi npm[1422]:   zigbee2mqtt:info 2019-4-11 16:48:42 Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
```

The log (from next restart) does not help, too:
```
2019-4-11 16:50:48 - info: Logging to directory: '/opt/zigbee2mqtt/data/log/2019-04-11.16-50-48'
2019-4-11 16:50:49 - info: Starting zigbee2mqtt version 1.2.1 (commit #b97f5d1)
2019-4-11 16:50:49 - info: Starting zigbee-shepherd
2019-4-11 16:50:49 - info: Error while starting zigbee-shepherd, attempting to fix... (takes 60 seconds)
```

Koenkk, 2019-04-12:
Could you re-plug the stick? Somehow it fails to open the port, could some other process be locking the port?

When starting with ` DEBUG=* npm start` make sure you have stopped the zigbee2mqtt service, you cannot start it twice.

hajo62, 2019-04-12:
Replug doesn't change anything and yes I had stop the service before.

- What I did today: Recovered state HA 0.90.2 -> zigbee2mqtt worked well
- Updated to HA 0.91.3 -> zigbee2mqtt stopped working

Figured out, that 
```
sudo ls -l /dev/ttyACM0
ls: Zugriff auf '/dev/ttyACM0' nicht möglich: Datei oder Verzeichnis nicht gefunden
```
The stick is now available as /dev/ttyACM1.  
I have changed the configuration.yaml and the sensor is online again.

**Any idea what could cause this change from /dev/ttyACM0 to /dev/ttyACM1?**

Koenkk, 2019-04-13:
Do you have other serial devices connected?

hajo62, 2019-04-13:
No, not connected.
I had configured a z-wave stick, but since I'm playing around with zigbee2mqtt, never plugged in and commented out the lines...
When zigbee2mqtt is running and stable, that would be the next challenge: How to ensure, that both sticks are running...

But back to zigbee2mqtt:
Made a diskdump and restarted the Pi today. -> /dev/ttyACM0 but was configured to /dev/ttyACM1
I changed the configuration and started zigbee2mqtt. Did not help.
Booted he Pi. Again /dev/ttyACM1    Argh...

Now - this moment - it's configured to /dev/ttyACM0, works well, when I start it with `sudo DEBUG=* npm start`. But when I start it with `sudo systemctl start zigbee2mqtt.service` it does not come up.
```
sudo systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2019-04-13 10:51:12 CEST; 5s ago
  Process: 7605 ExecStart=/usr/local/bin/npm start (code=exited, status=1/FAILURE)
 Main PID: 7605 (code=exited, status=1/FAILURE)

Apr 13 10:51:12 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
Apr 13 10:51:12 raspberrypi systemd[1]: zigbee2mqtt.service: Service hold-off time over, scheduling restart.
Apr 13 10:51:12 raspberrypi systemd[1]: Stopped zigbee2mqtt.
Apr 13 10:51:12 raspberrypi systemd[1]: zigbee2mqtt.service: Start request repeated too quickly.
Apr 13 10:51:12 raspberrypi systemd[1]: Failed to start zigbee2mqtt.
Apr 13 10:51:12 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Apr 13 10:51:12 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

Koenkk, 2019-04-13:
Perhaps you can use the serial id like mentioned here in step 1 http://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

hajo62, 2019-04-13:
Thanks a lot for your support, that sound good!

**But** I still have the problem, that manual starting works well, but as service does fail:
```
cat /etc/systemd/system/zigbee2mqtt.service
[Unit]
Description=zigbee2mqtt
After=network.target

[Service]
ExecStart=/usr/local/bin/npm start
WorkingDirectory=/opt/zigbee2mqtt
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
```
results in
```
sudo systemctl status zigbee2mqtt.service
● zigbee2mqtt.service - zigbee2mqtt
   Loaded: loaded (/etc/systemd/system/zigbee2mqtt.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Sat 2019-04-13 17:45:12 CEST; 36ms ago
  Process: 3159 ExecStart=/usr/local/bin/npm start (code=exited, status=1/FAILURE)
 Main PID: 3159 (code=exited, status=1/FAILURE)

Apr 13 17:45:12 raspberrypi systemd[1]: zigbee2mqtt.service: Unit entered failed state.
Apr 13 17:45:12 raspberrypi systemd[1]: zigbee2mqtt.service: Failed with result 'exit-code'.
```

closed 2019-04-13 by hajo62