Koenkk/zigbee2mqtt #1395: Availability only partly published in version 1.3

didiht, 2019-04-09:
After updating from version 1.2.1 to version 1.3 I observed that the availability state was published for only one router device.

I found the bug in the file **lib/extension/deviceAvailability.js**

After changing some code (see below) the availbility functionality works as expected again😄 

```
raspi:/opt/zigbee2mqtt/lib/extension $ diff deviceAvailability.js deviceAvailability.js.v1.3
87,89c87
< //    setTimer(ieeeAddr) {
<     setTimer(device) {
<         const ieeeAddr = device.ieeeAddr;
---
>     setTimer(ieeeAddr) {
95,96c93
< //            this.handleInterval(ieeeAddr);
<             this.handleInterval(device);
---
>             this.handleInterval(ieeeAddr);
```

Koenkk, 2019-04-09:
Thanks for reporting!

Can you check if this has been fixed in the latest dev branch?

didiht, 2019-04-10:
@Koenkk You are welcome!

Reviewed and tested, works perfectly.

closed 2019-04-10 by didiht