Koenkk/zigbee2mqtt #1398: Osram Smart+ Plug  does not work as router

MeisterQ, 2019-04-09:
## What happened

I bought some Osram Smart+ Plugs and after putting them into the socket while watching the log of zigbee2mqtt showed me this log
https://pastebin.com/DdpSS3St

After this, i got its message with its state and linkquality payload.

Then i tried to switch it on or off via mqtt. Worked perfect. Very nice!

Then i took the Plug out the socket, brought it somewhere between the place where i does not have any signals and where i wanted to put some zigbee stuff.

After putting it into the Socket again, it seemd to be connected. I tried it out via switching the plug on and off via mqtt. Worked, and got the feedback messages.

Well. Then i tried to switch my Xiaomi Mijia Door/Window sensors close to the plug (I never had any signals at this place)

Well... I still had no signal at this place then... Weird

## What did you expect to happen

I expected increased zigbee range. Didnt happen.

## How to reproduce it (minimal and precise)

Deleted the device from the list, did a new pair.

Can still switch it on/off via MQTT but it does not route any signals to my CC2531 stick.
I also waited afew hours/days and tried many times. No change.

I removed my Mijia DWS and paired again close to the Osram Smart+ Plug, still no change.

## Debug Info
zigbee2mqtt version: Updated today (09.04.2019) // DD.MM.YYYY
CC253X firmware version: CC2531ZNP-Prod.hex (23.02.2019) // DD.MM.YYYY

mpuff, 2019-04-10:
Hello,

i have the OSRAM Smart+ Gardenpole RGBW, it is connectet to zigbee2mqtt, but i cant control it.
i also get no data from the Osram light when i subscribe.

in the graphviz map i see it connected and online in the network.

andreasbrett, 2019-04-11:
It takes some time for the zigbee network mesh to rebuild/optimize. Did you give it enough time to let your door/window sensors find the new route (via SMART+ router to the coordinator)?

I have 2 SMART+ power sockets and some of my lights were connected to it at some point. I figured that out through the network map feature of zigbee2mqtt.

MeisterQ, 2019-04-11:
> It takes some time for the zigbee network mesh to rebuild/optimize. Did you give it enough time to let your door/window sensors find the new route (via SMART+ router to the coordinator)?
>
> I have 2 [...]

How much time? They are plugged in since days

andreasbrett, 2019-04-11:
That should be plenty of time.

closed 2019-06-17 by stale[bot]

Smiggel, 2019-12-08:
Same issue here for me. I have the same plug. I see it has a good signal (255) from my router to the plug, but it has not route back. It's 0/1. It looks like it's a one way device.