Koenkk/zigbee2mqtt #1401: Improve device pairing

Koenkk, 2019-04-10:
Some devices are hard to pair, this should be improved. This issue is created in order to gather the points of improvement.

- Xiaomi devices are hard to pair because the `modelId` can sometimes not be retrieved via the `simpleDesc` request (e.g. #1391). However, they also report the `modelId` through a `genBasic` `attReport` message. We should use this message and take the `modelId` from it when it's not present in `database.db` yet. E.g. `debug: Received zigbee message of type 'attReport' with data '{"cid":"genBasic","data":{"65281":{"1":3045,"3":30,"4":424,"5":184,"6":[0,0],"10":12142,"100":1},"modelId":"lumi.sensor_magnet.aq2"}}' of device 'undefined' (0x00158d000252b225) of endpoint 1`

ReneHonig, 2019-04-10:
that would be great, I often have great difficulty pairing the Xiaomi devices for the reasons stated above. Once they are paired, I find them very reliable with very clever battery management.

tb-killa, 2019-04-10:
https://github.com/Koenkk/zigbee2mqtt/issues/1378

Koenkk, 2019-04-17:
Implemented in the dev branch!

closed 2019-05-02 by Koenkk